OpenZWave/open-zwave #1036: Cannot find endpoint map to instance for Command Class... while accessing Qubino ZMNHTDx device

sbouchex, 2016-11-09:
I'm using the a Qubino ZMNHTDx device and from the OpenZWave Control Panel log, I see a log of:

2016-11-09 15:31:36.288 Error, Node003, Cannot find endpoint map to instance for Command Class COMMAND_CLASS_METER endpoint 1
2016-11-09 15:31:37.305 Detail, Node003,   Received: 0x01, 0x18, 0x00, 0x04, 0x00, 0x03, 0x12, 0x60, 0x0d, 0x01, 0x01, 0x32, 0x02, 0x41, 0x24, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xca
2016-11-09 15:31:37.305 Detail,

I'm not able to receive METER notification or send switch command.

Fishwaldo, 2016-12-19:
Need the full log file please (after removing zwcfg_*.xml or ozwcache_*.xml) and restarting

fbernard001, 2017-05-05:
Hello,
i have the same issue, how can i help in the resolution of this issue, how should i process to give you the requested logs ?

Netbulae, 2017-05-11:
Try disabling the endpoint mapping in ZMNHTDx.xml

` `
``

fbernard001, 2017-05-12:
Hello,
do i need to exclude and include again the device ? because tried just by changing the file and reboot system but nothing has changed

nechry, 2017-05-12:
Yes you must exclude and include again to got changes. But I'm mostly sure this eill not fix your issues...

Netbulae, 2017-05-12:
We've got the meter working with this config:

https://github.com/Netbulae/open-zwave/pull/1/commits/229175c3cf5d2abcb9197831a16915796c7341bb

Don't know if this is required but we followed the following steps to update the config:

- Update the xml
- stop the service
- remove the "zwcfg_*.xml" file
- Start service again and exclude -> include

nechry, 2017-05-12:
please get latest version 
stop the service
remove the "zwcfg_*.xml" file
and tell me if everything is fine

fbernard001, 2017-05-12:
Thanks all for your help, 
ill try this when back home, one question:  Removing this file will not make all my devices disappear ? because i have about 50 zwave devices and don't want to have to include them all again :)

nechry, 2017-05-12:
in this case you can only exclude and include again.

nechry, 2017-05-18:
shall I close this issue?

fbernard001, 2017-05-18:
Please not for now because the device is plugged between main power switch and the rest of my house and i need to find a suitable moment to shutdown everything and remove the device from the zwzve network.

I have tried with everything online but can’t remove it !

 

Cordialement / Kind regards, 

Franck

 

 

 <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> 

Garanti sans virus.  <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> www.avg.com

nechry, 2017-05-18:
Ok now rush, it's for my case following

Netbulae, 2017-05-18:
Now we removed the endpoint mapping, but maybe we need to fix the association. I haven't figured out how to implement the tip below though. Can I do this in the xml itself?

https://www.domoticz.com/forum/viewtopic.php?t=11951#

> Re: ZMNHBDx - meters and power don't match switches
> Postby juankar » Monday 15 August 2016 1:03
> 
> The refresh solution works, but I think association is better
> For my Qubino I have group 1 to node 1 (life line), don't change this... and the others groups are associated to node 1.1 (like an isntance of controler, node 1).
> This works for me and I don't need any refresh in config file.

nechry, 2017-12-26:
ping

Fishwaldo, 2018-10-26:
Closing due to lack of feedback

closed 2018-10-26 by Fishwaldo

Andor, 2020-07-30:
Hello!
Very similar issue with the latest version of Home Assistant (0.113.2).
It founds only `sensor.qubino_goap_zmnhxdx_3_phase_smart_meter_energy` entity and I see errors like this when I press "refresh node":
```
2020-07-30 23:08:49.027 Detail, Node002,   Received: 0x01, 0x0a, 0x00, 0x04, 0x00, 0x02, 0x02, 0x98, 0x40, 0xb5, 0x00, 0x9c
2020-07-30 23:08:49.027 Info, Node002, Received SecurityCmd_NonceGet from node 2
2020-07-30 23:08:49.027 Info, NONCES: 0x17, 0x67, 0x14, 0x5a, 0x31, 0xbf, 0x21, 0xee
2020-07-30 23:08:49.027 Info, NONCES: 0x07, 0x73, 0xd4, 0x7d, 0x1e, 0x7a, 0x4d, 0x85
2020-07-30 23:08:49.027 Info, NONCES: 0x2a, 0xc1, 0x33, 0x32, 0xd3, 0xbf, 0x53, 0x95
2020-07-30 23:08:49.028 Info, NONCES: 0x24, 0x4e, 0x4f, 0x6a, 0x84, 0x7b, 0x70, 0xb0
2020-07-30 23:08:49.028 Info, NONCES: 0xfd, 0xf7, 0xd7, 0x37, 0x89, 0xd4, 0x55, 0xab
2020-07-30 23:08:49.028 Info, NONCES: 0x93, 0x82, 0x06, 0xc1, 0x84, 0xca, 0x2d, 0xf5
2020-07-30 23:08:49.028 Info, NONCES: 0xd9, 0xe7, 0x08, 0x68, 0x9a, 0xc5, 0x04, 0x16
2020-07-30 23:08:49.028 Info, NONCES: 0xd4, 0x24, 0x47, 0x4c, 0x66, 0x6f, 0x25, 0x9b
2020-07-30 23:08:49.028 Info, Node002, Sending (Send) message (Callback ID=0x01, Expected Reply=0x00) - Nonce_Report - 0x01, 0x11, 0x00, 0x13, 0x02, 0x0a, 0x98, 0x80, 0x93, 0x82, 0x06, 0xc1, 0x84, 0xca, 0x2d, 0xf5, 0x05, 0x01, 0xa9:
2020-07-30 23:08:49.035 Detail,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2020-07-30 23:08:49.035 Detail,   ZW_SEND_DATA delivered to Z-Wave stack
2020-07-30 23:08:49.055 Detail,   Received: 0x01, 0x18, 0x00, 0x13, 0x01, 0x00, 0x00, 0x02, 0x00, 0xb3, 0x7f, 0x7f, 0x7f, 0x7f, 0x00, 0x00, 0x03, 0x00, 0x00, 0x00, 0x00, 0x03, 0x01, 0x00, 0x00, 0x45
2020-07-30 23:08:49.055 Detail,   ZW_SEND_DATA Request with callback ID 0x01 received (expected 0x01)
2020-07-30 23:08:49.268 Detail, Node002,   Received: 0x01, 0x2a, 0x00, 0x04, 0x00, 0x02, 0x22, 0x98, 0x81, 0x00, 0x82, 0x0f, 0xe0, 0x51, 0x87, 0x8c, 0x00, 0x05, 0x6d, 0xf7, 0x52, 0xa3, 0x80, 0x3a, 0x55, 0x99, 0xb4, 0xc8, 0x56, 0x0f, 0x89, 0xd1, 0x93, 0xc5, 0xee, 0x83, 0xa7, 0x10, 0x1e, 0xe3, 0x0a, 0xb5, 0x00, 0x74
2020-07-30 23:08:49.269 Info, Raw: 0x98, 0x81, 0x00, 0x82, 0x0f, 0xe0, 0x51, 0x87, 0x8c, 0x00, 0x05, 0x6d, 0xf7, 0x52, 0xa3, 0x80, 0x3a, 0x55, 0x99, 0xb4, 0xc8, 0x56, 0x0f, 0x89, 0xd1, 0x93, 0xc5, 0xee, 0x83, 0xa7, 0x10, 0x1e, 0xe3, 0x0a, 0xb5
2020-07-30 23:08:49.269 Detail, Node002, Decrypted Packet: 0x00, 0x60, 0x0d, 0x01, 0x01, 0x32, 0x02, 0x21, 0x34, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
2020-07-30 23:08:49.269 Detail,
2020-07-30 23:08:49.269 Error, Node002, Cannot find endpoint map to instance for Command Class COMMAND_CLASS_METER endpoint 1
2020-07-30 23:08:49.831 Detail, Node002,   Received: 0x01, 0x0a, 0x00, 0x04, 0x00, 0x02, 0x02, 0x98, 0x40, 0xb5, 0x00, 0x9c
2020-07-30 23:08:49.831 Info, Node002, Received SecurityCmd_NonceGet from node 2
2020-07-30 23:08:49.832 Info, NONCES: 0x17, 0x67, 0x14, 0x5a, 0x31, 0xbf, 0x21, 0xee
2020-07-30 23:08:49.832 Info, NONCES: 0x07, 0x73, 0xd4, 0x7d, 0x1e, 0x7a, 0x4d, 0x85
2020-07-30 23:08:49.832 Info, NONCES: 0x2a, 0xc1, 0x33, 0x32, 0xd3, 0xbf, 0x53, 0x95
2020-07-30 23:08:49.832 Info, NONCES: 0x24, 0x4e, 0x4f, 0x6a, 0x84, 0x7b, 0x70, 0xb0
2020-07-30 23:08:49.832 Info, NONCES: 0xfd, 0xf7, 0xd7, 0x37, 0x89, 0xd4, 0x55, 0xab
2020-07-30 23:08:49.832 Info, NONCES: 0x93, 0x82, 0x06, 0xc1, 0x84, 0xca, 0x2d, 0xf5
2020-07-30 23:08:49.832 Info, NONCES: 0xe6, 0x34, 0xf2, 0x65, 0x73, 0x19, 0xd1, 0xc4
2020-07-30 23:08:49.832 Info, NONCES: 0xd4, 0x24, 0x47, 0x4c, 0x66, 0x6f, 0x25, 0x9b
2020-07-30 23:08:49.832 Info, Node002, Sending (Send) message (Callback ID=0x01, Expected Reply=0x00) - Nonce_Report - 0x01, 0x11, 0x00, 0x13, 0x02, 0x0a, 0x98, 0x80, 0xe6, 0x34, 0xf2, 0x65, 0x73, 0x19, 0xd1, 0xc4, 0x05, 0x01, 0xd3:
2020-07-30 23:08:49.840 Detail,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2020-07-30 23:08:49.840 Detail,   ZW_SEND_DATA delivered to Z-Wave stack
2020-07-30 23:08:49.860 Detail,   Received: 0x01, 0x18, 0x00, 0x13, 0x01, 0x00, 0x00, 0x02, 0x00, 0xb5, 0x7f, 0x7f, 0x7f, 0x7f, 0x00, 0x00, 0x03, 0x00, 0x00, 0x00, 0x00, 0x03, 0x01, 0x00, 0x00, 0x43
2020-07-30 23:08:49.860 Detail,   ZW_SEND_DATA Request with callback ID 0x01 received (expected 0x01)
2020-07-30 23:08:49.888 Detail, Node002,   Received: 0x01, 0x2a, 0x00, 0x04, 0x00, 0x02, 0x22, 0x98, 0x81, 0x6b, 0xe2, 0x93, 0x63, 0x5d, 0x1e, 0x35, 0x12, 0x63, 0x06, 0xaa, 0xeb, 0xec, 0xb7, 0x3d, 0xa2, 0x3d, 0xe6, 0x29, 0x2f, 0xc3, 0x7e, 0xc8, 0xe6, 0x6b, 0x9a, 0xac, 0x1b, 0xf9, 0x24, 0x1e, 0x17, 0xb5, 0x00, 0x7c
2020-07-30 23:08:49.888 Info, Raw: 0x98, 0x81, 0x6b, 0xe2, 0x93, 0x63, 0x5d, 0x1e, 0x35, 0x12, 0x63, 0x06, 0xaa, 0xeb, 0xec, 0xb7, 0x3d, 0xa2, 0x3d, 0xe6, 0x29, 0x2f, 0xc3, 0x7e, 0xc8, 0xe6, 0x6b, 0x9a, 0xac, 0x1b, 0xf9, 0x24, 0x1e, 0x17, 0xb5
2020-07-30 23:08:49.888 Detail, Node002, Decrypted Packet: 0x00, 0x60, 0x0d, 0x02, 0x01, 0x32, 0x02, 0x21, 0x34, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
2020-07-30 23:08:49.888 Detail,
2020-07-30 23:08:49.888 Error, Node002, Cannot find endpoint map to instance for Command Class COMMAND_CLASS_METER endpoint 2
```

flechaig, 2021-02-04:
I solved the issue by refreshing the node.