OpenZWave/open-zwave #1037: OpenZWave 1.4.340 crash every time after adding security devices - RealLivingCapTouch

sleep888, 2016-11-10:
I got a assa abloy door lock, I used u-stick gen5  as control and Domoticz as UI running on windows 10. I think the pairing process was very smooth and it could pair up successfully and got message "Locked via Lock Button", then crash right after. I have spent quite a long time trying to debug this issue, but did not find root cause. Any idea what may cause this?

The last two line of "OZW_Log.txt"
2016-11-09 22:16:10.073 Info, Node002, Recieved a DoorLockLogging Record 195 which is "Locked via Lock Button"
2016-11-09 22:16:10.073 Detail, Node002, Initial read of value

The full log of "OZW_Log.txt":
Logging started 2016-11-09 22:14:02.561 

2016-11-09 22:14:02.568 Always, OpenZwave Version 1.4.340 Starting Up
2016-11-09 22:14:02.591 Info, Setting Up Provided Network Key for Secure Communications
2016-11-09 22:14:02.596 Info, mgr,     Added driver for controller COM3
2016-11-09 22:14:02.602 Info,   Opening controller COM3
2016-11-09 22:14:02.607 Info,     Trying to open serial port COM3 (Attempt 1)
2016-11-09 22:14:02.617 Info,     Serial port COM3 opened (attempt 1)
2016-11-09 22:14:02.622 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2016-11-09 22:14:02.628 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
2016-11-09 22:14:02.659 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES: 0x01, 0x03, 0x00, 0x05, 0xf9
2016-11-09 22:14:02.663 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_GET_CAPABILITIES: 0x01, 0x03, 0x00, 0x07, 0xfb
2016-11-09 22:14:02.667 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_SUC_NODE_ID: 0x01, 0x03, 0x00, 0x56, 0xaa
2016-11-09 22:14:02.671 Detail, 
2016-11-09 22:14:02.675 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x15) - FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2016-11-09 22:14:02.682 Detail, contrlr,   Received: 0x01, 0x10, 0x01, 0x15, 0x5a, 0x2d, 0x57, 0x61, 0x76, 0x65, 0x20, 0x33, 0x2e, 0x39, 0x35, 0x00, 0x01, 0x99
2016-11-09 22:14:02.686 Detail, 
2016-11-09 22:14:02.690 Info, contrlr, Received reply to FUNC_ID_ZW_GET_VERSION:
2016-11-09 22:14:02.694 Info, contrlr,     Static Controller library, version Z-Wave 3.95
2016-11-09 22:14:02.698 Detail, Node045,   Expected reply was received
2016-11-09 22:14:02.702 Detail, Node045,   Message transaction complete
2016-11-09 22:14:02.706 Detail, 
2016-11-09 22:14:02.710 Detail, contrlr, Removing current message
2016-11-09 22:14:02.714 Detail, 
2016-11-09 22:14:02.718 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x20) - FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
2016-11-09 22:14:02.724 Detail, contrlr,   Received: 0x01, 0x08, 0x01, 0x20, 0xcd, 0x64, 0x71, 0x9a, 0x01, 0x95
2016-11-09 22:14:02.729 Detail, 
2016-11-09 22:14:02.733 Info, contrlr, Received reply to FUNC_ID_ZW_MEMORY_GET_ID. Home ID = 0xcd64719a.  Our node ID = 1
2016-11-09 22:14:02.739 Detail, Node100,   Expected reply was received
2016-11-09 22:14:02.744 Detail, Node100,   Message transaction complete
2016-11-09 22:14:02.748 Detail, 
2016-11-09 22:14:02.753 Detail, contrlr, Removing current message
2016-11-09 22:14:02.758 Detail, 
2016-11-09 22:14:02.762 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x05) - FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES: 0x01, 0x03, 0x00, 0x05, 0xf9
2016-11-09 22:14:02.767 Detail, contrlr,   Received: 0x01, 0x04, 0x01, 0x05, 0x28, 0xd7
2016-11-09 22:14:02.771 Detail, 
2016-11-09 22:14:02.775 Info, contrlr, Received reply to FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES:
2016-11-09 22:14:02.780 Info, contrlr,     There is no SUC ID Server (SIS) in this network.
2016-11-09 22:14:02.784 Info, contrlr,     The PC controller is a primary controller.
2016-11-09 22:14:02.788 Detail, Node215,   Expected reply was received
2016-11-09 22:14:02.792 Detail, Node215,   Message transaction complete
2016-11-09 22:14:02.796 Detail, 
2016-11-09 22:14:02.801 Detail, contrlr, Removing current message
2016-11-09 22:14:02.805 Detail, 
2016-11-09 22:14:02.810 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x07) - FUNC_ID_SERIAL_API_GET_CAPABILITIES: 0x01, 0x03, 0x00, 0x07, 0xfb
2016-11-09 22:14:02.818 Detail, contrlr,   Received: 0x01, 0x2b, 0x01, 0x07, 0x01, 0x00, 0x00, 0x86, 0x01, 0x01, 0x00, 0x5a, 0xfe, 0x81, 0xff, 0x88, 0x4f, 0x1f, 0x00, 0x00, 0xfb, 0x9f, 0x7d, 0xa0, 0x67, 0x00, 0x00, 0x80, 0x00, 0x80, 0x86, 0x00, 0x00, 0x00, 0xe8, 0x73, 0x00, 0x00, 0x0e, 0x00, 0x00, 0x60, 0x00, 0x00, 0xfa
2016-11-09 22:14:02.822 Detail, 
2016-11-09 22:14:02.827 Info, contrlr,  Received reply to FUNC_ID_SERIAL_API_GET_CAPABILITIES
2016-11-09 22:14:02.831 Info, contrlr,     Serial API Version:   1.0
2016-11-09 22:14:02.835 Info, contrlr,     Manufacturer ID:      0x0086
2016-11-09 22:14:02.840 Info, contrlr,     Product Type:         0x0101
2016-11-09 22:14:02.844 Info, contrlr,     Product ID:           0x005a
2016-11-09 22:14:02.849 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_RANDOM: 0x01, 0x04, 0x00, 0x1c, 0x20, 0xc7
2016-11-09 22:14:02.853 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_GET_INIT_DATA: 0x01, 0x03, 0x00, 0x02, 0xfe
2016-11-09 22:14:02.858 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_SET_TIMEOUTS: 0x01, 0x05, 0x00, 0x06, 0x64, 0x0f, 0x97
2016-11-09 22:14:02.862 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_APPL_NODE_INFORMATION: 0x01, 0x07, 0x00, 0x03, 0x01, 0x02, 0x01, 0x00, 0xf9
2016-11-09 22:14:02.866 Detail,   Expected reply was received
2016-11-09 22:14:02.871 Detail,   Message transaction complete
2016-11-09 22:14:02.875 Detail, 
2016-11-09 22:14:02.880 Detail, contrlr, Removing current message
2016-11-09 22:14:02.884 Detail, 
2016-11-09 22:14:02.889 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x56) - FUNC_ID_ZW_GET_SUC_NODE_ID: 0x01, 0x03, 0x00, 0x56, 0xaa
2016-11-09 22:14:02.895 Detail, contrlr,   Received: 0x01, 0x04, 0x01, 0x56, 0x00, 0xac
[... 1341 lines cut ...]
2016-11-09 22:16:09.935 Info, Node002, Request RTT 100 Average Request RTT 96
2016-11-09 22:16:09.935 Detail,   Expected callbackId was received
2016-11-09 22:16:09.935 Detail, Node002,   Received: 0x01, 0x08, 0x00, 0x04, 0x00, 0x02, 0x02, 0x98, 0x40, 0x2b
2016-11-09 22:16:09.951 Info, Node002, Received SecurityCmd_NonceGet from node 2
2016-11-09 22:16:09.951 Info, NONCES: 0x22, 0x11, 0x74, 0x12, 0x8c, 0xd1, 0xa2, 0x5b
2016-11-09 22:16:09.951 Info, NONCES: 0x51, 0x64, 0x92, 0x33, 0xd9, 0xf7, 0xa9, 0xea
2016-11-09 22:16:09.967 Info, NONCES: 0xfa, 0x81, 0x73, 0x41, 0x8e, 0x6f, 0x7a, 0xed
2016-11-09 22:16:09.972 Info, NONCES: 0xa2, 0x2c, 0x73, 0x37, 0x93, 0x68, 0xb7, 0x77
2016-11-09 22:16:09.973 Info, NONCES: 0x36, 0x2b, 0xeb, 0xde, 0xf4, 0x06, 0x32, 0x4f
2016-11-09 22:16:09.973 Info, NONCES: 0x0a, 0xf1, 0xd0, 0x0c, 0x36, 0xca, 0xb1, 0xba
2016-11-09 22:16:09.973 Info, NONCES: 0xcc, 0xef, 0x30, 0x69, 0x17, 0x76, 0x94, 0x6e
2016-11-09 22:16:09.988 Info, NONCES: 0x36, 0x34, 0xf5, 0x0e, 0x4d, 0xe7, 0x06, 0x3a
2016-11-09 22:16:09.988 Info, Node002, Sending (Send) message (Callback ID=0x01, Expected Reply=0x04) - Nonce_Report - 0x01, 0x11, 0x00, 0x13, 0x02, 0x0a, 0x98, 0x80, 0x0a, 0xf1, 0xd0, 0x0c, 0x36, 0xca, 0xb1, 0xba, 0x05, 0x01, 0x39:
2016-11-09 22:16:10.004 Detail, Node002,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2016-11-09 22:16:10.004 Detail, Node002,   ZW_SEND_DATA delivered to Z-Wave stack
2016-11-09 22:16:10.020 Detail, Node002,   Received: 0x01, 0x07, 0x00, 0x13, 0x01, 0x00, 0x00, 0x02, 0xe8
2016-11-09 22:16:10.020 Detail, Node002,   ZW_SEND_DATA Request with callback ID 0x01 received (expected 0x01)
2016-11-09 22:16:10.035 Info, Node002, Request RTT 200 Average Request RTT 148
2016-11-09 22:16:10.051 Detail, Node002,   Received: 0x01, 0x28, 0x00, 0x04, 0x00, 0x02, 0x22, 0x98, 0x81, 0xf9, 0x5c, 0x51, 0xdb, 0xb6, 0xa8, 0x76, 0x7a, 0x7c, 0x89, 0x38, 0xc6, 0x28, 0x37, 0x08, 0xf1, 0x03, 0x9c, 0xeb, 0xb4, 0x03, 0xf6, 0x8a, 0x0a, 0x87, 0xad, 0x89, 0x7a, 0x03, 0x90, 0x93, 0x17, 0x41
2016-11-09 22:16:10.051 Info, Raw: 0x98, 0x81, 0xf9, 0x5c, 0x51, 0xdb, 0xb6, 0xa8, 0x76, 0x7a, 0x7c, 0x89, 0x38, 0xc6, 0x28, 0x37, 0x08, 0xf1, 0x03, 0x9c, 0xeb, 0xb4, 0x03, 0xf6, 0x8a, 0x0a, 0x87, 0xad, 0x89, 0x7a, 0x03, 0x90, 0x93, 0x17, 0x41
2016-11-09 22:16:10.051 Detail, Node002, Decrypted Packet: 0x00, 0x4c, 0x04, 0xc3, 0x07, 0xd0, 0x01, 0x01, 0x20, 0x00, 0x00, 0x03, 0x00, 0x71, 0x70
2016-11-09 22:16:10.069 Detail, 
2016-11-09 22:16:10.073 Info, Node002, Response RTT 238 Average Response RTT 274
2016-11-09 22:16:10.073 Info, Node002, Recieved a DoorLockLogging Record 195 which is "Locked via Lock Button"
2016-11-09 22:16:10.073 Detail, Node002, Initial read of value

sleep888, 2016-11-16:
it seems to be a windows platform only issue. I tried on RasberryPi, it worked great.

Fishwaldo, 2016-11-16:
Please try the latest snapshots from http://old.openzwave.com/snapshots/ to see if the problem is still present. (if it is, please reopen this case)

closed 2016-11-16 by Fishwaldo