OpenZWave/open-zwave #1038: BUS error in Driver::WriteNextMsg

rpentrala, 2016-11-11:
Pairing third device triggered crash and then repeatedly crashes. 

core dump:
#0  0x65646f4e in ?? ()
#1  0x7784c440 in OpenZWave::Driver::WriteNextMsg(OpenZWave::Driver::MsgQueue) () from /usr/local/lib/libopenzwave.so.1.1
#2  0x7784cb44 in OpenZWave::Driver::DriverThreadProc(OpenZWave::Event*) () from /usr/local/lib/libopenzwave.so.1.1
#3  0x77837184 in OpenZWave::ThreadImpl::Run() () from /usr/local/lib/libopenzwave.so.1.1
#4  0x778371c8 in OpenZWave::ThreadImpl::ThreadProc(void*) () from /usr/local/lib/libopenzwave.so.1.1
#5  0x75df3768 in start_thread () from /lib/libpthread.so.0
#6  0x75dec060 in __thread_start () from /lib/libpthread.so.0
Backtrace stopped: frame did not save the PC
(gdb) bt
#0  0x65646f4e in ?? ()
#1  0x7784c440 in OpenZWave::Driver::WriteNextMsg(OpenZWave::Driver::MsgQueue) () from /usr/local/lib/libopenzwave.so.1.1
#2  0x7784cb44 in OpenZWave::Driver::DriverThreadProc(OpenZWave::Event*) () from /usr/local/lib/libopenzwave.so.1.1
#3  0x77837184 in OpenZWave::ThreadImpl::Run() () from /usr/local/lib/libopenzwave.so.1.1
#4  0x778371c8 in OpenZWave::ThreadImpl::ThreadProc(void*) () from /usr/local/lib/libopenzwave.so.1.1
#5  0x75df3768 in start_thread () from /lib/libpthread.so.0
#6  0x75dec060 in __thread_start () from /lib/libpthread.so.0
[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/586341/OZW_Log.txt)
[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/586343/OZW_Log.txt)

Fishwaldo, 2016-11-12:
BUS errors are usually async. So you can not rely on a Backtrace. What program are you using. Can you reproduce it on OZWCP or MinOZW?

rpentrala, 2016-11-15:
No, i could not reproduce .

Fishwaldo, 2016-11-15:
Then I would say the root cause may lie with the application and not OZW.

closed 2016-11-15 by Fishwaldo