OpenZWave/open-zwave #1048: AEOTEC MultiSensor 6 -

rjblake, 2016-11-24:
After sorting the SmartSwitch 6 problem, I've now got time to sort the Multi Sensor 6 problem. Device are being added correctly and reported as requested, but the following logic error occurs:

1. Motion Sensor - triggers both motion and vibration
2. Vibration Sensor - triggers only vibration sensor. Does not reset after given time (240secs)
3. Vibration Sensor, followed by Motion Sensor - works correctly and both reset after

According to AEOTEC Support:

_I believe the burglar node is a references to the notification command class which in the end does not care what notification comes in.

Motion Sensor

When motion sensor is triggered, it sends a Basic Report (configurable) and a Notification report (to advertise that motion has been triggered as specified per the notification command class) which triggers the burglar node. The index sent in the notification command class is different in a separate case, but are still seen as a notification report by the software.

Both are disabled when a No More Motion has been triggered.

----

Vibration Sensor

This sends out a Notification Report when triggered, but does not have a timeout. Its only job is to send out a notification report to the gateway. Sends out a notification trigger (Notification type=07, Event=03).

This is normal activity for our Multisensors in this case depending on the software or gateway application of the notification command class.

The vibration is meant to be a quick notification which is why it does not have a timeout."_

I believe there is also an error in the command being sent to reset the sensor in (2.) above, as my Domoticz log shows the following at the time:

2016-11-23 20:50:03.212 Error, ERROR: ZW_SEND_DATA could not be delivered to Z-Wave stack
2016-11-23 20:50:03.212 Always, 
2016-11-23 20:50:03.212 Always, Dumping queued log messages
2016-11-23 20:50:03.212 Always, 
2016-11-23 20:50:03.212 Always, 
2016-11-23 20:50:03.212 Always, End of queued log message dump
2016-11-23 20:50:03.213 Always,

nechry, 2016-11-25:
I also have this device in Application Version 1.6 and the Motion sensor did not trigger the Vibration sensor.

First the vibration sensor is more a tamper sensor and is "normal" they don't reset after given time. A auto-reset is a job for your gateway in my point of view.

I also set param 4 to "Binary Sensor Report".

rjblake, 2016-11-25:
@nechry - interesting. I tested out using another gateway and problem does not exist. However, the errors in the log are coming from OpenZwave and not Domoticz, so the "Error, ERROR: ZW_SEND_DATA could not be delivered to Z-Wave stack" seems to be OpenZwave related. Or am I mistaken?

nechry, 2016-11-25:
Yes is a reported error from Openzwave library, but the meaning is the device did not accept your message. Some CC accepts Only Get and Refresh but not a Set message. It's what happens here, I'm suppose

nechry, 2016-11-25:
Please use malling list instead of a issue.
https://groups.google.com/forum/m/#!forum/openzwave

closed 2016-11-25 by nechry