OpenZWave/open-zwave #1049: PH-PAB01.eu and compatibility

Sebol0187, 2016-11-25:
Hello,
Sorry for my English.
I have a problem after installing Zipato PAB01. It is well recognized by Domoticz, but it doesn't return any value (cf OZC screen). Notably the cumulative energy.
I have the latest beta version of domoticz (3.5985), and openzwave (1.4-2247-g5046b13-dirty).
What do you think ?
Thank you !!
Olivier
[image]

nechry, 2016-11-25:
is suppose to do periodic report every 720sec (default value) check parameters

Sebol0187, 2016-11-25:
But I have nothing in Domoticz :/
[image]

nechry, 2016-11-25:
Ensure associations groups are ok

Sebol0187, 2016-11-26:
Group 1 : 1.1
Group 2 : 1.1
Group 3 : 1.1
Group 4 : -
Group 5 : -

Sebol0187, 2016-12-18:
Up ? ;)

Fishwaldo, 2016-12-19:
what about group 4 and 5? consult the manual and check what they are for.

Sebol0187, 2016-12-19:
Just groups 1 to 3 :(
[Manuel_Zipato_PAB01.pdf](https://github.com/OpenZWave/open-zwave/files/662329/Manuel_Zipato_PAB01.pdf)

nechry, 2016-12-20:
why you talk about 4 and 5?

nechry, 2016-12-20:
I try my PAB01 and I got energy report. 
[image]

the default reporting time is 720 sec.

Sebol0187, 2016-12-20:
Hello,
I don't have energy report with Domoticz : 
[image]
[image]
[image]

Fishwaldo, 2018-10-26:
Closing. Old issue

closed 2018-10-26 by Fishwaldo

Sebol0187, 2018-10-26:
Bonjour

Pourquoi clore une issue alors que le problème n'est toujours pas réglé à
ce jour et que nous sommes plusieurs dans le pétrin. Cela gêne ??

Cordialement,

Olivier THIRION