OpenZWave/open-zwave #1055: Qubino 0-10v ZMNHVD1

sumo-89, 2016-11-29:
Hi!
I think this is broken for the analogue input. From the manual:
**Endpoint 1:**
(0-10V output)
...

**Endpoint 2:**
Device Class:
ZWAVEPLUS_INFO_REPORT_ROLE_TYPE_SLAVE_ALWAYS_0N
GENERIC_TYPE_SENSOR_MULTILEVEL
SPECIFIC_TYPE_ROUTING_SENSOR_MULTILEVEL
Command Classes:
COMMAND_CLASS_ZWAVEPLUS_INFO_V2
COMMAND_CLASS_VERSION_V2
COMMAND_CLASS_ASSOCIATION_V2
COMMAND_CLASS_MULTI_CHANNEL_ASSOCIATION_V3
COMMAND_CLASS_ASSOCIATION_GRP_INFO_V2
COMMAND_CLASS_SENSOR_MULTILEVEL_V7

**Endpoint 3:**
Device Class:
ZWAVEPLUS_INFO_REPORT_ROLE_TYPE_SLAVE_ALWAYS_0N
GENERIC_TYPE_SENSOR_MULTILEVEL
SPECIFIC_TYPE_ROUTING_SENSOR_MULTILEVEL
Command Classes:
COMMAND_CLASS_ZWAVEPLUS_INFO_V2
COMMAND_CLASS_VERSION_V2
COMMAND_CLASS_ASSOCIATION_V2
COMMAND_CLASS_MULTI_CHANNEL_ASSOCIATION_V3
COMMAND_CLASS_ASSOCIATION_GRP_INFO_V2
COMMAND_CLASS_SENSOR_MULTILEVEL_V7

NOTE: The above list is valid for the product with a
temperature sensor connected to TS terminal. In case the
sensor is not connected then following command class
isn’t supported:
COMMAND_CLASS_SENSOR_MULTILEVEL_V7

I don't have a digital temp sensor on the TS terminal, only analogue PT100 -> 0-10V converter -> analogue input. I tried Domoticz now, but shouldn't I get two devices - one "Level" and one "Temp"?
I get "Level" and that works just fine! :)
But looking in the XML I only see two endpoints, is one missing?
And also - the note in the manual, does that mean that COMMAND_CLASS_SENSOR_MULTILEVEL_V7 is not supported on any endpoint if no TS temp sensor is connected?

Thanks! :)

Fishwaldo, 2016-12-19:
did you re-associate the device after connecting the analogue sensors? I understand you need to make sure the sensors are active when associating to your network.

sumo-89, 2016-12-19:
I did!
I emailed Qubino also, this was the answer I got:
"To read data from input I1, there must be configured multichannel
association lifeline, but there is a problem, domoticz does not support
multichannel association."

Don't know if it's a limitation to Domoticz or OpenZWave, it's out of my
league... :)

nechry, 2016-12-19:
you can simply add 

```<CommandClass id="142" ForceInstances="true"/>```

to the ZMNHVDx xml file, remove the zwcfg and check if fine. If yes I will add to the config file in ozw

sumo-89, 2016-12-19:
Thank you! :)
I've added the config and restarted Domoticz. Guess I need to
exclude/include again? Will try it later this week, since it controls my
chimney fan it's really not that easy to reach. :)

nechry, 2016-12-19:
If your remove your zwcfgxxx.xml file you don't have to exclude and include your device

sumo-89, 2016-12-19:
Domoticz recreated that file at startup... but, everything else just looks
the same to me. ;)
Should Domoticz automatically create a new Device for it?

From zwcfg now:
            <CommandClass id="142"
name="COMMAND_CLASS_MULTI_INSTANCE_ASSOCIATION" version="1"
request_flags="5" innif="true" ForceInstances="true">
                <Instance index="1" />
                <Instance index="2" endpoint="1" />
                <Instance index="3" endpoint="2" />
                <Associations num_groups="6">
                    <Group index="1" max_associations="1" label="Lifeline"
auto="true" multiInstance="true">
                        <Node id="1" />
                    </Group>
                    <Group index="2" max_associations="16" label="Basic
on/off of the input I1 state" auto="false" multiInstance="true" />
                    <Group index="3" max_associations="16" label="Level
change I1" auto="false" multiInstance="true" />
                    <Group index="4" max_associations="16"
label="Multilevel set of the Flush Dimmer" auto="false"
multiInstance="true" />
                    <Group index="5" max_associations="16"
label="Multilevel sensor report analogue sensor" auto="false"
multiInstance="true" />
                    <Group index="6" max_associations="16"
label="Multilevel sensor report temperature sensor" auto="false" />
                </Associations>
            </CommandClass>

Fishwaldo, 2017-05-30:
Ping? Is this fixed?

sumo-89, 2017-06-01:
Hi!
Sorry, I haven't have had time to take a look at it yet. I'll try to get it
going this or next weekend... :)

sumo-89, 2017-06-04:
So, I've included and excluded it again. zwcfg says:

            <CommandClass id="142" name="COMMAND_CLASS_MULTI_CHANNEL_ASSOCIATION" version="1" request_flags="5" innif="true" ForceInstances="true">
                <Instance index="1" />
                <Instance index="2" endpoint="1" />
                <Instance index="3" endpoint="2" />
                <Associations num_groups="6">
                    <Group index="1" max_associations="1" label="Lifeline" auto="true" multiInstance="true">
                        <Node id="1" instance="1" />
                    </Group>
                    <Group index="2" max_associations="16" label="Basic on/off of the input I1 state" auto="false" multiInstance="true" />
                    <Group index="3" max_associations="16" label="Level change I1" auto="false" multiInstance="true" />
                    <Group index="4" max_associations="16" label="Multilevel set of the Flush Dimmer" auto="false" multiInstance="true" />
                    <Group index="5" max_associations="16" label="Multilevel sensor report analogue sensor" auto="false" multiInstance="true" />
                    <Group index="6" max_associations="16" label="Multilevel sensor report temperature sensor" auto="false" />
                </Associations>
            </CommandClass>

But, still I don't get a device for it. I'm not sure how I'm supposed to get it in to Domoticz?

sumo-89, 2017-06-28:
I switched to Home Assistant and with the changes you proposed I can now
say that this works as expected! Thank you very much! :)

2017-06-01 21:16 GMT+02:00 Johan Linder <lindroz@gmail.com>:

closed 2018-10-26 by Fishwaldo