OpenZWave/open-zwave #1057: OTA updates

antares2001, 2016-12-01:
Hello,

AFAIK the ZWave Plus protocol supports OTA firmware updates. It seems that OpenZWave is not yet capable of this. Are there any plans to add support for OTA updates to OpenZWave?

Thanks and cheers!

Fishwaldo, 2016-12-19:
some vendors that support us with hardware have been reluctant to provide OTA firmware files for us to test with.... and I don't know of other vendors that publicly release their OTA firmware so I see no point at the moment.

closed 2016-12-19 by Fishwaldo

pbootsma, 2017-01-18:
Aeon Labs publishes their firmwares for the Aeotec products : [https://aeotec.freshdesk.com/support/home](url)

jvolkman, 2017-03-05:
HomeSeer publishes firmware for their line of switches and dimmers (rebranded DragonTech). http://www.homeseer.com/current-downloads.html

RoyKas, 2017-03-10:
Z-Uno supports firmware update OTA:
[https://z-uno.z-wave.me/z-wave/ota/](https://z-uno.z-wave.me/z-wave/ota/)

michapr, 2017-04-12:
will it be reopened? Or should be created a new issue?
For Z-Uno (as sample) it is very helpful to have this feature for updating sketches and firmware.
Z-Way support it (as sample) - yes, of course,... same vendor ;)

SuneRadich, 2017-04-30:
For what its worth, Logicsoft releases ota firmware for their stuff too

http://logichome.dk/index.php/10-support-categories/6-zhc5010-firmware

nechry, 2017-04-30:
HomeSeer will release soon a OTA tool
[image]

robbiet480, 2017-05-05:
FYI while their firmware is available for download, Homeseer has stated that their firmware is in an encrypted format which can only be written via HS3 or their new Z-Flash tool (which is a paid product btw).

[Source](https://forums.homeseer.com/showpost.php?p=1302933&postcount=8).

partytimeexcellent, 2017-12-17:
I'm regretting buying the Homeseer switches. I can't believe they want me to spend $45 to update the firmware after spending $500 on switches and their Z-stick. What a racket.

JudgeDreddKLC, 2017-12-17:
Looks like the firmware is free, but the software to do it costs money? I wonder if the files are compatible with other OTA devices

https://homeseer.com/current-downloads/

  *   Available as a kit with SmartStick+ or UZB stick.
  *   FREE with HS3PRO or any HomeTroller-SEL PRO or HomeTroller S6 PRO

https://shop.homeseer.com/products/homeseer-z-flash-z-wave-ota-firmware-update-software