OpenZWave/open-zwave #1062: Aeotec by Aeon Labs ZW095 Home Energy Meter not included as secured

oliverlj, 2016-12-07:
Hi contributors of great ozw project,

I recently bought an ZW095 Home Energy Meter with 1 amper clamp from aeotec device. If i included this one as secured, it works, dont't show me an unlock lock. If I reboot, the device is shown as insecured and the network become laggy. I included the log after the reboot. For the moment, I've included again the device as insecured and all seems good.

[openzwaved.zip](https://github.com/OpenZWave/open-zwave/files/637773/openzwaved.zip)

nechry, 2016-12-08:
use from Actions TAB the "Refresh node Informations" it should be ok after

Fishwaldo, 2016-12-19:
Fixed?

oliverlj, 2016-12-19:
Not tested yet. I'll try in the week

Fishwaldo, 2017-03-21:
Closing due to no feedback.

closed 2017-03-21 by Fishwaldo