OpenZWave/open-zwave #1063: Remotec ZXT-120 unable to set AC Cool Temperature < 26 Degrees

philhawthorne, 2016-12-08:
I have a Remotec ZXT-120 IR AC controller running with [Home Assistant](http://home-assistant.io). The device shows up fine in Home Assistant, and I can turn the heater on.

However, when it comes to turning the AC on, I am unable to set a temperature below 26 degrees C. If I set a temperature to 22 degrees, OZW will immediately bump the temperature back up to 26. Here's the OZW log that occurs.

```

2016-12-08 03:43:28.971 Info, Node038, Value::Set - COMMAND_CLASS_THERMOSTAT_SETPOINT - Cooling 1 - 2 - 1 - 22
2016-12-08 03:43:28.972 Detail, Node038, Queuing (Send) ThermostatSetpointCmd_Set (Node=38): 0x01, 0x0c, 0x00, 0x13, 0x26, 0x05, 0x43, 0x01, 0x02, 0x01, 0x16, 0x25, 0xe1, 0x50
2016-12-08 03:43:28.972 Info, Node038, ThermostatSetpointCmd_Get Not Supported on this node
2016-12-08 03:43:28.972 Detail, 
2016-12-08 03:43:28.972 Info, Node038, Sending (Send) message (Callback ID=0xe1, Expected Reply=0x13) - ThermostatSetpointCmd_Set (Node=38): 0x01, 0x0c, 0x00, 0x13, 0x26, 0x05, 0x43, 0x01, 0x02, 0x01, 0x16, 0x25, 0xe1, 0x50
2016-12-08 03:43:28.979 Detail, Node038,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2016-12-08 03:43:28.979 Detail, Node038,   ZW_SEND_DATA delivered to Z-Wave stack
2016-12-08 03:43:28.992 Detail, Node038,   Received: 0x01, 0x05, 0x00, 0x13, 0xe1, 0x00, 0x08
2016-12-08 03:43:28.992 Detail, Node038,   ZW_SEND_DATA Request with callback ID 0xe1 received (expected 0xe1)
2016-12-08 03:43:28.992 Info, Node038, Request RTT 19 Average Request RTT 24
2016-12-08 03:43:28.992 Detail,   Expected callbackId was received
2016-12-08 03:43:28.992 Detail,   Expected reply was received
2016-12-08 03:43:28.992 Detail,   Message transaction complete
2016-12-08 03:43:28.992 Detail, 
2016-12-08 03:43:28.992 Detail, Node038, Removing current message
2016-12-08 03:43:31.442 Detail, Node038,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x26, 0x03, 0x80, 0x03, 0x64, 0x30
2016-12-08 03:43:31.442 Detail, 
2016-12-08 03:43:31.442 Info, Node038, Received Battery report from node 38: level=100
2016-12-08 03:43:31.442 Detail, Node038, Refreshed Value: old value=100, new value=100, type=byte
2016-12-08 03:43:31.442 Detail, Node038, Changes to this value are not verified
2016-12-08 03:43:31.442 Detail, Node038, Notification: ValueChanged
2016-12-08 03:43:33.943 Detail, Node038,   Received: 0x01, 0x0b, 0x00, 0x04, 0x00, 0x26, 0x05, 0x31, 0x05, 0x01, 0x01, 0x1a, 0xfd
2016-12-08 03:43:33.944 Detail, 
2016-12-08 03:43:33.944 Info, Node038, Received SensorMultiLevel report from node 38, instance 1, Temperature: value=26C
2016-12-08 03:43:33.944 Detail, Node038, Refreshed Value: old value=26, new value=26, type=decimal
2016-12-08 03:43:33.944 Detail, Node038, Changes to this value are not verified
2016-12-08 03:43:33.944 Detail, Node038, Notification: ValueChanged
```

I've had this issue on 3 separate AC units from 2 manufacturer IR codes. So not sure what's going on there. I've also tried to update the value in Domoticz using OZWCP, but still same result.

If it helps, [here is](https://github.com/OpenZWave/open-zwave/files/638459/zwcfg_zxt120.zip) the device portion of my XML cache. 

Hoping there's someone that's also had this issue or knows what's up?

jotakar, 2016-12-09:
Are you using this device in a ZWAVE+ net? I can not config or change mode using Aetoec ZStick GEN5. 
I use to get a Timeout error notification, some times I can make it work, but just some times.

philhawthorne, 2016-12-09:
@jotakar to be honest I am not sure. However, I can tell you that I am using the ZXT-120 in "powered" mode, so it is always online and ready to accept commands.

If you are using the batteries, the device may go to sleep, and any config you try to send to it while sleeping won't be saved until it wakes up next.

jotakar, 2016-12-12:
My remotec is usb powered.But it look like an issue with Aeon Zstick  GEN5 or OZwave and this stick, because the device accept a command after send it about 10 times.
Thanks

jotakar, 2016-12-13:
Which Stick are you using? Yo can see it in Setup>Hardware>Setup, it must be the node 1 (controller)

philhawthorne, 2016-12-13:
Hmm not sure where Setup -> Hardware -> Setup is.

Using OZWCP, the product is listed as `Aeotec Z-Stick S2`.

I might write a Home Assistant script to call the set temperature to Cool 22 multiple times, and see if that makes the AC change modes.

jotakar, 2016-12-13:
Yes, this stick is ZWave and  Remotec works fine with it, or I thought so. I tested this stick and it worked: received several commands and so. But there must be a bug in anywhere because I have to send multiple times to set commands, like you with temperature.
Aeotec has a tool called Zensys-Tools that manages these sticks, it also works with S2. You can send commands and see if device receive them. It's a tool for windows. I tested the Remotecz with my stick GEN5 and it works fine: receive and set commands and parameters without any timeout or "error droping"

philhawthorne, 2016-12-13:
Interesting.

When I call `set_temperature` the command is received by the Remotec, and sent to the AC as the AC beeps to acknowledge the command. However, the AC isn't set to Cool (as you can physically tell its not blowing out cool air), and after a few seconds OZW reports the mode as Heat 26 Degrees.

Hopefully sending the cool setting several times to the Remotec will send the right command eventually.

daradib, 2016-12-14:
@jotakar wrote:
> Are you using this device in a ZWAVE+ net? I can not config or change mode using Aetoec ZStick GEN5. I use to get a Timeout error notification, some times I can make it work, but just some times.

I have the same problem. 10 manual tries and eventually it works. I tried battery-powered, USB-powered,  "Frequently Listening Routing Slaves" mode, and "Always Listening" mode. Tried changing the device configuration a bit (and removing it entirely). Thought that my device must be defective.

I also tried increasing `MAX_TRIES` in [`cpp/src/Defs.h`](https://github.com/OpenZWave/open-zwave/blob/master/cpp/src/Defs.h). Maybe changing `RETRY_TIMEOUT` would help if you're not using security and don't mind the delay. See also #989 and #994.

I am using the same Aeotec Gen5 controller and my other devices (Aeotec MultiSensor 6 and Aeotec Smart Switch 6) are also Z-Wave Plus. I am using a [network key](https://github.com/OpenZWave/open-zwave/wiki/Adding-Security-Devices-to-OZW) for secure inclusion with my other devices but not the ZXT-120 (included normally). Are you as well?

Here's my OZW log:

    07.911 Info, Node009, Value::Set - COMMAND_CLASS_THERMOSTAT_MODE - Mode - 0 - 1 - Heat
    07.911 Detail, Node009, Queuing (Send) ThermostatModeCmd_Set (Node=9): 0x01, 0x0a, 0x00, 0x13, 0x09, 0x03, 0x40, 0x01, 0x01, 0x25, 0x39, 0xb0
    07.911 Detail, Node009, Queuing (Send) ThermostatModeCmd_Get (Node=9): 0x01, 0x09, 0x00, 0x13, 0x09, 0x02, 0x40, 0x02, 0x25, 0x3a, 0xb3
    07.912 Info, Node009, Sending (Send) message (Callback ID=0x39, Expected Reply=0x13) - ThermostatModeCmd_Set (Node=9): 0x01, 0x0a, 0x00, 0x13, 0x09, 0x03, 0x40, 0x01, 0x01, 0x25, 0x39, 0xb0
    07.920 Detail, Node009,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
    07.921 Detail, Node009,   ZW_SEND_DATA delivered to Z-Wave stack
    09.183 Detail, Node009,   Received: 0x01, 0x07, 0x00, 0x13, 0x39, 0x00, 0x00, 0x7f, 0xad
    09.183 Detail, Node009,   ZW_SEND_DATA Request with callback ID 0x39 received (expected 0x39)
    09.183 Info, Node009, Request RTT 1272 Average Request RTT 668
    09.184 Detail, Node009, Removing current message
    09.184 Info, Node009, Sending (Send) message (Callback ID=0x3a, Expected Reply=0x04) - ThermostatModeCmd_Get (Node=9): 0x01, 0x09, 0x00, 0x13, 0x09, 0x02, 0x40, 0x02, 0x25, 0x3a, 0xb3
    09.192 Detail, Node009,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
    09.192 Detail, Node009,   ZW_SEND_DATA delivered to Z-Wave stack
    09.208 Detail, Node009,   Received: 0x01, 0x07, 0x00, 0x13, 0x3a, 0x00, 0x00, 0x02, 0xd3
    09.209 Detail, Node009,   ZW_SEND_DATA Request with callback ID 0x3a received (expected 0x3a)
    09.209 Info, Node009, Request RTT 24 Average Request RTT 346
    15.335 Detail, Node009,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x09, 0x03, 0x80, 0x03, 0x64, 0x1f
    15.335 Info, Node009, Response RTT 6150 Average Response RTT 3441
    15.335 Info, Node009, Received Battery report from node 9: level=100
    15.335 Detail, Node009, Refreshed Value: old value=100, new value=100, type=byte
    15.335 Detail, Node009, Changes to this value are not verified
    15.336 Detail, Node009, Notification: ValueChanged
    19.185 Error, Node009, ERROR: Dropping command, expected response not received after 1 attempt(s)
    19.185 Detail, Node009, Removing current message
    19.185 Detail, Node009, Notification: Notification - TimeOut

@philhawthorne Do you see any TimeOuts? If not, maybe @jotakar and I should open a separate issue.

Any better results if you enable COMMAND_CLASS_THERMOSTAT_SETPOINT? My device appears to support it, but OpenZWave disables it, so it could be firmware specific. In the device configuration, set `getsupported="true"` (or remove it entirely, probably the default I imagine):

    <CommandClass id="67" base="1" getsupported="true"/>

Lastly, what are your product types? I labeled them in [`config/manufacturer_specific.xml`](https://github.com/OpenZWave/open-zwave/blob/master/config/manufacturer_specific.xml):

    <Product type="0100" id="8377" name="ZXT-120US" config="remotec/zxt-120.xml"/>
    <Product type="0101" id="8377" name="ZXT-120EU" config="remotec/zxt-120.xml"/>
    <Product type="0102" id="8377" name="ZXT-120AU" config="remotec/zxt-120.xml"/>

Mine is the 0100 ("US"). Maybe my labels are wrong and they need separate config. Remotec never responded when I sent them an email asking for confirmation.

philhawthorne, 2016-12-14:
Hey @daradib 

Yeah I am not seeing any timeouts unfortunately. So perhaps they are separate issues. As I mentioned, each call to Z-wave results in a "beep" from my AC to acknowledge the call was received, but it just doesn't change the mode on the AC. Then straight away OpenZwave updates the node status, and its set to Heat 26 degrees.

My unit is the AU frequency IDs, which I submitted #974 for. So mine is `0102 (AU)`. 

Here's my device configuration XML

```
<CommandClass id="67" name="COMMAND_CLASS_THERMOSTAT_SETPOINT" version="1" request_flags="4" getsupported="false" innif="true" base="0">
```

Can I just update the `getsupported="true"` in the XML file and reload?

Cheers

daradib, 2016-12-14:
My problem is a bit reversed. Frequent timeouts, so these beep "acks" are missing or delayed, but when I eventually get a beep, I didn't observe problems with the modes. I had difficulty changing temperatures though. My hypothesis is that OZW is using the temperature of the sensor, not the target temperature. I think enabling COMMAND_CLASS_THERMOSTAT_SETPOINT get fixed this, but it's hard to know for sure with the timeouts.

> Can I just update the getsupported="true" in the XML file and reload?

I think so. You may need to make the change in your ozwcache*.xml file (or delete the file and wait for it to be slowly recreated, depending on how many nodes you have in your network).

philhawthorne, 2016-12-14:
> I think so. You may need to make the change in your ozwcache*.xml file (or delete the file and wait for it to be slowly recreated, depending on how many nodes you have in your network).

Hmm I was of the impression that the XML file was just a cache of the settings saved in the controller, I'll give it a go anyway, and see if that helps.

daradib, 2016-12-14:
> Hmm I was of the impression that the XML file was just a cache of the settings saved in the controller

I thought so too, but saw device configuration in it, so I got confused. Does ozwcache*.xml supercede manufacturer_specific.xml and other config when it exists?

philhawthorne, 2016-12-14:
Doesn't look like it

> This file is basically a cache of the current network configuration. It is created the first time Open-ZWave discovers your network and is subsequently used on future restarts to speed up the discovery of the network.
> Generally, users, and application developers should never modify this file, but sometimes it might be required to delete the file, so the library can refresh the network devices and configurations.

Might need to find where the `COMMAND_CLASS_THERMOSTAT_SETPOINT` is disabled in OZW, and then re-enable it manually

daradib, 2016-12-14:
> Might need to find where the COMMAND_CLASS_THERMOSTAT_SETPOINT is disabled in OZW, and then re-enable it manually

By device configuration XML, are you referring to ozwcache or [config/remotec/zxt-120.xml](https://github.com/OpenZWave/open-zwave/blob/master/config/remotec/zxt-120.xml)? I would try deleting the cache (or renaming it it) and changing the latter. May be installed to `/etc/openzwave` depending on how you installed OZW. Or did you already do that and I misunderstood?

philhawthorne, 2016-12-14:
> By device configuration XML, are you referring to ozwcache or config/remotec/zxt-120.xml?

I mean the ozwcache XML file. Yeah I agree, remove the XML cache file, and then update the zxt-120.xml file to have `getsupported="true"`

I haven't done this yet, so I'll play around with my system when I get a chance and see if that does anything. :)

daradib, 2016-12-14:
Good luck!

jotakar, 2016-12-14:
I think the problem of @philhawthorne  an our problem are related because I understand that he need to send mode command several times to get Remotec work.

> Hopefully sending the cool setting several times to the Remotec will send the right command eventually.

I think there is a problem with driver of Zstick and OZW. Zstick S2 works better then GEN5.
@philhawthorne which platform are you used? Linux, windows? Aeotec has a drive for cp210x (Linux), but it's dated on year 2013...

jotakar, 2016-12-15:
I've just discovered another thing: if I put **getsupported="false"**  for command class 64 (COMMAND_CLASS_THERMOSTAT_MODE) in zwcfg_0xxxxxx.xml file, then  **Remotec accepts and executes mode changes.** But the actual mode is not read from device .... So this is a trick not very good but the only way I've found.  Even if I refresh I get Off as mode, not the actual mode.
If I set **getsupported="true" in setpoint** I cannot send changes in setpoints!! but setpoins are read when controller (Domoticz) restart, in log I see an initial reading and their values are read.
What I can understand until now: commandSet is followed by a commandGet, if commandGet doesn't answer command is not sent. Soetimes commandGet answers and set works.
But if I put **getsupported** to false, commandSet is not followed by commandGet and the system works.
Is there any place where config Zwave files are exlained at all?
Bye

jotakar, 2016-12-15:
Again more info.
I've sent a ticket to Remotecabout supported class and they affirm that Remotec support sepoints Get commands, as thermostatemode command class, as I suspeccted.
Perhaps Remotec is too slow or doesn't admit Send followed by Get or it needs an time interval between both commands. ???

daradib, 2016-12-15:
@jotakar Thanks for the updates. I will try setting `getsupported="false"` for COMMAND_CLASS_THERMOSTAT_MODE. Might be worth adding a sleep to the RequestValue functions in [`cpp/src/command_classes/ThermostatMode.cpp`](https://github.com/OpenZWave/open-zwave/blob/master/cpp/src/command_classes/ThermostatMode.cpp) and [`cpp/src/command_classes/ThermostatMode.cpp`](https://github.com/OpenZWave/open-zwave/blob/master/cpp/src/command_classes/ThermostatMode.cpp), but that would be a terrible workaround.

jotakar, 2016-12-15:
Even Sleep(5) in ThermostatMode doesn't work always and it set a large delay in all thermostate devices in the net. The log shows an strange behaviour: first appears Command Set and after 5 seconds  appears CommansGet,  but the Set is sent after Get is in queue. And it supposes a big delay in the net.
My conclusion is that Remotec doesn't admit a commandSet and a commandGet in queue. Some times it works, some times it doesn't.
What I see and what I've made:
**Method Value::Set()** first send a **commandSet** to the queue and then a **commandGet** and this doesn't work for Remotec (And by the way, most ZWave+ device send their state to controller, why this Set/Get pair?). (See about lines 370..380 in value.cpp)
So if we remove the call to RequestValue just for this device then Remotec works fine, but we need refresh to get new values. I check GetProductName() and compare with ZXT-120EU in my case, yes I know this is ... ugly.
So in **zwcfg** I put a **RefreshTriggerValue triggered by CommandClass 128** (battery) because I've seen that this is request after send a command to Remotec. And eliminate getsupported for SETPOINTS command class.
Yes I know: this is an very, very ugly patch, but It's the only way I got device work. It would be better to add a config param to the device config file and use it instead of ProductName, or better: study about why Remotec doesn't admit Set and Get commands in queue, that's the queue management . 
All that issue is a bit crazy.

philhawthorne, 2016-12-18:
Okay quick update for @daradib suggestion.

I made the changes to the `zxt120.xml` file, and then re-linked my ZXT-120 in the living room. I also have a ZXT-120 in the bedroom, and both are experiencing this issue.

So, I now have the living room ZXT-120 linked to my Z-stick with `getsupported="true"` and the bedroom ZXT-120 with `getsupported="false"'.

The living room ZXT-120 no longer changes it setting temperature back to 26 degrees. So, if I set the mode to Cool, and the target temperature to 21, it stays on 21. However, the "current temperature" will never change from 26 degrees.

Doing the same action on the bedroom ZXT-120 results in the target temperature always being reset back to 26 degrees, regardless of how many times I try. So thank you @daradib we are making progress. 

Unfortunately I can't tell what setting the AC is being set to. It is blowing out cool air now, but I am not sure if the AC is being set to 21 degrees, or something else.

jotakar, 2016-12-18:
I've just opened another thread with the Remotec problem with Set/Get issue.

kyonz, 2016-12-20:
Having the same issue here with the ZXT-120AU, changing getsupported="true" appears to make the device not respond so no real difference noted. Running HASS on a pi3 with an Aeotec Z-Stick Gen5.

philhawthorne, 2016-12-20:
@kyonz interesting. I've also got the ZXT120AU. 

Where did you change the get supported flag? In your zxxxxcfg.xml file? It needs to be done in the OpenZwave files, then relinked to your stick.

kyonz, 2016-12-20:
Changed in the OpenZwave files and then purged the pyozw.sqlite and zwcfg files for it to be rebuilt (can see the changes reflected in the newly generated zxxxxcfg.xml file.

I think my issue is specifically along the lines of jotakar's raised [issue](https://github.com/OpenZWave/open-zwave/issues/1070). If getsupported="false" it will attempt to make the change (aircon gets adjusted) but the get fails and so the change isn't reflected. I've found that if I keep hitting it then it sometimes works..

Seeing the below in my OZW log.

```
Detail, Node013, Queuing (Send) ThermostatSetpointCmd_Set (Node=13): 0x01, 0x0c, 0x00, 0x13, 0x0d, 0x05, 0x43, 0x01, 0x02, 0x01, 0x13, 0x25, 0x2a, 0xb5
2016-12-20 07:57:33.248 Detail, Node013, Queuing (Send) ThermostatSetpointCmd_Get (Node=13): 0x01, 0x0a, 0x00, 0x13, 0x0d, 0x03, 0x43, 0x02, 0x02, 0x25, 0x2b, 0xa5
Error, Node013, ERROR: Dropping command, expected response not received after 1 attempt(s)
```

dcarrion87, 2017-09-19:
Just checking in to see if anyone has had any luck in resolving this. I seem to be having this issue as well.

jotakar, 2017-09-19:
Hi
WEll there is a solution to solve some problems with this device. It's in pull request #1090, but @fishwaldo doesn't like it, although there is another pull that perhaps fixes this problem, it's the pull request #1315

elad-bar, 2019-04-27:
@austwhite did you manage to fix it?
It seems that it creates in HA 4 devices - cooling, heating, unused 4 and 6,
cooling cannot set temp greater than 24 and heating cannot set below

austwhite, 2019-04-27:
@elad-bar:
It's temperamental on some setups, but I did get it working stable with a Virtual Machine setup.
If you want the device to report the Set Point rather than just the measured room temperature, you have to edit the zwcfg_xxxxx.xml file (when home assistant is not running) and set, if I recall correctly, in option 67 the get_setpoint or something similar that is set to FALSE and set it to TRUE, and then restart Home Assistant.  This is a kind of patch as the file is a cache file, but it works and sticks through upgrades, unless HA drops your Z-Wave adaptor, then it may recreate the cache file.
I think you can play with the manufacturer xml file to set it in OZW, but this will be over written when OZW is updated anyway.
I use the ZXT-120 on a device still, but I have also go the ZXT-600.   Unfortunately the config for the ZXT-600 also sets that setting to False, so it is the same modification needed on that too.
If someone knows how to submit a PR to OZW to update the config file, that could probably be rectified.

Note: The ZXT-120 does have timing issues in some setups and will stop responding.  I found it stable on my setup using a Virtual Machine, but if I install it on a Raspberry Pi 3, it fails to respond.
I would recommend the ZXT-600 if anyone is buying a device new.

austwhite, 2019-04-27:
on the 4 devices, you can just hide the two unused ones.
I use a LoveLace custom card called Dual thermostat that combines the heat and cool into one icon,

Lukas-19, 2019-06-24:
I have this model (remotex zxt-120) and it doesn't work properly. It has timeouts problems (I need to send the on/off command several times to get it in action, while log notifies about time out error), does anybody know a fix for this timeout issue? 
Thanks

rickdsanchez, 2019-06-25:
@Lukas-19 I have ZXT-120 and the device seems to work reliably in my network with openzwave. I did have problems with the timeouts during the initial device interview on Zwave network where it wouldn't advertise its services. I use GoControl  HUSBZB-1 USB Hub as a Z-wave controller and Home Assistant. I have less than 10 zwave devices in the network.

Lukas-19, 2019-06-25:
I have this device in networkd managed by Domoticz, with USB Aeon Stick (zwave 5 plus). Ubuntu 14.04 and the last version of OpenZwave.

Fishwaldo, 2019-06-25:
I'm so sick of this device. https://community.home-assistant.io/t/remotec-zxt-120-operational-questions-my-script/20260/41

rickdsanchez, 2019-06-25:
Correction to my earlier comment: the device I have that works well is Remotec ZXT-600, not ZXT-120.

I have ZXT-600 and the device seems to work reliably in my network with openzwave. I did have problems with the timeouts during the initial device interview on Zwave network where it wouldn't advertise its services. I use GoControl HUSBZB-1 USB Hub as a Z-wave controller and Home Assistant. I have less than 10 zwave devices in the network.

Lukas-19, 2019-06-25:
Oh, yes. Now I understand. I'll go on investigating about my ZXT-120... these devices are not cheap.

mhutch, 2019-07-16:
I'm seeing what looks exactly like this issue with a Remotec ZXT-600US

AdamWeglarz, 2020-07-03:
Hi,

can you try to apply this patch https://github.com/OpenZWave/open-zwave/issues/2286 and recompile z-wave ?
https://www.domoticz.com/forum/viewtopic.php?f=56&t=3074&start=60 -> here is explanation why this device has problems with receiveing commands.
On Domoticz it solves problems with this device.