OpenZWave/open-zwave #1065: Qubino ZMNHID1 cannot set mode

jotakar, 2016-12-10:
With the last update the thermostat ON/OFF ZMNHID1 does not accept the mode change. It remains in Off and can not change to Heat.

jotakar, 2016-12-11:
I've seen in valueList.cpp this code
```
//-----------------------------------------------------------------------------
// <ValueList::SetByValue>
// Set a new value in the device, selected by item index
//-----------------------------------------------------------------------------
bool ValueList::SetByValue
(
    int32 const _value
)
{
    // create a temporary copy of this value to be submitted to the Set() call and set its value to the function param
    ValueList* tempValue = new ValueList( *this );
    tempValue->m_valueIdx = GetItemIdxByValue(_value);

    // Set the value in the device.
    bool ret = ((Value*)tempValue)->Set();

    // clean up the temporary value
    delete tempValue;

    return ret;
}
```
I see that with
 ```
 ValueList* tempValue = new ValueList( *this );
    tempValue->m_valueIdx = GetItemIdxByValue(_value);
```
We send to the device the index of value in the List not the value of the item, because tempValue contains the index not the value. Could be this the problem?
I can't find the definition of Set for ValueList :(

jotakar, 2016-12-11:
And more:
In the file cpp/src/value_classes/ValueList.cpp  I've replaced
`tempValue->m_valueIdx = GetItemIdxByValue(_value);`
with
`tempValue->m_valueIdx = _value;`
And now I can set the Mode in Qubino and other device (Remotec ZXT120, wth the sme problem in temperature compensation)

madarwin, 2016-12-13:
I see the same issue when using my Honeywell thermostat. SetByLabel is broken as well due to this change. SetByLabel already calls GetItemIdxByValue and then it calls SetByValue. So with this change GetItemIdxByValue gets called twice.

jotakar, 2016-12-13:
No, SetByLabel doesnt call GetItemIdxByValue 
`bool ValueList::SetByLabel
(
    string const& _label
)
{
    // Ensure the value is one of the options
    int index = GetItemIdxByLabel( _label );
    if( index < 0 )
    {
        // Item not found
        return false;
    }
    return SetByValue( index );
}`
The argument for **SetByValue** is the **index** in the list. So in **SetByValue**, the call **GetItemIdxByValue** uses as argument _not a value_, but an _index_. 
The name of argument is confuse, I think it should be _index instead f _value.

Fishwaldo, 2016-12-19:
This is fixed after reverting the ValueList change recently committed.

closed 2016-12-19 by Fishwaldo