OpenZWave/open-zwave #1070: Remotec ZXT-120 and Stick GEN5

jotakar, 2016-12-18:
I've observed this issue with Remotec ZXT-120EU V1.8 (IR gateway for Air Conditioner) working with Aeon Z-Stick GEN5. 
There are more details in https://github.com/OpenZWave/open-zwave/issues/1063, but I think this issue deserves a new thread, specific for it.
When we set a parameter to this device ,  OZwave sends two commands: one for setting the value and one for getting this value. In log we see 
```
2016-12-18 11:45:45.302 Info, Node019, Value::Set - COMMAND_CLASS_THERMOSTAT_MODE - Mode - 0 - 1 - Heat
2016-12-18 11:45:45.302 Detail, Node019, Queuing (Send) ThermostatModeCmd_Set (Node=19): 0x01, 0x0a, 0x00, 0x13, 0x13, 0x03, 0x40, 0x01, 0x01, 0x25, 0x8b, 0x18
2016-12-18 11:45:45.303 Detail, Node019, Queuing (Send) ThermostatModeCmd_Get (Node=19): 0x01, 0x09, 0x00, 0x13, 0x13, 0x02, 0x40, 0x02, 0x25, 0x8c, 0x1f```
That's Ok and it works for any other device. But Remotec doesn¡t receive the send command and the get command produces an error:
```
2016-12-18 11:45:45.304 Info, Node019, Sending (Send) message (Callback ID=0x8b, Expected Reply=0x13) - ThermostatModeCmd_Set (Node=19): 0x01, 0x0a, 0x00, 0x13, 0x13, 0x03, 0x40, 0x01, 0x01, 0x25, 0x8b, 0x18
2016-12-18 11:45:45.313 Detail, Node019,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2016-12-18 11:45:45.313 Detail, Node019,   ZW_SEND_DATA delivered to Z-Wave stack
2016-12-18 11:45:45.330 Detail, Node019,   Received: 0x01, 0x07, 0x00, 0x13, 0x8b, 0x00, 0x00, 0x03, 0x63
2016-12-18 11:45:45.330 Detail, Node019,   ZW_SEND_DATA Request with callback ID 0x8b received (expected 0x8b)
2016-12-18 11:45:45.330 Info, Node019, Request RTT 26 Average Request RTT 25
2016-12-18 11:45:45.331 Detail, Node019, Removing current message
2016-12-18 11:45:45.331 Info, Node019, Sending (Send) message (Callback ID=0x8c, Expected Reply=0x04) - ThermostatModeCmd_Get (Node=19): 0x01, 0x09, 0x00, 0x13, 0x13, 0x02, 0x40, 0x02, 0x25, 0x8c, 0x1f
2016-12-18 11:45:45.340 Detail, Node019,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2016-12-18 11:45:45.340 Detail, Node019,   ZW_SEND_DATA delivered to Z-Wave stack
2016-12-18 11:45:45.357 Detail, Node019,   Received: 0x01, 0x07, 0x00, 0x13, 0x8c, 0x00, 0x00, 0x02, 0x65
2016-12-18 11:45:45.357 Detail, Node019,   ZW_SEND_DATA Request with callback ID 0x8c received (expected 0x8c)
2016-12-18 11:45:45.357 Info, Node019, Request RTT 25 Average Request RTT 25
2016-12-18 11:45:51.487 Detail, Node019,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x13, 0x03, 0x80, 0x03, 0x64, 0x05
2016-12-18 11:45:51.487 Info, Node019, Response RTT 6155 Average Response RTT 4701
2016-12-18 11:45:51.488 Info, Node019, Received Battery report from node 19: level=100
2016-12-18 11:45:51.488 Detail, Node019, Refreshed Value: old value=100, new value=100, type=byte
2016-12-18 11:45:51.489 Detail, Node019, Changes to this value are not verified
2016-12-18 11:45:51.489 Detail, Node019, Notification: ValueChanged
2016-12-18 11:45:55.332 Error, Node019, ERROR: Dropping command, expected response not received after 1 attempt(s)
2016-12-18 11:45:55.334 Detail, Node019, Removing current message
2016-12-18 11:45:55.334 Detail, Node019, Notification: Notification - TimeOut
```
If we avoid to use the get command (Value.cpp, line 278, delete cc->RequestValue(...)) Remotec receives the Send command, sets the parameter and sends the proper IR command to the AC, without any delay. 
Modify MAX_TRIES or RETRY_TIMEOUT or parameters in options.xml doesn't fix the issue. 
If we add getsupported = false in remotec config file the device work but we cannot read from it any value. This is not a solution, because the Get command is not send.
I suppose there is a way to send a Get command **after** Send command has finalized totally, because it seems that, now, the Get command reaches the device **before** set command is completed.
Any suggestion?

Fishwaldo, 2016-12-19:
As far as we know, there is no difference between a Gen5 stick and a original stick from the SerialAPI protocol, so if you are having issues with Gen5 sticks, I doubt its something we can fix (unless sigma release the SerialAPI specs, which they have said they wont). 

If its failing because its getting a Get straight after a Set - Crappy Product. 👎 Not much we can do as OZW is multithreaded on the sending queue. Putting Sleeps in is not going to help it, nor will retries etc. As soon as we get a "ZW_SEND_DATA Request with callback ID" message, that means that the device has acknowledged the message from the controller. 

I'd speak to Remotec about this. See if they have any insight.

jotakar, 2016-12-27:
Remotec doesn't answer. 
Meanwhile I wonder if it would be posible to set a delay after callback ID is received (perhaps only for slow devices). I suppose the file to modify is  Driver.cpp, is it right? If so I would try. 
Thanks

Fishwaldo, 2016-12-28:
yes. You would probably want to do it after recieving the ACK from the device in Driver::ProcessMsg

jotakar, 2016-12-28:
Thanks. 
I was not sure where to put the sleep in ProcessMsg, so  I added a delay in Driver::RemoveCurrentMsg 
The delay is set only if the affected node is the Remotec ZXT-120EU. It works with 5 mS (usleep(5000)).
Yes, it's a bit tricky to check the name of the node.  
So my next step will be to add a new  parameter to this device in zwcfg. This parameter value will be the sleeping interval  I want for the device. So the delay will be used if this new parameter is greater than 0.

jotakar, 2016-12-29:
I've received an answer from Remotec and they tell me what following:

We have analyzed, it might be the problem of the command class version. ZXT-120 doesn’t support Thermostat Mode Version 3 commands. Please follow below supported command class version.
[image]

I suppose OZwave can use any version of command class...

jotakar, 2016-12-30:
An more info from Remotec

> Hi Juan, 
> I don’t think it is related to ACK. In Z-Wave bottom layer( 300 series Z-Wave module), when it receives a completed/valid command, Z-Wave module stack will automatically send ACK out, which is not handled by us.
> 
> Best regards,

closed 2017-03-21 by jotakar

bbalesz, 2017-05-13:
Hi jotokar,
I'm facing the same issue with my ZXT-120. Did you find a good solution?
thank's,
Balint

jotakar, 2017-05-13:
Yes, I found a fix and I sent it to Gtihub as a pull request. You can see it in
https://github.com/OpenZWave/open-zwave/pull/1090
this pull request was assigned to @Fishwaldo  but I have no news  about it since February. It looks like he can not find time to merge this PR. So I'm waiting for this merge. 
But you can download the modified files and merge them in your copy of OpenZwave
Best regards

nongagf, 2017-08-06:
Hi @jotakar,

I'm experiencing the same issue with my ZXT-120 and Aeotec Z-Stick Gen5. Waiting for your pull request to finally go through...

Is this issue specific to the Aeotec sticks? I'm wondering if replacing my stick with a Sigma UZB stick would solve the problem...

Thanks!

jotakar, 2017-08-07:
Hi @nongagf 
Sorry, but I don't know if Sigma Stick solves this issue. 
This request is waiting for about 6 months!!!!!, I don't know the reason. (and I don't know how to change the person who is assigned to it).
Best regards

austwhite, 2018-09-07:
A year on.  Just wanted to note, after reading this, that the issue is unique to OpenZwave and still present.  Commercial stacks and even OpenHab work perfectly with the device.
@jotakar:  I know you have tried really hard, but maybe talking to CJ at OpenHab to see what his solution with this product was may help.  I have no coding experience to note so I am unable to help with a solution, but there must be one as non-OpenZwave based implementations work.

austwhite, 2018-09-07:
@jotakar
> And It would work fine with OpenZwave too aplying this fix: #1090.
I think it is more that there may be other implications of the code the way it currently stands with the patch.  I read a couple of threads that suggest it may cause other issues.

If you talk to Chris @openhab, he is the one that basically focused on all the z-wave stuff there, maybe he has a better solution that @Fishwaldo could find useful and maybe neater in OpenZWave.

Domoticz uses OpenZWave and the ZXT-120 does not work in that either, so it makes me think OpenHab must use a different Z-Wave stack, or at least a modified one.

Fishwaldo, 2018-09-07:
@jotakar once you pay my salary, and I earn a living from OZW, you are free to make as many criticisms of me and the progress of OZW as you wish... I have already mentioned that this is not the right solution and will not merge it.

If you had applied the same amount of energy bugging the vendor to fix their buggy firmware (instead of insisting on a workaround in OZW) this might be fixed right now. OZW works with ~982 other products without requiring this workaround. 

The negative energy you display really doesn't motivate me to commit my spare time to work on OZW when all people do is complain. 

Locking this thread as I'm sick of this discussion. Apologies to the others that participated, this isn't directed towards you.