OpenZWave/open-zwave #1072: "handle request for 0xa8" with Sigma Designs bridge controller

gdurand, 2016-12-21:
When using a Sigma Designs bridge controller UZB flashed with the firmware from SDK 6.61, some frames from a Popp thermostat are not interpreted, which prevents from getting temperatures, etc.
The Sigma Designs static controller (version unknown, but probably older) works fine though.
I don't know if previous versions of the bridge controller work fine.
The same problem appears with a switching actuator.

I attached the logs generated when interacting with the thermostat. For the static controller I receive a bunch of info and for the bridge this stops quickly.
[sigma-bridge.txt](https://github.com/OpenZWave/open-zwave/files/666472/sigma-bridge.txt)
[sigma-static.txt](https://github.com/OpenZWave/open-zwave/files/666473/sigma-static.txt)

gdurand, 2016-12-21:
sorry, for duplicate

closed 2016-12-21 by gdurand

sayegh, 2016-12-21:
Your log says:

```2016-12-21 13:07:23.546 Warning, Failed - Network Key Not Set```

Could you set the key first and post the result again?

Fishwaldo, 2016-12-21:
Bridge firmware is not supported