OpenZWave/open-zwave #1073: Manual "Refresh Node Info" wont clear cached Node config in ozwconfig.xml file.

RicardP, 2016-12-21:
Manual "Refresh Node Info" wont clear cached Node config in ozwconfig.xml file, you need to manually edit (delete) the ozw_config/cach.xml in order to restore the corrupt node config... really annoying :-)

Thanks for listening,
Ricard

Fishwaldo, 2016-12-21:
The cache is only saved when writeconfig is called did you do that?

RicardP, 2016-12-22:
Hi Fishwaldo, I should told I am using Domoticz on Windows 7, the OZW version included by Domoticz is Version: 1.4.2246.ga583ba8.dirty. I am a user not programmer, so the writeconfig I don't recognize as Domoticz user. My problem is that when I discover duplicate entries etc. with a node in the ozwcpl I must stop the Domoticz service and open ozwconfig.xml and delete the node manually. After that "hopefully" ozw populate the node config correctly in cachfile. 

But the main issue from begining is why nodes ends up corrupt with duplicates switches, temp sensors are rediscovered as new identities etc. this is the biggest issue as it happens regular.

I have for example 3 pcs of Fibaro Universal Sensor with 4 temp sensors on each, one of each temp sensor is installed in each room in my villa. However from time to time the 4 temp sensors are lost and instead seen as temp+humidity sensors. As I am using tem sensors to control a relay board switching on and of floor heating suddenly a room gets war or hot because the correct temp sensors are gone. 

In this situation I have to go into ozwconfig.xml and delete the related Fibaro Universal sensor to rediscover the 4 correct temp sensors.

So why does nodes get corrupt in the first place without system correcting it? And secondly by pushing "Refresh Node Info" is it possible already to let ozw first wipe the specific node config in .xml so I dont need to manually do it?

I have spent quite much money on z-wave equipment and other automation hardware, it gets some frustrating to see system critical functions, as home heating, fails on "small issues"(?). Of course I know what I am using Open Source and you are doing a impressive work for OZW, big thanks! :)

sidhuy, 2016-12-26:
@RicardP @Fishwaldo 
Please assist with this issue: https://github.com/OpenZWave/open-zwave/issues/1075
Really need help with this, having difficulty installing OpenZWave library

Fishwaldo, 2016-12-28:
@sidhuy please do not spam other issues. 

@RicardP these issues, seem to be related to how Domoticz is interfacing with OZW. If your not a programmer, suggest you ask the Domoticz guys to look at this with me. I can't help, as I don't use Domoticz.

sidhuy, 2016-12-28:
@Fishwaldo my apologies, it won't happen again

RicardP, 2016-12-28:
@Fishwaldo Ok so duplicates endpoints already seen in OZW node list as below clip (Philio PAN04-1B) may bee a Domoticz cause? I thought Domoticz just fetched data from OZW... Further, if using default "Native OZW code" (never seen) nodes data in cache file .xml shall be cleared if pushing "Refresh Node Info"? I also experience that my Fibaro temp sensors described above is most likely a Domoticz "spin off"?  

Kind regards,
Ricard

Fishwaldo, 2017-02-17:
I really don't follow what your saying but:
Node Numbers are directly retrieved from the Controller, not generated by OZW, so if your seeing duplicate nodes in your network - Somehow the controller is "re-adding" them to the Network. You can use the HasNodeFailed and RemoveFailedNode Commands to remove the "ghost" nodes, but you need to know how the new devices are appearing in the first place.

RicardP, 2017-02-17:
Sorry I thought I was explaining in details, no I don't get the node itself duplicated. Its the Node capabilities that is retrieved and built up at inclusion that is trashed more and more over time. Please take a look at the above screenshot that I have marked a false or "duplicate" field in this case only one false entry. On some Nodes I had at a time like 20-50 false entries building up in the cache file. A few "false" don't do any harm but after time as cache file grows by false data it slows down the system and false entries seems (not sure) increasing the speed of new false entries.

I am not alone... :)

First, why is the false entries popping up? I have seen it from different Manufactures by the way. Fibaro Universal Sensor, Philio PAN04-1B, etc.. Maybe OZW could handle it differently, check if an entry is false i.e. could the Controller ask the Node to confirm it before adding it as a new entry...?

Second, The question I started with.. and a half good solution if system cant keep Nodes fresh by itself: If manually using the "Refresh Node Info" in control panel Node should be rebuilt in cache file, or if not why is it called Refresh Node Info?

Many thanks! :-)

Fishwaldo, 2017-02-22:
ok.... I've not seen this before, so I need you to try to find out when the false ValueID's are being created. ie, when you do a certian action etc.

second - RefreshNodeInfo - Maybe this is not clear. NodeInfo (NI) is a specific packet in the ZWave Protocol, and not actually to "refresh the entire node configuration". a NodeInfo packet only contains the supported CommandClass information the node advertises. So you should not need to call this unless the Node Command Classes change (and the only device I've seen where this might happen is the Aeotec 5 in 1 MultiSensor where the CC's change if you switch between USB and Battery Powered. Just about every other device has a "static" NI packet and doesn't change at all.

Fishwaldo, 2017-02-22:
ok. Sorry, I've just re-read the code, and a RefreshNodeInfo might cause some corruption. There is too much State information stored in various classes to make it work as intended, so I'm about to push a fix to the Dev branch that will essentially delete a Node, and Reload it without reading any cache. This should fix the issues you are seeing.

But again, unless you know the CC or Capabilities of a node have changed (very very rare) or you have changed a Config file (only works on Dev branch) this isn't a Method you should be calling.

RicardP, 2017-02-22:
Hi, thanks for looking into it and that some issues was found, regardless my experiences. I never use the RefreshNodeInfo except if I see a Node have gone corrupt with false ValueID's, and the false populate the affected Node over time under normal usage. It could be because of bad signal/RF maybe causing the demodulated data being corrupt. However I heard the Z-Wave protocol have some error corcetion CRC to avoid corrupt data being accepted... But also, a new ValueID should be verified as "true" before added to the system. Well, or something like that...

As speaking, I saw this commit was added to the 1.6 milestoine, any plan for the launch or what achievement have to be fulfilled first? Thanks again for an exiting and useful OZW project :)

Fishwaldo, 2017-03-21:
Lack of Time is the limiting factor for me..... 

I'm assuming all is good now. Closing the issue.

closed 2017-03-21 by Fishwaldo

RicardP, 2017-03-21:
Hi Justin,

 

I understand, take care!

 

//Ricard

 

Detta e-postmeddelande har sökts igenom efter virus med antivirusprogram från Avast.
https://www.avast.com/antivirus