OpenZWave/open-zwave #1076: Fibaro FGWP(F)-102 no updates

ZyberSE, 2016-12-28:
I just added the new Fibaro FGWP(F)-102 Z-Plus Wall Plug, I had no issues including it and I see reading and can control on/off state.
But I don't get any updates from the device beside the on/off state. The only readings I see is after I restart OZW, then it reads kWh and W.

I have a plasma tv connected to it and I've seen readings ~196W.

This is from the OZW log then the tv turns off.
`2016-12-28 08:42:04.080 Detail, Node012,   Received: 0x01, 0x08, 0x00, 0x04, 0x00, 0x0c, 0x02, 0x98, 0x40, 0x25
2016-12-28 08:42:04.080 Info, Node012, Received SecurityCmd_NonceGet from node 12
2016-12-28 08:42:04.081 Info, NONCES: 0x8a, 0x4e, 0x37, 0xf6, 0xca, 0xf2, 0x55, 0xe7
2016-12-28 08:42:04.081 Info, NONCES: 0xad, 0xc9, 0x37, 0x91, 0x66, 0x30, 0xa3, 0x79
2016-12-28 08:42:04.081 Info, NONCES: 0x17, 0x9c, 0x13, 0x8d, 0x45, 0xde, 0xef, 0xfb
2016-12-28 08:42:04.081 Info, NONCES: 0x3a, 0x92, 0x2d, 0x75, 0x58, 0x96, 0xdb, 0xe1
2016-12-28 08:42:04.081 Info, NONCES: 0xe3, 0x12, 0x57, 0x2d, 0x83, 0xac, 0x14, 0x30
2016-12-28 08:42:04.081 Info, NONCES: 0xf4, 0xc9, 0xc1, 0xd9, 0xf8, 0xe3, 0x52, 0x8f
2016-12-28 08:42:04.081 Info, NONCES: 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
2016-12-28 08:42:04.082 Info, NONCES: 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
2016-12-28 08:42:04.082 Info, Node012, Sending (Send) message (Callback ID=0x01, Expected Reply=0x00) - Nonce_Report - 0x01, 0x11, 0x00, 0x13, 0x0c, 0x0a, 0x98, 0x80, 0xf4, 0xc9, 0xc1, 0xd9, 0xf8, 0xe3, 0x52, 0x8f, 0x05, 0x01, 0x04:
2016-12-28 08:42:04.088 Detail,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2016-12-28 08:42:04.088 Detail,   ZW_SEND_DATA delivered to Z-Wave stack
2016-12-28 08:42:04.104 Detail,   Received: 0x01, 0x07, 0x00, 0x13, 0x01, 0x00, 0x00, 0x02, 0xe8
2016-12-28 08:42:04.104 Detail,   ZW_SEND_DATA Request with callback ID 0x01 received (expected 0x01)
2016-12-28 08:42:04.122 Detail, Node012,   Received: 0x01, 0x1d, 0x00, 0x04, 0x00, 0x0c, 0x17, 0x98, 0x81, 0x55, 0x24, 0x70, 0x0f, 0x29, 0x47, 0x28, 0x7b, 0x8b, 0x03, 0xe7, 0x20, 0xf4, 0xc2, 0x11, 0x8f, 0xc0, 0xa3, 0x00, 0x57, 0xb8, 0xbc
2016-12-28 08:42:04.122 Info, Raw: 0x98, 0x81, 0x55, 0x24, 0x70, 0x0f, 0x29, 0x47, 0x28, 0x7b, 0x8b, 0x03, 0xe7, 0x20, 0xf4, 0xc2, 0x11, 0x8f, 0xc0, 0xa3, 0x00, 0x57, 0xb8, 0xbc
2016-12-28 08:42:04.122 Detail, Node012, Decrypted Packet: 0x00, 0x20, 0x01, 0xff
2016-12-28 08:42:04.122 Detail,
2016-12-28 08:42:04.122 Info, Node012, Received Basic set from node 12: level=255.  Sending event notification.
2016-12-28 08:42:04.123 Detail, Node012, Notification: NodeEvent`

This is when it turns on.
`2016-12-28 08:43:18.080 Detail, Node012,   Received: 0x01, 0x08, 0x00, 0x04, 0x00, 0x0c, 0x02, 0x98, 0x40, 0x25
2016-12-28 08:43:18.081 Info, Node012, Received SecurityCmd_NonceGet from node 12
2016-12-28 08:43:18.081 Info, NONCES: 0x8a, 0x4e, 0x37, 0xf6, 0xca, 0xf2, 0x55, 0xe7
2016-12-28 08:43:18.081 Info, NONCES: 0xad, 0xc9, 0x37, 0x91, 0x66, 0x30, 0xa3, 0x79
2016-12-28 08:43:18.081 Info, NONCES: 0x17, 0x9c, 0x13, 0x8d, 0x45, 0xde, 0xef, 0xfb
2016-12-28 08:43:18.082 Info, NONCES: 0x3a, 0x92, 0x2d, 0x75, 0x58, 0x96, 0xdb, 0xe1
2016-12-28 08:43:18.082 Info, NONCES: 0xe3, 0x12, 0x57, 0x2d, 0x83, 0xac, 0x14, 0x30
2016-12-28 08:43:18.082 Info, NONCES: 0xf4, 0xc9, 0xc1, 0xd9, 0xf8, 0xe3, 0x52, 0x8f
2016-12-28 08:43:18.082 Info, NONCES: 0xfe, 0xe4, 0x9b, 0xc2, 0x42, 0x0a, 0xbd, 0x7b
2016-12-28 08:43:18.082 Info, NONCES: 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
2016-12-28 08:43:18.083 Info, Node012, Sending (Send) message (Callback ID=0x01, Expected Reply=0x00) - Nonce_Report - 0x01, 0x11, 0x00, 0x13, 0x0c, 0x0a, 0x98, 0x80, 0xfe, 0xe4, 0x9b, 0xc2, 0x42, 0x0a, 0xbd, 0x7b, 0x05, 0x01, 0x2a:
2016-12-28 08:43:18.089 Detail,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2016-12-28 08:43:18.089 Detail,   ZW_SEND_DATA delivered to Z-Wave stack
2016-12-28 08:43:18.105 Detail,   Received: 0x01, 0x07, 0x00, 0x13, 0x01, 0x00, 0x00, 0x01, 0xeb
2016-12-28 08:43:18.106 Detail,   ZW_SEND_DATA Request with callback ID 0x01 received (expected 0x01)
2016-12-28 08:43:18.124 Detail, Node012,   Received: 0x01, 0x1d, 0x00, 0x04, 0x00, 0x0c, 0x17, 0x98, 0x81, 0x93, 0x41, 0xf5, 0xf4, 0x25, 0x60, 0xf9, 0xcc, 0x8e, 0x2f, 0xa9, 0xac, 0xfe, 0x27, 0xa1, 0xdd, 0x2a, 0xf0, 0xbe, 0x4c, 0x72, 0x1c
2016-12-28 08:43:18.124 Info, Raw: 0x98, 0x81, 0x93, 0x41, 0xf5, 0xf4, 0x25, 0x60, 0xf9, 0xcc, 0x8e, 0x2f, 0xa9, 0xac, 0xfe, 0x27, 0xa1, 0xdd, 0x2a, 0xf0, 0xbe, 0x4c, 0x72, 0x1c
2016-12-28 08:43:18.124 Detail, Node012, Decrypted Packet: 0x00, 0x20, 0x01, 0x00
2016-12-28 08:43:18.124 Detail,
2016-12-28 08:43:18.124 Info, Node012, Received Basic set from node 12: level=0.  Sending event notification.
2016-12-28 08:43:18.124 Detail, Node012, Notification: NodeEvent`

Here are the logs with "report" after a restart.
`pi@hass:/home/hass/.homeassistant $ cat OZW_Log.txt | grep report
2016-12-28 08:49:13.239 Info, **TODO: handle request for 0x0a** Please report this message.
2016-12-28 08:49:42.426 Info, Node012, Number of association groups reported for node 12 is 3.
2016-12-28 08:49:42.544 Info, Node012, Received Multi Instance Association report from node 12, group 1
2016-12-28 08:49:42.668 Info, Node012, Received Multi Instance Association report from node 12, group 2
2016-12-28 08:49:42.784 Info, Node012, Received Multi Instance Association report from node 12, group 3
2016-12-28 08:49:42.824 Info, Node012, Received a PowerLevel report: PowerLevel=Normal, Timeout=0
2016-12-28 08:49:42.961 Info, Node012, Received SwitchBinary report from node 12: level=On
2016-12-28 08:49:43.014 Info, Node012, Received SensorMultiLevel report from node 12, instance 1, Power: value=13.3W
2016-12-28 08:49:43.077 Info, Node012, Received Meter report from node 12: Energy=0.07kWh
2016-12-28 08:49:43.148 Info, Node012, Received Meter report from node 12: Power=13.4W
Reply`

YcKe, 2017-01-02:
Should be fixed with pull request #1080 . Can you please retest when the pull request is accepted? (And your system is also updated)

Fishwaldo, 2017-01-11:
check the configuration and see if there are timeouts etc associated with sending updates.

closed 2017-01-11 by Fishwaldo

nechry, 2017-01-11:
with new config,https://github.com/OpenZWave/open-zwave/pull/1080, file the power report will occure normally now.