OpenZWave/open-zwave #1077: changing report type of resessed door sensor failed.

remb0, 2016-12-29:
I use domoticz on a raspberry pi 3  with an zw089 recessed door sensor.

when I try to change:
 121. Sensor Binary Report -> This seems to be important because the Basic set did not work for me. 

When chaning this to basis my domoticz incl. openzwave controlpanel crashed.
when changing it from domoticz or the ozw panel.

I have excluded and included the sensor again. but no different. I hope this is the right spot to post this issue.

here I see the same problem from other users:
http://www.domoticz.com/forum/viewtopic.php?f=17&t=5099&start=20&hilit=Aeotec+ZW089+Recessed+Door+Sensor+Gen5

openzwave-dev-watcher, 2017-01-10:
any news on this one?

remb0, 2017-01-10:
No. I have no Clue.. and still the problem.  I have seen other members with same problems. with this sensor (zw089 but also with others). If I can help to debug please let me know.

nechry, 2017-01-11:
I have also this device and is working fine for me with Sensor Binary Report.

remb0, 2017-01-11:
But I can't change it to binary. when I go openzwavecontrol panel crashes?  can I do this by editing a file?

nechry, 2017-01-11:
No you can change on your zwcfg file. Ok yes you can but the device will not got this parameter in memory. Yous sur you use latest version off ozw and devices config?

remb0, 2017-01-12:
thanks for advice.  my version is: Version: 1.4-2279-g7c8c27d-dirty
and my zw089.xml is the same as here in the github repo.
my config file 
`<Node id="40" name="Voordeur" location="" basic="4" generic="7" specific="1" roletype="6" devicetype="3072" nodetype="0" type="Notification Sensor" listening="false" frequentListening="false" beaming="true" routing="true" max_baud_rate="40000" version="4" query_stage="CacheLoad">
        <Manufacturer id="86" name="Aeotec">
            <Product type="2" id="59" name="ZW089 Recessed Door Sensor Gen5" />
        </Manufacturer>
        <CommandClasses>
            <CommandClass id="32" name="COMMAND_CLASS_BASIC" version="1" request_flags="4" setasreport="true">
                <Instance index="1" />
                <Value type="byte" genre="basic" instance="1" index="0" label="Basic" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="0" />
            </CommandClass>
            <CommandClass id="48" name="COMMAND_CLASS_SENSOR_BINARY" version="1" request_flags="4" innif="true">
                <Instance index="1" />
                <Value type="bool" genre="user" instance="1" index="0" label="Sensor" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
                <SensorMap index="0" type="234" />
            </CommandClass>
            <CommandClass id="90" name="COMMAND_CLASS_DEVICE_RESET_LOCALLY" version="1" request_flags="4" after_mark="true" innif="true">
                <Instance index="1" />
            </CommandClass>
            <CommandClass id="94" name="COMMAND_CLASS_ZWAVE_PLUS_INFO" version="1" request_flags="4" innif="true">
                <Instance index="1" />
                <Value type="byte" genre="system" instance="1" index="0" label="ZWave+ Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="1" />
                <Value type="short" genre="system" instance="1" index="1" label="InstallerIcon" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="-32768" max="32767" value="3072" />
                <Value type="short" genre="system" instance="1" index="2" label="UserIcon" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="-32768" max="32767" value="3072" />
            </CommandClass>
            <CommandClass id="112" name="COMMAND_CLASS_CONFIGURATION" version="1" request_flags="4" innif="true">
                <Instance index="1" />
                <Value type="list" genre="config" instance="1" index="1" label="Sensor Binary Report Toggle" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="1" vindex="0" size="1">
                    <Help>Which value of the Sensor Binary Report will be sent when the magnet switch to open/close.</Help>
                    <Item label="Normal" value="0" />
                    <Item label="Inverse" value="1" />
                </Value>
                <Value type="list" genre="config" instance="1" index="3" label="Basic Set Toggle" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="1" vindex="0" size="1">
                    <Help>Which value of the Basic Set will be sent when the magnet switch to open/close.</Help>
                    <Item label="Normal" value="0" />
                    <Item label="Inverse" value="1" />
                </Value>
                <Value type="list" genre="config" instance="1" index="101" label="Low Battery Check Enable" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="1" vindex="0" size="1">
                    <Help>Enable/disable the function of low battery checking, when the current voltage is less than the warning voltage, it will send the Battery Low Warning Report.</Help>
                    <Item label="Disabled" value="0" />
                    <Item label="Enabled" value="1" />
                </Value>
                <Value type="int" genre="config" instance="1" index="111" label="Low Battery Check Interval" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="2147483647" value="86640">
                    <Help>This parameter is available when the low battery checking function is activated or the device was waked up by other actions (i.e. the z-wave button trigger, magnet switch trigger and the Wake Up Interval Set timeout trigger). The Recessed Door Sensor will check the battery power when it was wake up as other ways. For example: the z-wave button trigger, magnet switch trigger and the Wake Up Interval Set timeout trigger. Default setting: 86640</Help>
                </Value>
                <Value type="list" genre="config" instance="1" index="121" label="Report Type" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" vindex="1" size="4">
                    <Help>Configure which sensor report will be sent when the sensor is Opened/Closed</Help>
                    <Item label="Disabled" value="0" />
                    <Item label="Basic Set" value="256" />
                    <Item label="Sensor Binary Report" value="16" />
                    <Item label="Basic Set and Sensor Binary Report" value="272" />
                </Value>
[... cut ...]
                <Value type="string" genre="system" instance="1" index="2" label="Application Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="1.12" />
            </CommandClass>
            <CommandClass id="152" name="COMMAND_CLASS_SECURITY" version="1" request_flags="4" innif="true">
                <Instance index="1" />
                <Value type="bool" genre="system" instance="1" index="0" label="Secured" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
            </CommandClass>
        </CommandClasses>
    </Node>`

nechry, 2017-01-12:
I think you have include the device in secure mode? try to exclude it and include again but in non secure this time.

remb0, 2017-01-12:
finally it works! I pushed the button a few seconds so the sensor was not sleeping anymore. then I could change the parameter.

thanks for all help! this topic can be closed.

nechry, 2017-01-12:
the issue is already close few days ago, but cool if your problem is solve