OpenZWave/open-zwave #1083: Zipato Keypad RFID - Sending wake up no more info too early

Mavyre, 2017-01-05:
Hello,

It seems that the WakeUp_NoMoreInformation is sent too early to the device. HA software doesn't have the time to send the Binary Switch to change the notification, and is then sent to the next WakeUp, meaning the next time that a badge/code is validated. Notification sound are thus completely messed up.

Here are a part of the log that could help you:
`2017-01-05 16:37:05.819 Info, Node010, Sending (WakeUp) message (Callback ID=0x23, Expected Reply=0x13) - WakeUpCmd_NoMoreInformation (Node=10): 0x01, 0x09, 0x00, 0x13, 0x0a, 0x02, 0x84, 0x08, 0x25, 0x23, 0x67
2017-01-05 16:37:05.828 Detail, Node010,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2017-01-05 16:37:05.828 Detail, Node010,   ZW_SEND_DATA delivered to Z-Wave stack
2017-01-05 16:37:05.844 Detail, Node010,   Received: 0x01, 0x07, 0x00, 0x13, 0x23, 0x00, 0x00, 0x02, 0xca
2017-01-05 16:37:05.844 Detail, Node010,   ZW_SEND_DATA Request with callback ID 0x23 received (expected 0x23)
2017-01-05 16:37:05.844 Info, Node010, Request RTT 24 Average Request RTT 24
2017-01-05 16:37:05.844 Info, Node010,   Node 10 has been marked as asleep
2017-01-05 16:37:05.844 Detail,   Expected callbackId was received
2017-01-05 16:37:05.844 Detail,   Expected reply was received
2017-01-05 16:37:05.844 Detail,   Message transaction complete
2017-01-05 16:37:05.844 Detail,
2017-01-05 16:37:05.844 Detail, Node010, Removing current message
2017-01-05 16:37:05.845 Detail, Node010, Notification: Notification - Node Asleep
2017-01-05 16:37:05.850 Detail,
2017-01-05 16:37:05.850 Info, Node010, Sending (Send) message (Callback ID=0x24, Expected Reply=0x13) - SwitchBinaryCmd_Set (Node=10): 0x01, 0x0a, 0x00, 0x13, 0x0a, 0x03, 0x25, 0x01, 0xff, 0x25, 0x24, 0x35
2017-01-05 16:37:05.858 Detail, Node010,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2017-01-05 16:37:05.858 Detail, Node010,   ZW_SEND_DATA delivered to Z-Wave stack
2017-01-05 16:37:10.135 Detail, Node010,   Received: 0x01, 0x07, 0x00, 0x13, 0x24, 0x01, 0x01, 0xac, 0x63
2017-01-05 16:37:10.135 Detail, Node010,   ZW_SEND_DATA Request with callback ID 0x24 received (expected 0x24)
2017-01-05 16:37:10.135 Info, Node010, WARNING: ZW_SEND_DATA failed. No ACK received - device may be asleep.
2017-01-05 16:37:10.135 Info, Node010, Node not responding - moving message to Wake-Up queue: SwitchBinaryCmd_Set (Node=10): 0x01, 0x0a, 0x00, 0x13, 0x0a, 0x03, 0x25, 0x01, 0xff, 0x25, 0x24, 0x35
2017-01-05 16:37:10.135 Info, Node010, Node not responding - moving message to Wake-Up queue: SwitchBinaryCmd_Get (Node=10): 0x01, 0x09, 0x00, 0x13, 0x0a, 0x02, 0x25, 0x02, 0x25, 0x25, 0xca
2017-01-05 16:37:10.136 Info, Node010, Node not responding - moving message to Wake-Up queue: UserCodeCmd_Get (Node=10): 0x01, 0x0a, 0x00, 0x13, 0x0a, 0x03, 0x63, 0x02, 0x03, 0x25, 0x22, 0x8a
2017-01-05 16:37:10.607 Detail, Node010,   Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x0a, 0x0a, 0x71, 0x05, 0x00, 0x00, 0x00, 0xff, 0x06, 0x06, 0x01, 0x01, 0x60
`

Fishwaldo, 2017-01-06:
Need the full log to see what's going on.

Mavyre, 2017-01-09:
Leaving you all the logs I could find. Note that my Keypad is Node010 : 

```
2017-01-09 20:22:28.315 Detail, Node010,   Received: 0x01, 0x14, 0x00, 0x04, 0x00, 0x0a, 0x0e, 0x63, 0x03, 0x00, 0x00, 0x31, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xba
2017-01-09 20:22:28.315 Detail,
2017-01-09 20:22:28.315 Info, Node010, User Code Packet is 10
2017-01-09 20:22:28.315 Detail, Node010, Initial read of value
2017-01-09 20:22:28.315 Info, Node010, Received User Code Report from node 10 for User Code 0 (Available)
2017-01-09 20:22:28.315 Detail, Node010, Notification: ValueChanged
2017-01-09 20:22:28.333 Detail, Node010,   Received: 0x01, 0x08, 0x00, 0x04, 0x00, 0x0a, 0x02, 0x84, 0x07, 0x78
2017-01-09 20:22:28.333 Detail,
2017-01-09 20:22:28.333 Info, Node010, Received Wakeup Notification from node 10
2017-01-09 20:22:28.333 Info, Node010,   Node 10 has been marked as awake
2017-01-09 20:22:28.333 Detail, Node010, Queuing (WakeUp) UserCodeCmd_Get (Node=10): 0x01, 0x0a, 0x00, 0x13, 0x0a, 0x03, 0x63, 0x02, 0x03, 0x25, 0x65, 0xcd
2017-01-09 20:22:28.333 Detail, Node010, Queuing (WakeUp) WakeUpCmd_NoMoreInformation (Node=10): 0x01, 0x09, 0x00, 0x13, 0x0a, 0x02, 0x84, 0x08, 0x25, 0x78, 0x3c
2017-01-09 20:22:28.333 Detail, Node010, Notification: Notification - Node Awake
2017-01-09 20:22:28.340 Detail,
2017-01-09 20:22:28.340 Info, Node010, Sending (WakeUp) message (Callback ID=0x65, Expected Reply=0x04) - UserCodeCmd_Get (Node=10): 0x01, 0x0a, 0x00, 0x13, 0x0a, 0x03, 0x63, 0x02, 0x03, 0x25, 0x65, 0xcd
2017-01-09 20:22:28.348 Detail, Node010,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2017-01-09 20:22:28.348 Detail, Node010,   ZW_SEND_DATA delivered to Z-Wave stack
2017-01-09 20:22:28.365 Detail, Node010,   Received: 0x01, 0x07, 0x00, 0x13, 0x65, 0x00, 0x00, 0x02, 0x8c
2017-01-09 20:22:28.366 Detail, Node010,   ZW_SEND_DATA Request with callback ID 0x65 received (expected 0x65)
2017-01-09 20:22:28.366 Info, Node010, Request RTT 25 Average Request RTT 24
2017-01-09 20:22:28.366 Detail,   Expected callbackId was received
2017-01-09 20:22:28.379 Detail, Node010,   Received: 0x01, 0x14, 0x00, 0x04, 0x00, 0x0a, 0x0e, 0x63, 0x03, 0x03, 0x01, 0x8f, 0xb5, 0xaa, 0xb2, 0x66, 0x50, 0x01, 0x04, 0x00, 0x00, 0x98
2017-01-09 20:22:28.379 Detail,
2017-01-09 20:22:28.379 Info, Node010, Response RTT 38 Average Response RTT 38
2017-01-09 20:22:28.379 Info, Node010, User Code Packet is 10
2017-01-09 20:22:28.379 Detail, Node010, Initial read of value
2017-01-09 20:22:28.379 Info, Node010, Received User Code Report from node 10 for User Code 3 (Occupied)
2017-01-09 20:22:28.379 Detail, Node010, Queuing (Query) UserCodeCmd_Get (Node=10): 0x01, 0x0a, 0x00, 0x13, 0x0a, 0x03, 0x63, 0x02, 0x04, 0x25, 0x79, 0xd6
2017-01-09 20:22:28.379 Detail, Node010,   Expected reply and command class was received
2017-01-09 20:22:28.379 Detail, Node010,   Message transaction complete
2017-01-09 20:22:28.379 Detail,
2017-01-09 20:22:28.379 Detail, Node010, Removing current message
2017-01-09 20:22:28.380 Detail, Node010, Notification: ValueChanged
2017-01-09 20:22:28.383 Detail,
2017-01-09 20:22:28.383 Info, Node010, Sending (WakeUp) message (Callback ID=0x78, Expected Reply=0x13) - WakeUpCmd_NoMoreInformation (Node=10): 0x01, 0x09, 0x00, 0x13, 0x0a, 0x02, 0x84, 0x08, 0x25, 0x78, 0x3c
2017-01-09 20:22:28.392 Detail, Node010,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2017-01-09 20:22:28.392 Detail, Node010,   ZW_SEND_DATA delivered to Z-Wave stack
2017-01-09 20:22:28.408 Detail, Node010,   Received: 0x01, 0x07, 0x00, 0x13, 0x78, 0x00, 0x00, 0x03, 0x90
2017-01-09 20:22:28.408 Detail, Node010,   ZW_SEND_DATA Request with callback ID 0x78 received (expected 0x78)
2017-01-09 20:22:28.409 Info, Node010, Request RTT 25 Average Request RTT 24
2017-01-09 20:22:28.409 Info, Node010,   Node 10 has been marked as asleep
2017-01-09 20:22:28.409 Detail,   Expected callbackId was received
2017-01-09 20:22:28.409 Detail,   Expected reply was received
2017-01-09 20:22:28.409 Detail,   Message transaction complete
2017-01-09 20:22:28.409 Detail,
2017-01-09 20:22:28.409 Detail, Node010, Removing current message
2017-01-09 20:22:28.409 Detail, Node010, Notification: Notification - Node Asleep
2017-01-09 20:22:28.417 Detail,
2017-01-09 20:22:28.417 Info, Node010, Sending (Query) message (Callback ID=0x79, Expected Reply=0x04) - UserCodeCmd_Get (Node=10): 0x01, 0x0a, 0x00, 0x13, 0x0a, 0x03, 0x63, 0x02, 0x04, 0x25, 0x79, 0xd6
2017-01-09 20:22:28.425 Detail, Node010,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2017-01-09 20:22:28.425 Detail, Node010,   ZW_SEND_DATA delivered to Z-Wave stack
2017-01-09 20:22:32.702 Detail, Node010,   Received: 0x01, 0x07, 0x00, 0x13, 0x79, 0x01, 0x01, 0xac, 0x3e
2017-01-09 20:22:32.702 Detail, Node010,   ZW_SEND_DATA Request with callback ID 0x79 received (expected 0x79)
2017-01-09 20:22:32.702 Info, Node010, WARNING: ZW_SEND_DATA failed. No ACK received - device may be asleep.
2017-01-09 20:22:32.702 Info, Node010, Node not responding - moving message to Wake-Up queue: UserCodeCmd_Get (Node=10): 0x01, 0x0a, 0x00, 0x13, 0x0a, 0x03, 0x63, 0x02, 0x04, 0x25, 0x79, 0xd6
2017-01-09 20:24:11.674 Detail, Node010,   Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x0a, 0x0a, 0x71, 0x05, 0x00, 0x00, 0x00, 0xff, 0x06, 0x06, 0x01, 0x02, 0x63
2017-01-09 20:24:11.675 Detail,
2017-01-09 20:24:11.675 Info, Node010, Received Alarm report: type=0, level=0, sensorSrcID=0, type:Access Control event:6, status=255
[... cut ...]
2017-01-09 20:24:15.975 Detail, Node010,   Received: 0x01, 0x07, 0x00, 0x13, 0x7b, 0x01, 0x01, 0xa3, 0x33
2017-01-09 20:24:15.975 Detail, Node010,   ZW_SEND_DATA Request with callback ID 0x7b received (expected 0x7b)
2017-01-09 20:24:15.975 Info, Node010, WARNING: ZW_SEND_DATA failed. No ACK received - device may be asleep.
2017-01-09 20:24:15.976 Info, Node010, Node not responding - moving message to Wake-Up queue: UserCodeCmd_Get (Node=10): 0x01, 0x0a, 0x00, 0x13, 0x0a, 0x03, 0x63, 0x02, 0x05, 0x25, 0x7b, 0xd5
```

Fishwaldo, 2017-01-11:
Nothing is wrong here. It appears that the Application is triggering a UserCode_Get message maybe upon a ValueChanged event? 

The Log starts with 2 messages queued up - UserCode_Get and NoMoreInformation.
We recieve a WakeUP Message - And the UserCode_Get and NoMoreInformation is successfully sent
Then the application immediately request another UserCode_Get (why?) but of course, the device is asleep again, so it gets queued up.

And then the cycle repeats. 

I suggest checking your application out. This is not a bug.

closed 2017-01-11 by Fishwaldo

Mavyre, 2017-01-11:
Okay, first in my log, there are the SwitchBinaryCmd_Set missing. Then, let me explain to you why, in the case of the Zipato Keypad, we should wait before sending a NoMoreInformation:
[image]

As you can see, the keypad wakes up right after sending the Alarm Report, and the controller must send SwitchBinaryCmd_Set. But, most of the software controllers don't have the time to queue up the SwitchBinaryCmd_Set right after getting the Alarm Report and then everytime, they queue it up after the NoMoreInformation.
The keypad then triggers the notification the next time it wakes up. 

After testing, I used my Keypad, and did not get the notification, because it was queued up after the NoMoreInformation. Then, two hours later, I heared the beep notification in my hall while I was watching TV.

Do you get it, or need further explaination?

I'll tag @nechry here, that is an active member in the forum of the Home Automation software that I use, and may be interested in that issue.

Fishwaldo, 2017-01-11:
as long as a command is "sent" prior to receiving a WakeUP Message, it will be sent prior to a NoMoreInformation. 
But as we are multi-threaded - If you "send" a message after a WakeUP is received, it will be luck of the draw if your messages gets sent before a NoMoreInformation or not (as we basically see if the queue of messages is 0 or not after sending each message to a sleeping device.)

I'm assuming the application sends the SwitchBinaryCmd_Set based on the AlarmReport message. if not, then you should work with the app team first.

Mavyre, 2017-01-11:
But, the Keypad may expect that you send a SwitchBinaryCmd_Set to 0xFF to start the notification, then wait 5 seconds, and send a SwitchBinaryCmd_Set to 0x00 to stop the notification and finally send the NoMoreInformation.
This is actually not doable with OpenZWave...

Mavyre, 2017-01-11:
Just crossed the merge request #1090 
Maybe setting one for the Zipato RFID should resolve this issue for most HA software.