OpenZWave/open-zwave #1085: Versioning seems incorrect

TimelessNL, 2017-01-06:
Hi,

There is a Release tag for v1.5 which is 1 day after tagged version V1.4. This seems strange in my opinion.

Normally a release tag should contain the latest (stable) release. And should not be some kind of future development version. A development version could use some kind of reversion(trunk) number.

**Example:**
Release = **v1.5**
Development(trunk) = **v1.5.r14785** or **r14785**

With the current use of tags its seems that there are 5 stable releases.
While there are actually:
- 2 retro release tags
- 2 release tags
- 1 development tag.

I thought that the newest release was v1.5 and happily compiled and used that for a while. until someone showed me that the downloaded archive used files from 11-jan-2016.

With the current use of tagging every release will create 1 "stable" release tag and directly after the release tag is 1 new trunk tag. So every release will bump the version number 2 times.

Regards,
Timeless

Fishwaldo, 2017-01-11:
If you read the mailing list and look at the tags, you will see 1.5 is tagged against the Dev branch, not master.

And as noted before Even - Stable. Odd - Development.

closed 2017-01-11 by Fishwaldo

TimelessNL, 2017-01-11:
Yeah I noticed. Thats also in my issue message.
But thats not a reason? More like a second explanation.

And why is my issue closed?

Fishwaldo, 2017-01-11:
because thats the way I'm developing the software. The version number is generated from the tags on the branch its checkout from. To distinguish between Stable and Dev I'm using 1.4 and 1.5.

TimelessNL, 2017-01-11:
I'm sorry to say but it seems that more people don't understand your way of versioning.

**Example see:**
Debian using 1.5 (which is v1.4 +1commit): [Link](https://packages.debian.org/sid/openzwave)
Buildroot bumped version 1.4 to 1.5 (which is again basically v1.4 +1 commit): [Buildroot changeset](https://git.busybox.net/buildroot/diff/package/openzwave/openzwave.mk?id=f5cfefffda6a5aab7efc83bc295672d558401e54)

And the build bot is building v1.4 as both Stable and Snapshot:
[Stable Build Bot](http://old.openzwave.com/downloads/)
[Snapshot (dev?) build bot](http://old.openzwave.com/snapshots/)

So, sure... using EVEN for STABLE and ODD for DEVELOPMENT is a way of versioning. But at least mention it somewhere because people are interpreting it the wrong way currently. And the buildbot is either configured wrong or it means something else I don't understand.

Also when visiting the releases here on [github](https://github.com/OpenZWave/open-zwave/releases), v1.5 is mentioned as a release. How can a development version be a release? Maybe I'm wrong but shouldn't a release be stable and not some sort of development ongoing release candidate?

_I see allot of people using OpenZwave and many of them are using the latest development version because the latest stable relatively old. And because it's easier for them because the versioning system is inconsistent and hard to understand._ 
Another example: [Domoticz](http://www.domoticz.com/forum/viewtopic.php?f=17&t=4909&sid=291471b4e922abce26547f65944773cc&start=40#p112652)

TimelessNL, 2017-01-17:
Any update on this?