OpenZWave/open-zwave #1094: Update/notification issues Qubino ZMNHBD1

ogroot, 2017-01-12:
Situation: 
OZW Version: 1.4-2327-g8b1973a-dirty
3 Qubino ZMNHBD1 switches do not report status changes of wall switch I2/Q2 properly: 

```
Case 1: [ Q1 input/switch = OFF ]
       Q2 input/switch change: off --> on  is reported to domoticz
       Q2 input/switch change: on --> off  is reported to domoticz
Case 2: [ Q1 input/switch = ON
       Q2 input/switch change: off --> on  is reported to domoticz
       Q2 input/switch change: on --> off  is NOT ALWAYS reported to domoticz (+/- 70% of the cases it is not) 
```
I upgraded to 1.4-2327-g8b1973a-dirty, because I saw many similar reports and several (MultiInstanceAssociation ) changes in the code base and XML config: 

**Other similar reports/issues:**
[(https://github.com/OpenZWave/open-zwave/issues/880)]
[https://github.com/OpenZWave/open-zwave/issues/862]
[https://github.com/OpenZWave/open-zwave/issues/857]
[https://github.com/OpenZWave/open-zwave/issues/857]
[https://github.com/Nico0084/open-zwave/commit/c49a955900af34589d808f584314694bee9bdc30]
**OZW Changes**
I saw several MultiInstanceAssociation related changes. E.g.:
cpp/src/Driver.h 
cpp/src/command_classes/MultiInstanceAssociation.cpp 
cpp/src/command_classes/MultiInstanceAssociation.h 
OZW Changes in config/qubino/ZMNHBDx.xml
```
Added: 
        <CommandClass id="37">
                <TriggerRefreshValue Genre="user" Index="0" Instance="1">
                        <RefreshClassValue CommandClass="37" RequestFlags="0" Index="0" Instance="2" />
                        <RefreshClassValue CommandClass="37" RequestFlags="0" Index="0" Instance="3" />
                </TriggerRefreshValue>
        </CommandClass>
Removed: 
        <!-- Map endpoints to instances --> <CommandClass id="96" mapping="endpoints" />
```
So, I upgraded to 1.4-2327-g8b1973a-dirty, which did **NOT** solve the problem. 
However, now frequent **timeouts** appeared (while z-wave mesh coverage is very good).

There is still something to improve with these double (Multi Instance) switches like ZMNHBD1 and FGS223.

YcKe, 2017-01-12:
Hi ogroot,

Have you already tried putting the association 1.1 on group 1, 2 and 3? This works for me.

ogroot, 2017-01-12:
Hi YcKe, 
Thanks for your suggestion. Yes, I added these associations. 
However, without success.

YcKe, 2017-01-12:
Hi ogroot,

What is the software version of your modules?

ogroot, 2017-01-12:
YcKe, they are build in currently, but as I remember they are version S4

YcKe, 2017-01-13:
Hi ogroot,

mine are of version s5. That might explain the difference. I agree with you that there is an issue that needs to be investigated.

Fishwaldo, 2017-02-17:
need full log files please.

jotakar, 2017-02-22:
There is a message from goap in OZW google groups
[https://groups.google.com/forum/#!topic/openzwave/355Vr3s4f00](url)

jotakar, 2017-02-23:
I have S4 and S5 versions of this device and I don't need  the triggerrefresh command in config file nor polling. 
My OZW version is  1.4-2334-gaf55dc8-dirty.
I hope this data be useful.

Fishwaldo, 2018-10-26:
This should be fixed according to the Mailing List Post

closed 2018-10-26 by Fishwaldo