OpenZWave/open-zwave #1099: double free or corruption during  OpenZWave::Manager::GetValueAsFloat causing crashes?

partofthething, 2017-01-16:
I've been having [trouble in an application that uses openzwave](https://github.com/home-assistant/home-assistant/issues/5272) with frequent segfault/SIGABRTs. They happen every day or so and always correspond to some activity in `OZW_Log.txt`. I built a debug environment and caught a few exceptions that were in openzwave (note I'm using the version wrapped by `python-openzwave`). 

I thought perhaps you could give me some insight into this issue. Is it pointing to an issue with openzwave? 

Some gdb output follows. The bug, whatever it is, mostly just corrupts the stack so tracebacks are usually impossible to get. This one did make some sense. 
```
Program received signal SIGABRT, Aborted.
0x76d55f70 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
56      ../nptl/sysdeps/unix/sysv/linux/raise.c: No such file or directory.
#0  0x76d55f70 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x76d57324 in __GI_abort () at abort.c:89
#2  0x76d91954 in __libc_message (do_abort=<optimized out>, fmt=0x76e476e8 "*** Error in `%s': %s: 0x%s ***\n")
    at ../sysdeps/posix/libc_fatal.c:175
#3  0x76d97b80 in malloc_printerr (action=1, str=0x76e47924 "double free or corruption (fasttop)", 
    ptr=<optimized out>) at malloc.c:4996
#4  0x76d98b24 in _int_free (av=<optimized out>, p=<optimized out>, have_lock=1870162464) at malloc.c:3840
#5  0x6f68d508 in OpenZWave::Manager::GetValueAsFloat(OpenZWave::ValueID const&, float*) ()
   from /opt/hass-dev/lib/python3.4/site-packages/libopenzwave-0.3.1-py3.4-linux-armv7l.egg/libopenzwave.cpython-34m.so
#6  0x6f66fbfc in __pyx_f_12libopenzwave_getValueFromType (
    __pyx_v_manager=0x6f5beea8 <std::string::_Rep::_S_empty_rep_storage>, __pyx_v_valueId=0x63363dd0)
    at src-lib/libopenzwave/libopenzwave.cpp:4151
#7  0x6f670768 in __pyx_pf_12libopenzwave_9PyManager_192getValue (__pyx_v_self=<optimized out>, 
    __pyx_v_id=<optimized out>) at src-lib/libopenzwave/libopenzwave.cpp:23098
#8  __pyx_pw_12libopenzwave_9PyManager_193getValue (__pyx_v_self=<optimized out>, __pyx_v_id=<optimized out>)
    at src-lib/libopenzwave/libopenzwave.cpp:23074
#9  0x0008b834 in PyEval_EvalFrameEx ()
Cannot access memory at address 0x1
#10 0x000a84a4 in ?? ()
Cannot access memory at address 0x1
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 
```

relevant `OZW_Log.txt` log (actually from a different crash, but hopefully related) below. Basically, the application is in the middle of polling node 8 for an update but in the middle of the poll, node 4 chimes in with an update. 
```
2017-01-15 22:12:03.303 Detail, 
2017-01-15 22:12:03.303 Detail, Node008, Removing current message
2017-01-15 22:12:03.303 Detail, Node008, Notification: ValueChanged
2017-01-15 22:12:08.306 Detail, Node008, Polling: COMMAND_CLASS_SWITCH_BINARY index = 0 instance = 1 (poll queue has 0 messages)
2017-01-15 22:12:08.306 Detail, Node008, Queuing (Poll) SwitchBinaryCmd_Get (Node=8): 0x01, 0x09, 0x00, 0x13, 0x08, 0x02, 0x25, 0x02, 0x25, 0x29, 0xc4
2017-01-15 22:12:08.309 Detail, 
2017-01-15 22:12:08.309 Info, Node008, Sending (Poll) message (Callback ID=0x29, Expected Reply=0x04) - SwitchBinaryCmd_Get (Node=8): 0x01, 0x09, 0x00, 0x13, 0x08, 0x02, 0x25, 0x02, 0x25, 0x29, 0xc4
2017-01-15 22:12:08.318 Detail, Node008,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2017-01-15 22:12:08.319 Detail, Node008,   ZW_SEND_DATA delivered to Z-Wave stack
2017-01-15 22:12:08.340 Detail, Node008,   Received: 0x01, 0x07, 0x00, 0x13, 0x29, 0x00, 0x00, 0x03, 0xc1
2017-01-15 22:12:08.340 Detail, Node008,   ZW_SEND_DATA Request with callback ID 0x29 received (expected 0x29)
2017-01-15 22:12:08.340 Info, Node008, Request RTT 30 Average Request RTT 30
2017-01-15 22:12:08.340 Detail,   Expected callbackId was received
2017-01-15 22:12:08.355 Detail, Node008,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x08, 0x03, 0x25, 0x03, 0xff, 0x20
2017-01-15 22:12:08.356 Detail, 
2017-01-15 22:12:08.356 Info, Node008, Response RTT 46 Average Response RTT 45
2017-01-15 22:12:08.356 Info, Node008, Received SwitchBinary report from node 8: level=On
2017-01-15 22:12:08.356 Detail, Node008, Refreshed Value: old value=true, new value=true, type=bool
2017-01-15 22:12:08.356 Detail, Node008, Changes to this value are not verified
2017-01-15 22:12:08.356 Detail, Node008,   Expected reply and command class was received
2017-01-15 22:12:08.356 Detail, Node008,   Message transaction complete
2017-01-15 22:12:08.357 Detail, 
2017-01-15 22:12:08.357 Detail, Node008, Removing current message
2017-01-15 22:12:08.357 Detail, Node008, Notification: ValueChanged
2017-01-15 22:12:08.444 Detail, Node004,   Received: 0x01, 0x14, 0x00, 0x04, 0x08, 0x04, 0x0e, 0x32, 0x02, 0x21, 0x64, 0x00, 0x06, 0x78, 0xbe, 0x00, 0x78, 0x00, 0x06, 0x78, 0xbe, 0xe0
2017-01-15 22:12:08.444 Detail, 
2017-01-15 22:12:08.444 Detail, Node004, Refreshed Value: old value=false, new value=false, type=bool
2017-01-15 22:12:08.444 Detail, Node004, Changes to this value are not verified
2017-01-15 22:12:08.444 Info, Node004, Received Meter report from node 4: Energy=424.126kWh
2017-01-15 22:12:08.445 Detail, Node004, Refreshed Value: old value=424.126, new value=424.126, type=decimal
2017-01-15 22:12:08.445 Detail, Node004, Changes to this value are not verified
2017-01-15 22:12:08.445 Info, Node004,     Previous value was 424.126kWh, received 120 seconds ago.
2017-01-15 22:12:08.445 Detail, Node004, Refreshed Value: old value=424.126, new value=424.126, type=decimal
2017-01-15 22:12:08.445 Detail, Node004, Changes to this value are not verified
2017-01-15 22:12:08.445 Detail, Node004, Refreshed Value: old value=120, new value=120, type=int
[... 26 lines cut ...]
2017-01-15 22:12:08.629 Detail, Node004, Refreshed Value: old value=0.000, new value=0.000, type=decimal
2017-01-15 22:12:08.629 Detail, Node004, Changes to this value are not verified
2017-01-15 22:12:08.629 Info, Node004,     Previous value was 424.126W, received 120 seconds ago.
2017-01-15 22:12:08.629 Detail, Node004, Refreshed Value: old value=424.126, new value=424.126, type=decimal
2017-01-15 22:12:08.629 Detail, Node004, Changes to this value are not verified
2017-01-15 22:12:08.629 Detail, Node004, Refreshed Value: old value=120, new value=120, type=int
2017-01-15 22:12:08.629 Detail, Node004, Changes to this value are not verified
2017-01-15 22:12:08.629 Detail, Node004, Notification: ValueChanged
2017-01-15 22:12:08.646 Detail, Node004, Notification: ValueChanged
2017-01-15 22:12:08.663 Detail, Node004, Notification: ValueChanged
2017-01-15 22:12:08.684 Detail, Node004, Notification: ValueChanged
2017-01-15 22:12:08.702 Detail, Node004,   Received: 0x01, 0x14, 0x00, 0x04, 0x00, 0x04, 0x0e, 0x32, 0x02, 0x21, 0x74, 0x00, 0x00, 0x00, 0x00, 0x00, 0x78, 0x00, 0x06, 0x78, 0xbe, 0x38
2017-01-15 22:12:08.703 Detail, 
2017-01-15 22:12:08.703 Detail, Node004, Refreshed Value: old value=false, new value=false, type=bool
2017-01-15 22:12:08.703 Detail, Node004, Changes to this value are not verified
2017-01-15 22:12:08.703 Info, Node004, Received Meter report from node 4: Power=0.000W
2017-01-15 22:12:08.703 Detail, Node004, Refreshed Value: old value=0.000, new value=0.000, type=decimal
2017-01-15 22:12:08.703 Detail, Node004, Changes to this value are not verified
2017-01-15 22:12:08.703 Info, Node004,     Previous value was 424.126W, received 120 seconds ago.
2017-01-15 22:12:08.703 Detail, Node004, Refreshed Value: old value=424.126, new value=424.126, type=decimal
2017-01-15 22:12:08.703 Detail, Node004, Changes to this value are not verified
2017-01-15 22:12:08.703 Detail, Node004, Refreshed Value: old value=120, new value=120, type=int
2017-01-15 22:12:08.704 Detail, Node004, Changes to this value are not verified
2017-01-15 22:12:08.704 Detail, Node004, Notification: ValueChanged
```

partofthething, 2017-02-05:
home-assistant is calling openzwave things from multiple threads but I'm not sure if it's done thread-safely or not. From what I understand (based on [this](https://groups.google.com/forum/#!topic/openzwave/UJFceAz07zE), openzwave access from multiple external threads must be handled in a way that things don't get called at the same time on the zwave network.

Fishwaldo, 2017-02-17:
Hard to say whats going on, but as its limited to just python-ozw users, I'd hesitate a guess its how the application/wrapper is talking to OZW:
1) Python-OpenZWave is passing in a unallocated pointer for the Call. In this case - need to audit the PYOZW code.

Would be best if you can build the OZW library with Debugging enabled (so we can get exact line numbers etc from OZW) to see what variable its hating...

I don't know how pyozw builds ozw, but esentially you want the make command to be:
`BUILD=debug make`

Fishwaldo, 2017-03-21:
Closing - No Feedback

closed 2017-03-21 by Fishwaldo