OpenZWave/open-zwave #1106: Fibaro FGS223 (Gen5) - no include possible

michapr, 2017-01-23:
Using 1.4-2347-g2d91fb1-dirty  (latest Domoticz version) the dual switch can not be included.
Have tried several times, multiple wakeups, factory reset, different controllers, secure and insecure inclusion.

Latest XML from GitHub was used.

Attached the log, I hope it help.
[fib_fgs223.txt](https://github.com/OpenZWave/open-zwave/files/723516/fib_fgs223.txt)

Michael

Bswanny, 2017-01-27:
Hi Michael,

On my first attempt two weeks ago, my FGS-223 was included, but switch 1 could not be controlled. The reading of the manual switch status and power consumed was ok.

After updating Domoticz from 3.6277 to 3.6518 and including again, all worked fine.

I run Domoticz on a linux pc with an AEON contoler USB stick. For testing and programming purpose i have made a temporary switchboard with a double switch an two lightbulbs. depending on you geographical location i could help you with this maybe.

Are u sure the FGS-223 is ok?

Bart

michapr, 2017-01-27:
Hi,

yes, I think FGS is ok - this were two different modules.
Have you other OZW version now - newer as  1.4-2347-g2d91fb1-dirty ?

It was prepared for customer installation - so I have replaced it with FGS 222 ...

Michael

michapr, 2017-01-29:
see http://www.domoticz.com/forum/viewtopic.php?f=6&t=15375 and
http://www.domoticz.com/forum/viewtopic.php?f=6&t=13209

Problem is there still existing, not sure why it is working at your installation...

nechry, 2017-01-29:
The question is, why isn't working at your installation, we are many users using thsi device on jeedom gateway without problems

martian, 2017-02-08:
I have had some trouble using FGS-223 from domoticz immediately after including the module, most notably on/off commands initiated from domoticz not arriving. But after restarting domoticz they seem to work fine.

michapr, 2017-02-12:
It's now clear what happen.

The source of this FGS223 problems seems to be in a Fibaro firmware "issue".
Device is reporting that it is supporting "switch multilevel" - but it is not right,... it is a binary switch only.
In Fibaro FGS config XML was inserted that's why:

```
    <!-- COMMAND_CLASS_SWITCH_MULTILEVEL. This class is in the list reported by the device, but it does not respond to requests -->
    <CommandClass id="38" getsupported="false"/>

```
In Domoticz config XML ist still this CC inside, with getsupported="false" - but seems to be active anyway.
So sometimes the switching command want to use the multilevel CC - what will be wrong and an eror will come up.

It was reported to Domoticz development, so can be closed here.

closed 2017-02-12 by michapr

nechry, 2017-02-12:
I think the COMMAND_CLASS_SWITCH_MULTILEVEL is reported only for Association purposes