OpenZWave/open-zwave #1119: Unable to Pair - FGBHMS-001

japita-se, 2017-02-01:
Unable to pair the Motion FGBHMS-001.
When running the add node operation, and following the instruction of FIBARO (holding down the press button) nothing happens.

nechry, 2017-02-01:
Always perform a exclude before a include. Check log for remove nodeid 0 after do the include procedure.

Beware to do exactly the click step, one more and you stop the process

nechry, 2017-02-01:
Your issue is not related to openzwave librairy please use mailling list instead.

closed 2017-02-01 by nechry

openzwave-dev-watcher, 2017-02-02:
Which mailing list?
This is not supposed to be the mailing list?
Regards

nechry, 2017-02-02:
if you look at github ozw home page:
https://github.com/OpenZWave/open-zwave

it's explained

So the malling list is her: 
https://groups.google.com/forum/#!forum/openzwave

openzwave-dev-watcher, 2017-02-03:
The same I writing in :)