OpenZWave/open-zwave #1129: COMMAND_CLASS_ALARM version 2, Type/Level

gizmocuz, 2017-02-11:
In the current implementation when receiving a version 2 alarm we send:

type = data[1]
level = data[2]

That causes it to always be zero.
I think for version 2, this should be changed to:

type = data[5]
level = data[6]

Also the string 'type' is based on data[5]

Fishwaldo, 2017-02-17:
Doing that will break Alarm Version 1 devices. Anyway, a rewrite is being done in the Dev branch for the Notification Class (renamed Alarm Class)

closed 2017-02-17 by Fishwaldo

gizmocuz, 2017-02-17:
Not entirely true if you use the if/then

```
		if( GetVersion() == 1 )
		{
			Log::Write( LogLevel_Info, GetNodeId(), "Received Alarm report: type=%d, level=%d", _data[1], _data[2] );
		}
		else
		{
			string alarm_type =  ( _data[5] < Alarm_Count ) ? c_alarmTypeName[_data[5]] : "Unknown type";

			Log::Write( LogLevel_Info, GetNodeId(), "Received Alarm report: type=%d, level=%d, sensorSrcID=%d, type:%s event:%d, status=%d",
							_data[1], _data[2], _data[3], alarm_type.c_str(), _data[6], _data[4] );
		}
```

You can see here, for version '2', that BYTE 5/6 is used