OpenZWave/open-zwave #1136: MutexImpl::Unlock error 0 (1) error every couple of days

kylehendricks, 2017-02-15:
### Issue Guidelines

I have been using OpenZWave via Home Assistant for a while and have started to get a `MutexImpl::Unlock error 0 (1)` error every few days and need to end up restarting home assistant.

Seems to be `pthread_mutex_unlock` returning an EPERM error which either means the current thread never locked the mutex or the mutex is already unlocked.  Could this be an issue with the usage of the OpenZWave python library in Home Assistant?

Not sure if it matters but I have Home Assistant running in a docker container.

tobiasgraf, 2017-02-15:
This sounds like this, doesn't it? https://github.com/OpenZWave/open-zwave/pull/1105

kylehendricks, 2017-02-15:
Ah yes.  That does seem like the same thing.

Fishwaldo, 2017-02-17:
PR is applied, and some additional code to detect unbalanced Lock/Unlock calls added. Please test.

kylehendricks, 2017-02-17:
Got it running now.  Probably will have to wait a week or so to really see if it's still happening.

Fishwaldo, 2017-02-22:
hows it going?

kylehendricks, 2017-02-22:
So far so good!

Fishwaldo, 2017-03-21:
Thanks. Closing.

closed 2017-03-21 by Fishwaldo