OpenZWave/open-zwave #1171: Cannot find endpoint map to instance for Command Class COMMAND_CLASS_CENTRAL_SCENE endpoint 0

AlCalzone, 2017-03-11:
I'm trying to get my new Aeotec WallMote to work and encounter the following error in OZW_Log.txt whenever I touch the device:
`Error, Node002, Cannot find endpoint map to instance for Command Class COMMAND_CLASS_CENTRAL_SCENE endpoint 0`

I've installed OZW over the ioBroker adapter which afaik downloads and compiles the newest version of OZW on each install.
[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/835698/OZW_Log.txt) is attached, any idea how to fix this?

nechry, 2017-03-11:
You must use dev branch for use CC CentralScene

AlCalzone, 2017-03-11:
Any idea when that is going to be included in the master branch? Is using the dev branch going to work with node-openzwave-shared or does that package need changes?

Anyways, I'll talk to the iobroker devs and see what can be done on their side.

AlCalzone, 2017-03-14:
I finally managed to get the Dev branch to work with openzwave-shared on my Raspberry. The OZW_Log.txt now reads `OpenZwave Version 1.5.0 Starting Up`. 

However the above error still persists.

Fishwaldo, 2017-03-21:
See #1125

closed 2017-03-21 by Fishwaldo