OpenZWave/open-zwave #1172: Error building Driver.cpp

AlCalzone, 2017-03-11:
I'm trying to build the Dev branch on a Raspberry Pi, but I'm getting an error. Here are the steps I took:

1. Download the zip archive:
`sudo curl -L -O https://github.com/OpenZWave/open-zwave/archive/Dev.zip`
2. Unzip it
3. build/install:
`sudo make && sudo make install`

This leads to the following error:

    Building OpenZWave Version
    Building Driver.o
    /opt/open-zwave-Dev/cpp/src/Driver.cpp: In member function 'void OpenZWave::Driver::processConfigRevision(OpenZWave::DNSLookup*)':
    /opt/open-zwave-Dev/cpp/src/Driver.cpp:7098:30: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
         if (m_mfs->getRevision() < atol(result->result.c_str())) {
                                  ^
    cc1plus: all warnings being treated as errors
    /opt/open-zwave-Dev/cpp/build/support.mk:130: recipe for target '/opt/open-zwave-Dev/.lib/Driver.o' failed
    make[1]: *** [/opt/open-zwave-Dev/.lib/Driver.o] Error 1
    make[1]: Leaving directory '/opt/open-zwave-Dev/cpp/build'
    Makefile:20: recipe for target 'all' failed
    make: *** [all] Error 2

Is there something I can do on my end to get this to work?

hvbommel, 2017-03-13:
add the casts as shown below and it should build
if ((uint)m_mfs->getRevision() < (uint)atol(result->result.c_str())) {

AlCalzone, 2017-03-13:
Build completed, thanks!

commit e0bca3ff (2017-03-21): Fix #1172 - Compile Error
closed 2017-03-21 by Fishwaldo