OpenZWave/open-zwave #1179: 2Gig CT101 (Iris) Humidity Reporting Bug

elbarsal, 2017-03-15:
I recently purchased two CT101 (Iris) thermostats to follow up an initial CT101 (non-Iris) thermostat that I had successfully started using.

There are actually multiple issues with the behavior of the thermostats (both versions) that can be worked around, but there is a larger issue in the Iris version reporting humidity.

On polling for humidity, the thermostat responds with a MultiChannelEncap message, but the response uses an incorrect sub-command (0x06 for MultiInstanceEncap instead of 0x0d for MultiChannelEncap), resulting in an inability to parse the reply, as the length and positions of the data are different.

I implemented a work-around in my local version of the code that attempts to parse a failed 0x06 message in HandleMultiInstanceEncap as a 0x0d by calling HandleMultiChannelEncap with the same data, and it works nicely.

I'm not sure if this is the right way or best place to fix this. The firmware on the CT101 can't be updated according to RadioThermostat support - so a work-around seems like the only solution.

Where is the right place / what is the right way to implement this workaround? I would be happy to supply mine, or make modifications elsewhere in the code, if there is a better place.

Fishwaldo, 2017-04-05:
Can you provide logs (after removing ozwconfig_*.xml etc) and restarting......

elbarsal, 2017-04-19:
OK will do, sorry for the delay.

elbarsal, 2017-04-19:
Here is the log file. Node 21 is a Radio Thermostat version of the device (I had to add its config, but it works well) and Nodes 22 and 24 are the Iris version - the ones that misreport humidity.

You can see at 11:38:50.704 there is a correctly received Relative Humidity from Node 21.

For Node 22, at 11:38:54.024 it appears that the device replies to the request with the wrong sub-command, which gets parsed with no error, but does not result in getting any data.

In my experimentation, I modified the parsing code such that if it doesn't find the expected sub-command, it tries the other, and that successfully gets the data.

[OZW_Log_Apr19_New.txt](https://github.com/OpenZWave/open-zwave/files/935781/OZW_Log_Apr19_New.txt)

elbarsal, 2017-06-05:
Is there anything else you need? I can give you the text of my modification if needed.

jeschmidt, 2017-10-21:
See parallel thread for Openhab here: https://community.openhab.org/t/ct101-thermostat-not-supported-in-openhab2-z-wave-binding/14384/33

THANK YOU very much to elbarsal for giving me the details necessary to get a workaround running successfully.  Let me know if there are other tweaks you have made.

danbowkley, 2018-09-04:
Has there been any progress on this? @elbarsal could you please post your modification? I'm trying to get my CT101 to report humidity to Home Assistant, same issue...

elbarsal, 2018-10-01:
> Has there been any progress on this? @elbarsal could you please post your modification? I'm trying to get my CT101 to report humidity to Home Assistant, same issue...

@danbowkley - 

Haven't seen any movement on this. Anyway, the change was pretty simple. In the function  MultiInstance::HandleMultiInstanceEncap in file command_classes/MultiInstance.cpp find the if statement 

` if( CommandClass* pCommandClass = node->GetCommandClass( commandClassId ) )
                {
                        Log::Write( LogLevel_Info, GetNodeId(), "Received a MultiInstanceEncap from node %d, instance %d, for Command Class %s", GetNodeId(), instance, pCommandClass->GetCommandClassName().c_str() );
                        pCommandClass->ReceivedCntIncr();
                        pCommandClass->HandleMsg( &_data[3], _length-3, instance );
                }`

and add an else to it:

` else
                {
                       Log::Write( LogLevel_Warning, GetNodeId(), "Received invalid MultiInstanceReport from node %d. Attempting to process as MultiChannel", GetNodeId());
                       HandleMultiChannelEncap( _data, _length );
                }
`

What this does is if a MultiInstanceEncap is received but cannot be processed, attempt to process it as a MultiChannelEncap (which is what the data actually is, even though the sub-command says otherwise).

Let me know if that does the trick (I hope it does!)

Fishwaldo, 2018-10-02:
Ok. I've committed this fix.

closed 2018-10-02 by Fishwaldo

chilicheech, 2018-10-11:
@Fishwaldo can also put this change in the Dev branch? It will help me with my device as well.

Cheers,

danbowkley, 2018-10-12:
Well, I tried..alas, I'm running Home Assistant in a Python venv and have basically no clue how to go about updating openzwave within that. Can anyone please give me the total newb's guide to not completely demolishing my server? I'm running Ubuntu 16.04 (I'd love to upgrade, but it's also running my UniFi NVR which relies on an older MongoDB) and just upgraded Hass today.

dummptyhummpty, 2018-10-12:
Would the steps [here](https://www.reddit.com/r/homeassistant/comments/65kz2x/proper_way_to_update_open_zwave/) work?

danbowkley, 2018-10-14:
Well, I'm blaming my bald spot on Zwave.

I did get it built properly; I just made and installed it outside the venv and used the python-openzwave "shared" flavor. I'd tried that previously and when it didn't get the humidity sensor I'd assumed that was because it was still using its own openzwave libraries...and I was wrong.

So what's happening now, is when I stop the network and delete the zwcfg_xxxxx.xml file and restart the network, I get the thermostat back but minus the humidity sensor and with a new "indicator" sensor. The CT-101 does have a filter change indicator, and I'm thinking that's probably what that sensor is for. But still no humidity.

While dredging through the nether regions of the web, I ran across this: https://groups.google.com/forum/#!topic/openremotecommunity/MH18-CsN8p8 (next to last message) and think this might have something to do with it. What's the elegant way to set an association within Openzwave? Or is this a red herring? My eyes started to bleed slightly while trying to grok the zwcfg xml file...

danbowkley, 2018-10-17:
Twas a red herring.

Turns out Home Assistant requires python_openzwave==0.4.9, so I had to install the Git flavor of that specific version. Once I got that sorted out, yes indeed the humidity sensor does now work. 

Thank you all!