OpenZWave/open-zwave #1183: Aeotec ZW112 not detected

KRens, 2017-03-19:
I have a ZW112 Aeotec triangle, but it cannot be properly detected.
The product name stays empty and I can't commands to it.

An older version (doorwindow.xml) did work.
The newer **/config/aeotec/zw112.xml** file does not work.

Here you can find the diff working vs non-working:
https://www.diffchecker.com/S5bP3AAL

nechry, 2017-03-19:
I have this device and is working as expected with the zw112.xml file. Can you more specific about your issue?

KRens, 2017-03-19:
There are 3 product types of this device, maybe you have a different type?

<Product type="0002" id="0070" name="ZW112 Door Window Sensor 6"
<Product type="0102" id="0070" name="ZW112 Door Window Sensor 6"
<Product type="0202" id="0070" name="ZW112 Door Window Sensor 6"

The issue is that the product name is not recognized (stays empty) and that I do not get any events from the device when I remove the 2 magnets.
All other devices are working properly.

If I replace the xml with the old doorwindow.xml it's working again.

nechry, 2017-03-19:
I have the type 0002 for Europe region. The 2 others type share same configuration, but for different region. I wrote this configuration file. The old file didn't match the configuration parameters form Aeotec.

nechry, 2017-03-19:
Did you include in secure way or in unsecured way? Try first in unsecure way

KRens, 2017-03-19:
Ok, I'm using Europe region as well.
Any tips on how to investigate which parameter or field is causing the issue?
I have 3 Aeotec triangles, none of them get detected properly.

nechry, 2017-03-19:
Just to be sure, you tell if you replace the contents of the zw112.xml with something else the device is included successfully? But with this file the latest version your device productname don't show up?

nechry, 2017-03-19:
Be sure the content of your zwcfg file is same as
https://github.com/OpenZWave/open-zwave/blob/master/config/aeotec/zw112.xml

KRens, 2017-03-19:
All devices have already been included.
I just updated the config folder. 

Since the update, these 3 devices are not recognized anymore.
I used the latest version from github for testing the new config.

nechry, 2017-03-19:
So perform a exclude and include it again

nechry, 2017-03-19:
Allso Check for error in your log file

nechry, 2017-03-19:
I just take a look, my zw112 have been included in December, with the lasted xml of November 2016.

KRens, 2017-03-22:
Ok, excluding and including solved the issue.
Strange.
Thank you for the suggestion.

closed 2017-03-22 by KRens