OpenZWave/open-zwave #1186: Fibaro Button (FGPB-101) Problems

ghost72, 2017-03-21:
Hello

I try to get run the Fibaro Button FGPB-101. I see successful message in the logfile when I wake up the button by pressing it 4 time.
But when I press the button one, two or three times, i get always following error message for each event:
ApplicationCommandHandler - Unhandled Command Class 0x60

So nothing happens when i push the button, only wake-up (4 times pushing) is working!

Does someone have an idea how to solve this problem and get the button running?

Many thanks for your help!
Kind regards
Roger

Some more log informations:

2017-03-21 23:36:05.362 Detail, Node025,   Received: 0x01, 0x0f, 0x00, 0x04, 0x00, 0x19, 0x09, 0x60, 0x0d, 0x00, 0x01, 0x5b, 0x03, 0x69, 0x00, 0x01, 0xb8
2017-03-21 23:36:05.363 Detail, 
2017-03-21 23:36:05.363 Info, Node025, ApplicationCommandHandler - Unhandled Command Class 0x60
2017-03-21 23:36:05.519 Detail, Node025,   Received: 0x01, 0x0d, 0x00, 0x04, 0x00, 0x19, 0x07, 0x60, 0x0d, 0x00, 0x01, 0x20, 0x01, 0x00, 0xa5
2017-03-21 23:36:05.519 Detail, 
2017-03-21 23:36:05.519 Info, Node025, ApplicationCommandHandler - Unhandled Command Class 0x60

ZyberSE, 2017-03-21:
Hi, FGPB-101 uses Central Scene commands which isn't yet supported by OZW. There is a PR for it here though and afaik it shall be included in the 1.6 release.

https://github.com/OpenZWave/open-zwave/pull/1125

Fishwaldo, 2017-04-05:
Closing as this is pending #1125

closed 2017-04-05 by Fishwaldo

KptnKMan, 2017-04-05:
@ghost72 I also have a FGPB-101 which I have been trying to get working for months without any luck.

This is all related to the CentralScene issue in OZW, which seems to be persisting without end.

As far as I'm aware this issue goes pretty far back to #367 (which is still open, opened March 2015)
There was also a PR #1124 that was to stage #993 
As @ZyberSE and @Fishwaldo mentioned there was there is #1125 (continuing #993 + #1124)
The PR #993 was merged/closed today referencing #1125 so lets hope this can finally be put to rest.

As for #367 I have no idea when that will be settled.