OpenZWave/open-zwave #1188: Heiman gas sensor HS1CG - alarm not detected

michapr, 2017-03-23:
HS1CG is a notification device - if alarm is given, a notification will be sent.

In OZW-Log will be seen the status change, but "Unknown type event 0" / "Unknown type event 2" only.

Not sure how it can be fixed.

`2017-03-23 08:26:28.837 Info, Node010, Received Alarm report: type=0, level=0, sensorSrcID=0, type:Unknown type event:2, status=255`

`2017-03-23 08:26:30.164 Info, Node010, Received Alarm report: type=0, level=0, sensorSrcID=0, type:Unknown type event:0, status=255`

Attached two reports - inclusion (to check the CC) and alarm report.

Thanks!
Michael

[HS1CG_notification_report.txt](https://github.com/OpenZWave/open-zwave/files/863978/HS1CG_notification_report.txt)
[HS1CG_inclusion_report.txt](https://github.com/OpenZWave/open-zwave/files/863985/HS1CG_inclusion_report.txt)

Fishwaldo, 2018-10-26:
This should be fixed now

closed 2018-10-26 by Fishwaldo

mike02spb, 2020-03-23:
Hello! I have same problem! My gas sensor has only 3 entities: "alarm type", "alarm level" and "sourcenodeid". And there is no binary_sensor.
I've insert lines with binary_sensor from heiman smoke sensor, and 2 more entities are appear: "sensor" (that must change state) and "gas alarm", but they are didn't change state when gas test is pushed.
Tell me please, what exactly I need to do to fix this problem? I really need this sensor!
Thank you!

myGithub-Markus, 2020-05-24:
Hello! I have also the same problem! Only this 3 entities. But the entities alarm level and type are always 0 zero. Only in the OZW-Log you can see this to changes
type:Unknown type event:2
type:Unknown type event:0