OpenZWave/open-zwave #1190: AeonLabs meters start reporting negative kWh?

ado464, 2017-03-29:
Hello all, the meters were ok until they hit a 7th digit number and then somehow reversed the counting. Can someone check this in openzwave? Thanks. Attaching OZW_Log.txt

[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/878181/OZW_Log.txt)

tobiasgraf, 2017-04-20:
The Meter Value is a signed value by definition. Singed values are used to also display negative numbers. You can read about how it works here: https://en.wikipedia.org/wiki/Two%27s_complement  
It uses 4 bytes (32bit) wit ha precision of 3 (10^3), so the maximum positive Value this can have is: 2 147 483,647 kWh.

OZW is acting according to the speccs. Sooo, i guess you just reached (or surpassed) the maximum Value the device can measure.

Out of curiosity, what are you measuring there?

p0fi, 2017-04-20:
Who are you? Elon Musk?

ado464, 2017-04-20:
Then the limit of these meters has been reached. Is there a simple command to reset the kWh counter? Hopefully a command that is easily run through a bash script and on all devices ;)

The measuring point is a big transformation station ;)

Thanks for all the help Tobias :)

openzwave-dev-watcher, 2017-04-21:
Probably the current transformer clamps are used in the wrong direction so 
the cumulated energy is decreasing. Once you are at 0 it further decreases 
into negative values.

Solution 1:
Check the current transformer clamps and turn them

Solution 2:
Set configuration Reverse Clamping to false, this ignores the direction of 
current in the clamps and only increase the energy counter

ado464, 2017-08-02:
Hello sorry for late reply, the reverse clamping does not work on the kWh, only on the Wattage. Is there a way to implement reset kwh into the xml code? I know you can reset the counter, i have seen it done on the Razberry interface. Help is hugely appreciated :)

best regards,
Ado

ado464, 2017-08-03:
Hello, in meter.cpp file there is this code of string

´´´
//-----------------------------------------------------------------------------
// <Meter::SetValue>
// Set the device's scale, or reset its accumulated values.
//-----------------------------------------------------------------------------
bool Meter::SetValue
(
	Value const& _value
)
{
	if( MeterIndex_Reset == _value.GetID().GetIndex() )
	{
		ValueButton const* button = static_cast<ValueButton const*>(&_value);
		if( button->IsPressed() )
		{
			Msg* msg = new Msg( "MeterCmd_Reset", GetNodeId(), REQUEST, FUNC_ID_ZW_SEND_DATA, true );
			msg->SetInstance( this, _value.GetID().GetInstance() );
			msg->Append( GetNodeId() );
			msg->Append( 2 );
			msg->Append( GetCommandClassId() );
			msg->Append( MeterCmd_Reset );
			msg->Append( GetDriver()->GetTransmitOptions() );
			GetDriver()->SendMsg( msg, Driver::MsgQueue_Send );
			return true;
		}
	}

	return false;
}
´´´

How could we make this executable with a bash command, reseting the accumulated kWh in value2? Anyone?

best regards,
Ado

tobiasgraf, 2017-08-03:
You can use the manager method `Manager::PressButton` to press the reset button.

something like this could work (did not test this), **# this would reset all meters it can find!**
Using the example Main.cpp as base

```
// Since the configuration file contains command class information that is only 
// known after the nodes on the network are queried, wait until all of the nodes 
// on the network have been queried (at least the "listening" ones) before
// writing the configuration file.  (Maybe write again after sleeping nodes have
// been queried as well.)
if( !g_initFailed )
    {
    pthread_mutex_lock( &g_criticalSection );
    // Loop through all the nodes 
    for( list<NodeInfo*>::iterator it = g_nodes.begin(); it != g_nodes.end(); ++it )
    {
        NodeInfo* nodeInfo = *it;

        // skip the controller (most likely node 1)
        if( nodeInfo->m_nodeId == 1) continue;

        printf("NodeID: %d \n ", nodeInfo->m_nodeId);
        printf("\t NodeName: %s \n ", Manager::Get()->GetNodeName(nodeInfo->m_homeId,nodeInfo->m_nodeId).c_str());
        printf("\t ManufacturerName: %s \n ", Manager::Get()->GetNodeManufacturerName(nodeInfo->m_homeId,nodeInfo->m_nodeId).c_str());
        printf("\t NodeProductName: %s \n ", Manager::Get()->GetNodeProductName(nodeInfo->m_homeId,nodeInfo->m_nodeId).c_str());

        // loop through all the valueIDs this nodes has, if a meter with a reset button is found, press it.
        for( list<ValueID>::iterator it2 = nodeInfo->m_values.begin(); it2 != nodeInfo->m_values.end(); ++it2 )
        {
            ValueID v = *it2;
            printf("\t ValueLabel: %s \n", Manager::Get()->GetValueLabel(v).c_str());
            printf("\t\t CommandClass: 0x%02X (%d) \n", v.GetCommandClassId(),v.GetCommandClassId());
            printf("\t\t Genre: %d \n", v.GetGenre());
            printf("\t\t instance: %d \n", v.GetInstance());
            printf("\t\t index: %d \n", v.GetIndex());
            printf("\t\t ValueType: %d \n", v.GetType());
            printf("\t\t ValueHelp: %s \n", Manager::Get()->GetValueHelp(v).c_str());
            printf("\t\t ValueUnits: %s \n", Manager::Get()->GetValueUnits(v).c_str());
            printf("\t\t ValueMin: %d \n", Manager::Get()->GetValueMin(v));
            printf("\t\t ValueMax: %d \n", Manager::Get()->GetValueMax(v));

            if ( v.GetCommandClassId() == 0x32 && v.GetIndex() = 33 && v.GetType() == ValueID::ValueType_Button)
            {
                // this will press the button!
                //Manager::Get()->PressButton(v);
            }
        }
    }
}
```

ado464, 2017-08-03:
Hello tobias, could possibly work. But how would i execute this? I am guessing it has to be placed somewhere specific?

thanks for the support!
Ado

tobiasgraf, 2017-08-03:
Like I said, this uses the [example](https://github.com/OpenZWave/open-zwave/blob/master/cpp/examples/MinOZW/Main.cpp) as a base. 
So, copy the code above into the [Main.cpp](https://github.com/OpenZWave/open-zwave/blob/master/cpp/examples/MinOZW/Main.cpp#L321), [fire up your compiler](https://github.com/OpenZWave/open-zwave/blob/master/INSTALL) and run the MinOZW example.

ado464, 2017-08-03:
Thanks again tobias, will do!

best regards,
Ado

ado464, 2017-08-09:
Succesfully implemented with jsonrpc communication!

closed 2017-08-09 by ado464