OpenZWave/open-zwave #1199: pthread_join blocks sometimes

sayegh, 2017-04-01:
When terminating OZW esp. during the startup query process, we experienced that its thread doesn't terminate (on Darwin and Linux). The reason was that `SerialControllerImpl::Read` didn't return, hence the thread could not terminate.

There are probably better ways to solve this, but that's how we did it:

In **platform/Thread.h**:
Make `virtual bool IsSignalled()` public.

In **platform/unix/SerialControllerImpl.cpp**, `SerialControllerImpl::Read`:

Add this after the select() system call:

```
if (m_pThread->IsSignalled())
  return ;
```

So far it works well.

sayegh, 2017-04-05:
Thanks for your reply, and there probably is something else going on, it's just what I found. 

Read() just never leaves the while(1) loop. Maybe a more detailed description helps: Our application auto-detects when a Z-Wave USB adapter is attached/detached, and then resp. starts/stops OZW. The problem usually occurs when I attach the USB adapter, and don't wait for the OZW initialisation to finish but remove the adapter (causing another thread to shut down OZW). Also, the behaviour is non-deterministic, though it happens most of the time.

The few lines of code above solved the issue for us and works as a safeguard. We just wanted to share in case someone else has similar issues.

Fishwaldo, 2017-06-29:
> Did some more digging. Possibly the Linux problem was just a coincidence. I found quite some complaints that pthread_cancel() doesn't work on Darwin. However, maybe it's still not the worst idea to replace while(1) with with while(!m_pThread->IsSignalled())... what do you think? Do you see negative implications?

Sounds good. Can you do a PR against the Dev tree for this?

sayegh, 2017-07-04:
Sorry - I'm now familiar with how to create a pull request (as you can see). Don't want to mess up anything - so a little help would be great.

xbeaudouin, 2017-08-23:
@sayegh If you have a patch (diff -u file.c.old file.c > file.diff) I can do the PR for you.

Fishwaldo, 2019-05-13:
Closing as we didnt get a PR...

closed 2019-05-13 by Fishwaldo