OpenZWave/open-zwave #1202: Missing unsolicited report configuration from Enerwave ZWN-RSM1 Plus

cexshun, 2017-04-02:
The docs from Enerwave detail the configuration for unsolicited reporting:

> Unsolicited Report Configuration ZWN-RSM1-PLUS can send unsolicited status reports to the primary controller (Node ID 0x1) when the switch is toggled if the controller is designed as a gateway. If your controller is not a gateway or does not need the status reported or you
> think it could confuse your Z-Wave network, you can use Command_Class_Configuration to disable this function. By
> default this function is disabled.
> Configuration Details: 
> − Parameter number: 3 
> − Parameter size: 1 
> − Parameter value: 0.........................Disabled
>  1.............Send Broadcast 

Without this configuration, if the relay is manually switched, the zwave status does not update. Is it possible to have this configuration added?

nechry, 2017-04-03:
We don't have any definition for this device please provide useful details if you want a new configuration file. see more https://github.com/OpenZWave/open-zwave/wiki/Adding-Devices

dnschneid, 2017-05-07:
Isn't that https://github.com/OpenZWave/open-zwave/blob/master/config/enerwave/zwnrsm1plus.xml ?  It does seem to be missing that config item, which is described in the manual linked to by the file.

dnschneid, 2017-05-07:
Try https://github.com/dnschneid/open-zwave/commit/fc6c1a1398588c497c8aac27b4ec00ec80eae43a ?  I'm not seeing any broadcast messages after sending the config, but maybe I'm doing something wrong.

cexshun, 2017-05-07:
> Try dnschneid@fc6c1a1

Tried that, and I tried adding something similar by editing the XML myself.
Doesn't work. I've been in contact with the manufacturer, and they claim
it's an incompatibility with OpenZWave as it works fine in Wink/Smart
Things.

nechry, 2017-12-26:
ping!

nordicblue, 2018-01-30:
Anyone ever figure out a solution to this issue yet? I am experiencing the same and having to manually poll the device to get its status, when the switch is turned on at the switch.

cexshun, 2018-01-30:
Nope. And it looks like they have been discontinued, so I doubt any further development will take place.  The manufacturer tech support was unable to help after weeks of emailing back and forth. But the SmartThings folks seemed to have gotten it working through a custom handler. Unfortunately when I attempted to port their fix, it still does not work with OpenZWave.

nordicblue, 2018-01-30:
Well, that's a bummer. Thanks for the update.

Fishwaldo, 2018-10-26:
Closing due to lack of updates

closed 2018-10-26 by Fishwaldo