OpenZWave/open-zwave #1205: Security 2 Framework

danieldeusing, 2017-04-11:
Hello folks,

I just want to clarify if this update has any relevance for open-zwave:

https://castercomm.com/pressrelease/mandatory-security-implementation-z-wave-certified-iot-devices-takes-effect-today/

Kind regards,
Daniel

Fishwaldo, 2017-04-12:
its been coming for a while, but so far I've not seen any actual devices on the market that support S2.

Fishwaldo, 2017-04-12:
Hi,
This is a issue tracker and not a support forum. 

Please use our Forums for Questions regarding OpenZWave: 
https://groups.google.com/forum/#!forum/openzwave

Thanks

closed 2017-04-12 by Fishwaldo

aromanro, 2018-12-12:
Just in case others are looking for this info, as far as I can tell from looking over the sources and documentation, no, S2 is not supported right now. It's quite a nuisance to implement (I know, I'm working on it), just the keys exchange is described with 30 steps in the documentation. The documentation (the public one, anyway) is a nightmare, you don't have for example test vectors to check your code results against some known results.

NicSil, 2019-02-28:
Why is the issue closed? If not implemented, the issue belongs...

enobrev, 2019-10-23:
I imagine this would count more as a feature request than a support request. 

While there may not have been any S2 products on the market when this ticket was created, there definitely are now.  Are there plans now or in the near future for S2 support.