OpenZWave/open-zwave #1209: Seeing blank zwave devices/nodes

jsg4, 2017-04-15:
I recently tried to replace a couple devices, and afterwards, my network started to go a little haywire. Some existing unrelated devices were lost and I had to exclude them and readd them, but even more strange is I see some blank devices in the logs now and when I add new devices, some nodes are skipped because of these blank devices.

Is there anyway to fix this and rebuild so to speak the network without having to repair everything? It seems to work fine now, I also deleted the xml file and let it rebuild itself.

jsg4, 2017-04-18:
Here is part of the log for example.

https://pastebin.com/6SBe53sR

Fishwaldo, 2017-05-18:
Thats just a logfile issue. those Nodes don't really exist, its just the way the Log Code is handling node 0 or messages that are not destined for a actual device.

closed 2017-05-18 by Fishwaldo