OpenZWave/open-zwave #1217: Support for missing coolcam Neo types

Luigi8723, 2017-05-05:
I would like to use the following type of product in my domoticz system, but there is no config file available.
the product i would like to use are:

The "NAS-RC01Z"
[this one on Aliexpress](https://nl.aliexpress.com/store/product/2016-New-Z-Wave-PLUS-Product-Smart-Home-Automation-Wireless-Wifi-Remote-Controller-SOS-NAS-RC01Z/1805523_32624196163.html)
 
And the "NAS-SC01Z-2"
[this one on Aliexpress](https://nl.aliexpress.com/item/Smart-Home-Z-wave-Wall-Light-Switch-Home-Automation-Z-Wave-Wireless-Smart-Remote-Control-Light/32795668194.html?spm=2114.48010308.4.44.6Lxajd)

Could some one help me out woth this one?
Thanks

nechry, 2017-05-05:
Without the products type and id I can't do nothing.

Luigi8723, 2017-05-05:
ah oke i am new with this so,
I have found the codes for the wall switch "NAS-SC01Z-2"
Manufacture ID:    0x0258
Product ID:       0x008B (US)   
Product ID:       0x108B (EU)
[Neo CoolCam Z-wave switch manual_Optimize.pdf](https://github.com/OpenZWave/open-zwave/files/980008/Neo.CoolCam.Z-wave.switch.manual_Optimize.pdf)

the remote is no where to find.
So if your can help me only with the wall switch i am happy

Fishwaldo, 2017-05-08:
https://github.com/OpenZWave/open-zwave/wiki/Adding-Devices

nechry, 2017-05-18:
@Luigi8723 I still don't have the product Type.
We need the Manufacture ID, Product Type and Product ID.

Do you have the device in hand ?

Luigi8723, 2017-05-22:
Sorry for my late response, i was in vacation 

@nechry No i don't have the device here. I wanted to be sure if it will work before i will order.
but if were are missing this data it is not possible to get it working in open z-wave i assume, right?
if so we can close the issue, because i have no further data.

nechry, 2017-05-22:
You must have the device, and we can try to get the device work. but we can't be sure he will work.

It's not just a question of products datas, the product type and id allow to set de product name and add device specific configuration.

closed 2017-05-22 by nechry

syphernl, 2017-07-26:
I have just finished installing a NAS-SC01ZE. This is the Touch Wall Switch Single (EU version)

Information I obtained from the logs:
```
Neo CoolCam Unknown: type=0003, id=108c Switch
```

```
Node Identity Codes: 0258:0003:108c
```

@nechry Does this help? If any further information is required, please let me know!