OpenZWave/open-zwave #1221: Config Parameter "value changed"

japita-se, 2017-05-15:
I try to set a parameter by using setValue.
I supposed that when the device receive this message, it will reply with an event "value changed".
But, tt seems that for some devices this does not happens.
For example the Door/Windows of Aeotec DSB29, if I send "17" for class "121" (Binay Report), it will behave correctly even if no value changed events has been fired.

Fishwaldo, 2017-05-18:
Can you provide log files. all ValueID's should report ValueChanged if in fact it is changed.

Fishwaldo, 2017-05-29:
ping...

japita-se, 2017-06-08:
Actually, it does not work at all.
We are trying to set the configuration parameter, setConfParam.
This is the sequence:
1. setConfParam to some node
2. OpenZwave "fire" value changed event.
3. Restart the software (scan of the network).
4. The device respond with old values!

nechry, 2017-06-08:
you have to manually wake up the sensor if you want the parameter will be apply. If you restart the network the pending message will be cleared.

japita-se, 2017-06-08:
Thnk you nechry. Although I think it is not the case,I will try again.

japita-se, 2017-06-08:
This is what I get after setting the Wattage Threshold param to 60 for node 14 (Energy Meter):

```
2017-06-08 12:29:58.832 Info, Node014, Value::Set - COMMAND_CLASS_CONFIGURATION - Wattage Threshold Whole HEM - 4 - 1 - 60
2017-06-08 12:29:58.834 Info, Node014, Configuration::Set - Parameter=4, Value=60 Size=2
2017-06-08 12:29:58.836 Detail, Node014, Queuing (Send) ConfigurationCmd_Set (Node=14): 0x01, 0x0d, 0x00, 0x13, 0x0e, 0x06, 0x70, 0x04, 0x04, 0x02, 0x00, 0x3c, 0x25, 0xef, 0x6d
2017-06-08 12:29:58.839 Detail, Node014, Queuing (Send) ConfigurationCmd_Get (Node=14): 0x01, 0x0a, 0x00, 0x13, 0x0e, 0x03, 0x70, 0x05, 0x04, 0x25, 0xf0, 0x4f
2017-06-08 12:29:58.841 Detail, 
2017-06-08 12:29:58.841 Info, Node014, Sending (Send) message (Callback ID=0xef, Expected Reply=0x13) - ConfigurationCmd_Set (Node=14): 0x01, 0x0d, 0x00, 0x13, 0x0e, 0x06, 0x70, 0x04, 0x04, 0x02, 0x00, 0x3c, 0x25, 0xef, 0x6d
2017-06-08 12:29:58.850 Detail, Node014,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2017-06-08 12:29:58.850 Detail, Node014,   ZW_SEND_DATA delivered to Z-Wave stack
2017-06-08 12:29:58.866 Detail, Node014,   Received: 0x01, 0x07, 0x00, 0x13, 0xef, 0x00, 0x00, 0x02, 0x06
2017-06-08 12:29:58.866 Detail, Node014,   ZW_SEND_DATA Request with callback ID 0xef received (expected 0xef)
2017-06-08 12:29:58.867 Info, Node014, Request RTT 24 Average Request RTT 23
2017-06-08 12:29:58.867 Detail,   Expected callbackId was received
2017-06-08 12:29:58.867 Detail,   Expected reply was received
2017-06-08 12:29:58.867 Detail,   Message transaction complete
2017-06-08 12:29:58.868 Detail, Node014, Removing current message
2017-06-08 12:29:58.868 Detail, 
2017-06-08 12:29:58.868 Info, Node014, Sending (Send) message (Callback ID=0xf0, Expected Reply=0x04) - ConfigurationCmd_Get (Node=14): 0x01, 0x0a, 0x00, 0x13, 0x0e, 0x03, 0x70, 0x05, 0x04, 0x25, 0xf0, 0x4f
2017-06-08 12:29:58.876 Detail, Node014,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2017-06-08 12:29:58.877 Detail, Node014,   ZW_SEND_DATA delivered to Z-Wave stack
2017-06-08 12:29:58.893 Detail, Node014,   Received: 0x01, 0x07, 0x00, 0x13, 0xf0, 0x00, 0x00, 0x02, 0x19
2017-06-08 12:29:58.893 Detail, Node014,   ZW_SEND_DATA Request with callback ID 0xf0 received (expected 0xf0)
2017-06-08 12:29:58.893 Info, Node014, Request RTT 24 Average Request RTT 23
2017-06-08 12:29:58.894 Detail,   Expected callbackId was received
2017-06-08 12:29:58.904 Detail, Node014,   Received: 0x01, 0x0c, 0x00, 0x04, 0x00, 0x0e, 0x06, 0x70, 0x06, 0x04, 0x02, 0x00, 0x3c, 0xb3
2017-06-08 12:29:58.904 Detail, 
2017-06-08 12:29:58.905 Info, Node014, Response RTT 36 Average Response RTT 35
2017-06-08 12:29:58.905 Detail, Node014, Refreshed Value: old value=60, new value=60, type=short
2017-06-08 12:29:58.906 Detail, Node014, Changes to this value are not verified
2017-06-08 12:29:58.906 Info, Node014, Received Configuration report: Parameter=4, Value=60
2017-06-08 12:29:58.907 Detail, Node014,   Expected reply and command class was received
2017-06-08 12:29:58.907 Detail, Node014,   Message transaction complete
2017-06-08 12:29:58.908 Detail, 
2017-06-08 12:29:58.908 Detail, Node014, Removing current message
2017-06-08 12:29:58.909 Detail, Node014, Notification: ValueChanged

```

japita-se, 2017-06-08:
...but after restart the OZW I get:
```
2017-06-08 15:38:11.520 Info, Node014, Response RTT 35 Average Response RTT 35
2017-06-08 15:38:11.520 Info, Node014, ZWave+ Info Received from Node 14
2017-06-08 15:38:11.520 Info, Node014,   Zwave+ Node Type  (0x00) - Z-Wave+ node. Mandatory Command Classes:
2017-06-08 15:38:11.520 Info, Node014,     NONE
2017-06-08 15:38:11.520 Info, Node014,   Zwave+ Device Type  (0x1000) - Sub Energy Meter. Mandatory Command Classes:
2017-06-08 15:38:11.520 Info, Node014,     COMMAND_CLASS_DEVICE_RESET_LOCALLY
2017-06-08 15:38:11.520 Info, Node014,     COMMAND_CLASS_ZWAVE_PLUS_INFO
2017-06-08 15:38:11.520 Info, Node014,     0x59 (Not Supported)
2017-06-08 15:38:11.521 Info, Node014,     COMMAND_CLASS_MANUFACTURER_SPECIFIC
2017-06-08 15:38:11.521 Info, Node014,     COMMAND_CLASS_POWERLEVEL
2017-06-08 15:38:11.521 Info, Node014,     COMMAND_CLASS_ASSOCIATION
2017-06-08 15:38:11.521 Info, Node014,     COMMAND_CLASS_VERSION
2017-06-08 15:38:11.521 Info, Node014,     COMMAND_CLASS_CRC_16_ENCAP
2017-06-08 15:38:11.521 Info, Node014,     COMMAND_CLASS_METER
2017-06-08 15:38:11.521 Info, Node014,   ZWave+ Role Type  (0x31) - Always On Slave
2017-06-08 15:38:11.521 Info, Node014,     COMMAND_CLASS_DEVICE_RESET_LOCALLY
2017-06-08 15:38:11.521 Detail, Node014, Initial read of value
2017-06-08 15:38:11.521 Detail, Node014, Initial read of value
2017-06-08 15:38:11.521 Detail, Node014, Initial read of value
2017-06-08 15:38:11.521 Detail, Node014,   Expected reply and command class was received
2017-06-08 15:38:11.521 Detail, Node014,   Message transaction complete
2017-06-08 15:38:11.521 Detail, 
2017-06-08 15:38:11.522 Detail, Node014, Removing current message
2017-06-08 15:38:11.522 Detail, Node014, Notification: ValueChanged
2017-06-08 15:38:11.522 Detail, Node014, Notification: ValueChanged
2017-06-08 15:38:11.522 Detail, Node014, Notification: ValueChanged
2017-06-08 15:38:11.522 Detail, Node014, Query Stage Complete (NodePlusInfo)
2017-06-08 15:38:11.522 Detail, Node014, AdvanceQueries queryPending=0 queryRetries=0 queryStage=SecurityReport live=1
2017-06-08 15:38:11.522 Detail, Node014, QueryStage_SecurityReport
2017-06-08 15:38:11.522 Info, Node014,   Opening config param file /usr/local/etc/openzwave/aeotec/zw095.xml
```
and the parameter I set is gone!

nechry, 2017-06-08:
you also remove your zwcfg file ?
2017-06-08 15:38:11.522 Info, Node014,   Opening config param file /usr/local/etc/openzwave/aeotec/zw095.xml
this entry is a new Add to the zwcfg file. at this step ozw file the zwcfg file with default values not the stored values stored in device memories. you must perform a refresh values and you will restore in zwcfg file the internal parameters. 

so it's not a issue is the default behavior.

japita-se, 2017-06-08:
So, I have to call requesAllConfigParams on all nodes?

nechry, 2017-06-08:
Yse if you, or your gateway remothe zwcfg file, it's the network cache. But normally you don't have to do this action

closed 2017-06-18 by Fishwaldo