OpenZWave/open-zwave #1230: Alarm Command Class V2: Incorrect AlarmCmd in command_classes/Alarm.cpp

skptic, 2017-05-26:
**Problem**

I have an alarm access keypad device which claims to send an Alarm Command Class Version 2 with an extra byte containing the UserCode.

The alarm type is `Alarm_Access_Control` and as described in the [wiki](https://github.com/OpenZWave/open-zwave/wiki/Alarm-Command-Class):

> the Alarm_Access_Control Type and Alarm_Access_Control_KeyPad_Lock Event exposes a Additional ValueByte containing the UserCode Position that was entered on the device.

This type is only available in Version 2 of the command class. 

Yet for the device to be recognised as Version 2 (and thus for OZW to process the extra ValueByte) the `command_classes/Alarm.cpp` [code](https://github.com/OpenZWave/open-zwave/blob/ad0d31e407aca6be328c82e58aa8b4ddcded42a4/cpp/src/command_classes/Alarm.cpp#L40-L47) requires that the byte after the command class (`0x71`) to be:

*Version 1:*
- AlarmCmd_Get = `0x04`
- AlarmCmd_Report = `0x05`

*Version 2:*
- AlarmCmd_SupportedGet -> `0x07`
- AlarmCmd_SupportedReport -> `0x08`

This is clearly not happening with this device with the message of the command recieved being for example:

`0x71, 0x05, 0x00, 0xff, 0x00, 0xff, 0x06, 0x05, 0x01, 0x02, 0x87`

Given that that `Alarm::HandleMsg` is evaluating:

`(AlarmCmd)_data[0]` == `0x05` == `AlarmCmd_Report`

it seems it is not possible to retrieve the extra ValueByte as it would if it was evaluated as `AlarmCmd_SupportedReport` and entered this condition [here](https://github.com/OpenZWave/open-zwave/blob/ad0d31e407aca6be328c82e58aa8b4ddcded42a4/cpp/src/command_classes/Alarm.cpp#L257-L290).

From this I have deduced that one of either two things is happening. Either:

1. OZW is incorrectly requiring the byte from `(AlarmCmd)_data[0]` to be `0x07` or `0x08` to be recognised as a version 2 command and thus retrieve the extra ValueByte.

2. The manufacturer is not adhering to version 2 and the device should be using `0x08`.

**Proposed Solution**
Either way, it seems that the `command_classes/Alarm.cpp` could be modified to allow the capture of the additional ValueByte.

**Device: Zipato Mini RFiD Keypad** 
- Open Zwave Config [here](https://github.com/OpenZWave/open-zwave/blob/ad0d31e407aca6be328c82e58aa8b4ddcded42a4/config/zipato/MiniKeypad.xml)
- Manual [here](https://www.zipato.com/wp-content/uploads/2015/10/wt-rfid-Zipato-Mini-Keypad-RFiD-Z-Wave-User-Manual-v1.4.pdf)

**Example Command Message Recieved:**
`0x71, 0x05, 0x00, 0xff, 0x00, 0xff, 0x06, 0x05, 0x01, 0x02, 0x87`

*Index -> Value:*
- 0: Command Class (113 = COMMAND_CLASS_ALARM)
- 1: AlarmCmd Report
- 2: Alarm Type
- 3: Alarm Level (0 or 255)
- 4: SourceNodeID
- 9: User Code
- 10: Burglar

Fishwaldo, 2017-05-26:
Firstly SupportedReport != Report. 

SupportedGet/Report is check the capabilities of the device. In this case, the type of Alarms it supports. Its not used to report Alarm Updates.

secondly, 
This has been addressed in the Dev Branch, where we support upto Version 3 of the AlarmClass. 

But we need to continue to work on it to get it upto Version 8 of the Notification CC (Renamed from Alarm Class after Version 3)

closed 2017-05-26 by Fishwaldo

skptic, 2017-05-26:
Thanks for the quick reply. The new code in Dev Branch makes a lot more sense, will try it out.

Fishwaldo, 2017-05-26:
It still needs work, and the way we expose things will change, but its a start.... Version 8 of the Notification CC has a lot of extra features. How to "expose" them to the app I haven't figured out yet.

openzwave-dev-watcher, 2017-05-29:
Do you need official  documentation ?

openzwave-dev-watcher, 2017-05-29:
The documentation is public, so no need. 

Justin Hammond

On 29 May 2017, at 4:28 PM, giuseppe d <demarco.giu@gmail.com <mailto:demarco.giu@gmail.com> > wrote:

Do you need official  documentation ?

Il giorno venerdì 26 maggio 2017 12:32:06 UTC+2, Justin Hammond ha scritto:

It still needs work, and the way we expose things will change, but its a start.... Version 8 of the Notification CC has a lot of extra features. How to "expose" them to the app I haven't figured out yet.

—

 

You received this message because you are subscribed to the Google Groups "OpenZWave" group.
To unsubscribe from this group and stop receiving emails from it, send an email to openzwave+unsubscribe@googlegroups.com <mailto:openzwave+unsubscribe@googlegroups.com> .
For more options, visit https://groups.google.com/d/optout.