OpenZWave/open-zwave #1235: Bitfield.h changes broke windows build

dotMorten, 2017-05-31:
The commit https://github.com/OpenZWave/open-zwave/commit/13e1585a659239ec2f715dce79260192b6ea495a has introduced a bunch of linker errors in the UWP Library build:

```
2>OpenZWave.lib(CommandClass.obj) : error LNK2019: unresolved external symbol "public: __cdecl OpenZWave::Bitfield::Bitfield(void)" (??0Bitfield@OpenZWave@@QEAA@XZ) referenced in function "public: __cdecl OpenZWave::CommandClass::CommandClass(unsigned int,unsigned char)" (??0CommandClass@OpenZWave@@QEAA@IE@Z)
2>OpenZWave.lib(CommandClass.obj) : error LNK2019: unresolved external symbol "public: __cdecl OpenZWave::Bitfield::~Bitfield(void)" (??1Bitfield@OpenZWave@@QEAA@XZ) referenced in function "int `public: __cdecl OpenZWave::CommandClass::CommandClass(unsigned int,unsigned char)'::`1'::dtor$0" (?dtor$0@?0???0CommandClass@OpenZWave@@QEAA@IE@Z@4HA)
2>OpenZWave.lib(CommandClass.obj) : error LNK2019: unresolved external symbol "public: void __cdecl OpenZWave::Bitfield::Set(unsigned int)" (?Set@Bitfield@OpenZWave@@QEAAXI@Z) referenced in function "public: void __cdecl OpenZWave::CommandClass::SetInstance(unsigned char)" (?SetInstance@CommandClass@OpenZWave@@QEAAXE@Z)
2>OpenZWave.lib(CommandClass.obj) : error LNK2019: unresolved external symbol "public: bool __cdecl OpenZWave::Bitfield::IsSet(unsigned int)const " (?IsSet@Bitfield@OpenZWave@@QEBA_NI@Z) referenced in function "public: void __cdecl OpenZWave::CommandClass::SetInstance(unsigned char)" (?SetInstance@CommandClass@OpenZWave@@QEAAXE@Z)
2>OpenZWave.lib(MultiInstance.obj) : error LNK2001: unresolved external symbol "public: bool __cdecl OpenZWave::Bitfield::IsSet(unsigned int)const " (?IsSet@Bitfield@OpenZWave@@QEBA_NI@Z)
2>OpenZWave.lib(CommandClass.obj) : error LNK2019: unresolved external symbol "public: unsigned int __cdecl OpenZWave::Bitfield::Iterator::operator*(void)const " (??DIterator@Bitfield@OpenZWave@@QEBAIXZ) referenced in function "public: bool __cdecl OpenZWave::CommandClass::RequestStateForAllInstances(unsigned int,enum OpenZWave::Driver::MsgQueue)" (?RequestStateForAllInstances@CommandClass@OpenZWave@@QEAA_NIW4MsgQueue@Driver@2@@Z)
2>OpenZWave.lib(CommandClass.obj) : error LNK2019: unresolved external symbol "public: class OpenZWave::Bitfield::Iterator & __cdecl OpenZWave::Bitfield::Iterator::operator++(void)" (??EIterator@Bitfield@OpenZWave@@QEAAAEAV012@XZ) referenced in function "public: bool __cdecl OpenZWave::CommandClass::RequestStateForAllInstances(unsigned int,enum OpenZWave::Driver::MsgQueue)" (?RequestStateForAllInstances@CommandClass@OpenZWave@@QEAA_NIW4MsgQueue@Driver@2@@Z)
2>OpenZWave.lib(CommandClass.obj) : error LNK2019: unresolved external symbol "public: bool __cdecl OpenZWave::Bitfield::Iterator::operator!=(class OpenZWave::Bitfield::Iterator const &)" (??9Iterator@Bitfield@OpenZWave@@QEAA_NAEBV012@@Z) referenced in function "public: bool __cdecl OpenZWave::CommandClass::RequestStateForAllInstances(unsigned int,enum OpenZWave::Driver::MsgQueue)" (?RequestStateForAllInstances@CommandClass@OpenZWave@@QEAA_NIW4MsgQueue@Driver@2@@Z)
2>OpenZWave.lib(CommandClass.obj) : error LNK2019: unresolved external symbol "public: class OpenZWave::Bitfield::Iterator __cdecl OpenZWave::Bitfield::Begin(void)const " (?Begin@Bitfield@OpenZWave@@QEBA?AVIterator@12@XZ) referenced in function "public: bool __cdecl OpenZWave::CommandClass::RequestStateForAllInstances(unsigned int,enum OpenZWave::Driver::MsgQueue)" (?RequestStateForAllInstances@CommandClass@OpenZWave@@QEAA_NIW4MsgQueue@Driver@2@@Z)
2>OpenZWave.lib(CommandClass.obj) : error LNK2019: unresolved external symbol "public: class OpenZWave::Bitfield::Iterator __cdecl OpenZWave::Bitfield::End(void)const " (?End@Bitfield@OpenZWave@@QEBA?AVIterator@12@XZ) referenced in function "public: bool __cdecl OpenZWave::CommandClass::RequestStateForAllInstances(unsigned int,enum OpenZWave::Driver::MsgQueue)" (?RequestStateForAllInstances@CommandClass@OpenZWave@@QEAA_NIW4MsgQueue@Driver@2@@Z)
```

From the looks of it, bitfield.h got its implementation removed, but was never moved somewhere else.

Fishwaldo, 2017-05-31:
I had to run half way though doing this... committed the missing file now.

closed 2017-05-31 by Fishwaldo

dotMorten, 2017-05-31:
@Fishwaldo  The file also needs to be added to the two vcxproj files.

Btw we could probably unify the two project files and have a shared project file to ease maintenance. That would mean requiring VS2015 or up, and drop the older VS2012/2012/2013 support. If you're good with that, I'll be happy to refactor the vcxproj files a little to simplify maintenance.

dotMorten, 2017-06-01:
Made a PR for updating the project files: https://github.com/OpenZWave/open-zwave/pull/1238