OpenZWave/open-zwave #1236: Deprecating scenes broke the UWP build

dotMorten, 2017-05-31:
UWP is failing the build on any deprecated calls. The recent deprecation of all scene methods means the build is now failing when linking.

I removed all the deprecated calls, but Manager.cpp still has a call to `RemoveScene( i )` in `RemoveAllScenes`, causing the build to fail even when not using deprecated methods.

Building for Win32 succeeds, but you still see this build warning
```
1>..\..\..\src\Manager.cpp(4059): warning C4996: 'OpenZWave::Manager::RemoveScene': was declared deprecated
1>\open-zwave\cpp\src\Manager.h(2075): note: see declaration of 'OpenZWave::Manager::RemoveScene'
```

@Fishwaldo

dotMorten, 2017-05-31:
Just noticed the build server also reporting this issue: 
http://bamboo.my-ho.st/bamboo/browse/OZW-OZWLNX6-77

Fishwaldo, 2017-05-31:
I love compiler differences! this worked fine on Clang!

dotMorten, 2017-05-31:
@Fishwaldo Close but no cigar :-)
```
1>c:\github\openzwave\openzwave-dotnet-uwp\open-zwave\cpp\src\manager.cpp(3564): error C2059: syntax error: '}'
1>c:\github\openzwave\openzwave-dotnet-uwp\open-zwave\cpp\src\manager.cpp(4052): error C2059: syntax error: '}'
1>c:\github\openzwave\openzwave-dotnet-uwp\open-zwave\cpp\src\manager.cpp(4061): error C2059: syntax error: '}'
1>c:\github\openzwave\openzwave-dotnet-uwp\open-zwave\cpp\src\manager.cpp(4062): error C2059: syntax error: '}'
1>c:\github\openzwave\openzwave-dotnet-uwp\open-zwave\cpp\src\manager.cpp(4845): error C2760: syntax error: expected ')' not 'end of file'
1>c:\github\openzwave\openzwave-dotnet-uwp\open-zwave\cpp\src\manager.cpp(4048): fatal error C1075: the left brace '{' was unmatched at the end of the file
```

dotMorten, 2017-05-31:
Looks like a closing parenthesis is missing here in `Defs.h`
```
#define OPENZWAVE_DEPRECATED_WARNINGS_OFF 	__pragma( warning(push) )\
											__pragma( warning(disable: 4996) 
```

Fishwaldo, 2017-06-01:
Doh! Fixed. (I hope)

closed 2017-06-01 by Fishwaldo