OpenZWave/open-zwave #1240: Aeotec MultiSensor 6 (ZW100) temperature calibration parameter not conform Help

Fuzzillogic, 2017-06-04:
Since firmware v1.07 of this sensor, the type of parameter 201 for temperature calibration has been changed from byte to short. It now also incorporates the type of adjustment (C/F). The change from Byte to Short is done in commit https://github.com/OpenZWave/open-zwave/commit/1d59209fbed61c70495ce0339985a7d6f1cec2e7. However, the Help section of this parameter hasn't been updated accordingly, it still reflects the old situation. Someone (like me) trying to use the calibration feature following the Help will get incorrect results or simply no results at all.

Aeotec has posted [more information and a spreadsheet](https://aeotec.freshdesk.com/support/solutions/articles/6000120736-calibrating-offsetting-temperature-on-multisensor-6-excel-easy-sheet-included) for the new situation. The values given by the spreadsheet works for positive adjustment, i.e. the measured temperature is too low. For negative adjustments it doesn't, at least using Domoticz, as the resulting values are >2^15-1. The 2's complement value must be entered here for it to work. E.g. I'm now using -2815 as value to adjust the measured temperature by -1.1°C in Domoticz.

According to the information it should still is possible to use Byte instead of Short for this parameter for v1.07. To me it appears that this will improve compatibility with firmwares older than v1.07, "fix" the help and is much easier to comprehend and calculate.

Fishwaldo, 2017-06-29:
Can you open a PR with the requested changes?

cincuranet, 2017-10-04:
I believe this can be closed, because it's `short` in [current code](https://github.com/cincuranet/open-zwave/blob/ae54a1fe797d71b9bd43df4a3a60ec06ec35fc32/config/aeotec/zw100.xml#L308).

arigit, 2018-09-14:
I ran into the same problem, having had to substract 65536 to the value calculated following aeotec instructions. The OZW help for the parameter shows:
 
A short between 4294934528 and 32767

Which is wrong (as the OP above, I am using a negative value here). Maybe the problem is that ozw currently takes a signed short instead of an unsigned. By changing to unsigned, ozw would be aligned with the sensor documentation for the parameter and also the current displayed range would make sense

closed 2018-10-26 by Fishwaldo