OpenZWave/open-zwave #1243: OZW fails to start when config file is removed

dechutes, 2017-06-06:
Hey all,
First, thanks for this software, it's been awesome for several years now. I recently upgraded to the latest code.  Everything worked fine for a week or two.  Yesterday I had to delete the cache file (used to be the config file) and now the system won't finish starting.  The last lines in the log file are:
2017-06-06 16:57:56.520 Info, Download Finished: C:\OZWServer\config\/fortrezz/mimolite.xml (Node: 7)
2017-06-06 16:57:56.523 Info, Node007, Reloading Node after new Config File Loaded
2017-06-06 16:57:56.526 Detail, Node007, Reloading Node
2017-06-06 16:58:06.310 Info, HttpThread Exiting. No Transfers in timeout period

The entire log file is attached.  

A few things:
The location in the log file is obviously wrong, I have used the custom locations for years and always worked fine.  Even now the files seem to load correctly, but the log seems to add / instead of \.  

The issue seems to revolve around the auto-downloading of the config files.  I have a lot of *.xml.1, *.xml.2 etc over the course of just a week or two. I have read up on that new feature and it says there is an option in the options file to disable it, but I can't find what that option is in any of the documentation.  Perhaps I am overlooking it?

After that message is logged the process goes to full cpu utilization on that thread.  

If I put my old cache file back everything works great again.

Any body have any thoughts?  

Thanks!
[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/1056477/OZW_Log.txt)

dechutes, 2017-06-06:
Update:
I looked and further up in the log is the below.  The log file indicates it needs revision 3, but when I manually download that file it says Revision 1.  Maybe that's the issue?  

2017-06-06 17:20:38.517 Info, Node007, Sending (Query) message (Callback ID=0x25, Expected Reply=0x04) - ManufacturerSpecificCmd_Get (Node=7): 0x01, 0x09, 0x00, 0x13, 0x07, 0x02, 0x72, 0x04, 0x25, 0x25, 0x96
2017-06-06 17:20:38.527 Detail, Node007,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2017-06-06 17:20:38.530 Detail, Node007,   ZW_SEND_DATA delivered to Z-Wave stack
2017-06-06 17:20:38.544 Detail, Node007,   Received: 0x01, 0x07, 0x00, 0x13, 0x25, 0x00, 0x00, 0x03, 0xcd
2017-06-06 17:20:38.547 Detail, Node007,   ZW_SEND_DATA Request with callback ID 0x25 received (expected 0x25)
2017-06-06 17:20:38.550 Info, Node007, Request RTT 30 Average Request RTT 30
2017-06-06 17:20:38.553 Detail,   Expected callbackId was received
2017-06-06 17:20:38.556 Detail, Node007,   Received: 0x01, 0x0e, 0x00, 0x04, 0x00, 0x07, 0x08, 0x72, 0x05, 0x00, 0x84, 0x04, 0x53, 0x01, 0x10, 0x4f
2017-06-06 17:20:38.560 Detail, 
2017-06-06 17:20:38.562 Info, Node007, Response RTT 43 Average Response RTT 43
2017-06-06 17:20:38.565 Info, Node007,   Opening config param file C:\OZWServer\config\/fortrezz/mimolite.xml
2017-06-06 17:20:38.568 Detail, Node007, Initial read of value
2017-06-06 17:20:38.570 Info, Node007, Queuing Lookup on 0110.0453.0084.db.openzwave.com for Node 7
2017-06-06 17:20:38.573 Detail, Node007, Initial read of value
2017-06-06 17:20:38.575 Info, Node007,   Configuration File Revision is 1
2017-06-06 17:20:38.578 Info, LookupTxT Checking 0110.0453.0084.db.openzwave.com
2017-06-06 17:20:38.581 Info, Lookup for 0110.0453.0084.db.openzwave.com returned 3
2017-06-06 17:20:38.586 Info, Node007, Received manufacturer specific report from node 7: Manufacturer=FortrezZ, Product=MIMOlite Wireless Interface/Bridge Module
2017-06-06 17:20:38.589 Info, Node007, Node Identity Codes: 0084:0453:0110
2017-06-06 17:20:38.592 Detail, Node007,   Expected reply and command class was received
2017-06-06 17:20:38.595 Detail, Node007,   Message transaction complete
2017-06-06 17:20:38.597 Detail, 
2017-06-06 17:20:38.600 Detail, Node007, Removing current message
2017-06-06 17:20:38.602 Detail, Node007, Notification: ValueChanged
2017-06-06 17:20:38.652 Detail, Node007, Notification: ValueChanged
2017-06-06 17:20:38.718 Detail, Node007, Notification: NodeNaming
2017-06-06 17:20:38.782 Detail, Node007, Initial read of value
2017-06-06 17:20:38.785 Warning, Node007, Config File for Device "MIMOlite Wireless Interface/Bridge Module" is out of date
2017-06-06 17:20:38.788 Info, Queuing download for http://download.db.openzwave.com/0110.0453.0084.xml (Node 7)
2017-06-06 17:20:38.792 Detail, Node007, Notification: ValueChanged
2017-06-06 17:20:38.844 Detail, Node007, Notification: User Alert - Config File out of Date
2017-06-06 17:20:38.893 Info, Download Finished: C:\OZWServer\config\/fortrezz/mimolite.xml (Node: 7)
2017-06-06 17:20:38.896 Info, Node007, Reloading Node after new Config File Loaded
2017-06-06 17:20:38.899 Detail, Node007, Reloading Node

dechutes, 2017-06-08:
Done some more testing, and I can confirm it is something with the auto-update code.  I added a host file record for openzwave.com to 127.0.0.1 (blocking all access to that site) and the file creates successfully and everything seems to work fine.  Please let me know if I can do any testing to help find this bug.

Thanks.

Fishwaldo, 2017-06-08:
Thanks for testing. If you have a debugger it would be good just to get a back trace when the CPU goes 100%. 

The infrastructure for the update code is in place but needs to be manually synced from the repository which is why the version numbers look a bit weird. 

(In fact the Dev Branch is a bit unstable right now, lots of new code going in... so it might be a fun ride for a few more weeks)

dechutes, 2017-06-08:
I'm just a lowly .net guy.  Haven't ever worked with c++.  I have visual studios on that system and can run in debug mode.  If you could give me some steps I could do that for you.

Fishwaldo, 2017-06-18:
Hrm... I'm not a .net guy... so I can only guess depending on how the wrapper is done @OpenZWave/openzwave-dotnet-uwp  maybe can help.

I would imagine you would just run your app under Visual Studio, and when it hangs, hit the pause button and you should get a backtrace of all the threads?

Fishwaldo, 2018-10-26:
Closing due to lack of feddback

closed 2018-10-26 by Fishwaldo