OpenZWave/open-zwave #1249: Endpoint on SES302 with SES001

Servuc, 2017-06-14:
Hello guys,

I have a problem with the [SES302](http://products.z-wavealliance.org/products/1240) and [SES001](http://www.newagesystems.eu/resources/uploads/Thermostat/SEC_SIR321/Probe_datasheet.pdf).

SES001 is an external temperature sensor, connected by One Wire physically and by endpoint on ZWave network on SES302, like Secure support said to me : 

> Hi 
> It looks like that the association has not been set with enpoint2(SES001 nedpoint).
> Have you created the Multichannel association with endpoint2?

So I try to set in OpenZWave Control Panel (OZCP), but I can't.

I see after that SES302 configuration ( [link](https://github.com/OpenZWave/open-zwave/blob/master/config/horstmann/ses302.xml) ) doesn't contain : `<CommandClass id="96" mapping="endpoints" />` like Fibaro Two Switch ( [link](https://github.com/OpenZWave/open-zwave/blob/master/config/fibaro/fgs222.xml#L147) ). I'm not sure this is the solution.

I've had it in `.xml` configuration file of SES302, but nothing appear in configuration possibilities on OZCP (after restart it).

Have I miss someting ? 

Have a good day :)

Servuc, 2017-06-18:
Sorry for my late response,

I've removed the SES 302, re-add it, but isn't seem enough. 
Nevertheless, a new value appear in OZCP, but value is very strange,(-270°C approx - not Kelvin value).

I'm working on it at my job, I will give more update few ;)

Thanks ;)

Fishwaldo, 2018-10-26:
closing due to lack of feedback

closed 2018-10-26 by Fishwaldo