OpenZWave/open-zwave #1250: POPP 10 year Smoke sensor recognized as POPP Solar Siren

Momotica, 2017-06-14:
The POPP 10 year Smoke Sensor is recognized as POPP Solar Siren.

http://products.z-wavealliance.org/products/2078

nechry, 2017-06-14:
the product type 0x0004  and id 0x0004 mentioned in z-wavealliancedatabase is the entry I put in ozw file:
https://github.com/OpenZWave/open-zwave/blob/master/config/manufacturer_specific.xml#L832

it's possible popp use the same devices ids with 2 different devices

Momotica, 2017-06-14:
Thanks for the reply.
I just tested 4 smoke sensors, but all ar recognized as type="0004" id="0002", same as the Outdoor siren. 

I will test the siren to see how that device is include.
The battery needs to be charged first ;)

Momotica, 2017-06-16:
I received an firmware update from POPP, all sensors got the wrong firmware :(

nechry, 2017-06-16:
Can you be more explicit?

geirra, 2017-06-18:
This is actually documented in config file of the 10 year unit.

Popp managed to send smoke sensors to market with the wrong pcb, which included also wrong firmware.
The unit works, but has wrong ID.

Check if your PCB is v1.5 or higher, and you need to firmware 1.14+ 
Check info in the file 
https://github.com/OpenZWave/open-zwave/blob/master/config/popp/009402.xml

Fishwaldo, 2017-06-23:
this is not a OZW issue then....

closed 2017-06-23 by Fishwaldo