OpenZWave/open-zwave #1259: wrong manufacturer DOMUX in XML

michapr, 2017-06-21:
<Manufacturer id="0260" name="Domux">
=> this seems to be wrong.

id 0260 - this is Heiman - can be checked at alliance page
as sample
  <ManufacturerId>0x0260</ManufacturerId>
  <ProductTypeId>0x8004</ProductTypeId>
  <ProductId>0x1000</ProductId>

  <Name>Smart Water Leakage Sensor</Name>
  <Brand>HEIMAN</Brand>
  <Identifier>HS1WL-Z</Identifier>
  <CertificationNumber>ZC10-17055580</CertificationNumber>

DOMUX seems to be an OEM - it is not a memer of alliance.
All device names (like "DX1WL-Z") must be removed and a folder for Heiman should be created.

Devices can be found here:
http://products.z-wavealliance.org/Search/Index?regionId=-1&searchText=heiman

Michael

nechry, 2017-06-21:
Can you provide a PR with all renamed files?

michapr, 2017-06-21:
[corrected_heiman.zip](https://github.com/OpenZWave/open-zwave/files/1091947/corrected_heiman.zip)

please remove domux directory.
Have corrected the manufacturer_specific.xml too.

commit 074cc16e (2017-06-21): Fix wrong manufacturer Heiman

fix #1259
DOMUX is the OEM version
closed 2017-06-21 by nechry