OpenZWave/open-zwave #1263: Binary Switch V2 CC

Fishwaldo, 2017-06-29:
Add Version 2 of the Spec to our command class. The main difference is:
1) added duration to SET messages (4.12.2)
2) Target Value from a Report message (4.12.3) - This would address #1261 if thats a v2 implementation
3) Duration Value from a Report message

added bonus points: update Value::VerifyRefreshedValue to not have to do another GET if we received a target value.

AlCalzone, 2017-09-06:
I'm on it, more later...

Fishwaldo, 2017-09-06:
Cool. Against the Dev Branch please!

AlCalzone, 2017-09-08:
Regarding 2.:
> Target Value from a Report message (4.12.3) - This would address #1261 if thats a v2 implementation
Would you expose the target value to applications as a separate value? 
Or just use it internally? If so, how?

closed 2017-09-21 by Fishwaldo