OpenZWave/open-zwave #1266: Cooper RFTR9505-T Get/Set Association Loop

waxhell, 2017-07-06:
When a Cooper RFTR9505-T is added to a network, the get/set association routines are looping -- possibly due to the special handling case where the number of association groups is 255.

I have attached two log files -- both of a pairing as well as starting after it has been paired. Node 14 was the first attempt at adding one of these devices and Node 15 was a second attempt with a different device (to rule out an issue with the first device). For the pairing log, I cleared out the config xml files prior to starting openzwave.
[OZW_Log-Pairing.txt](https://github.com/OpenZWave/open-zwave/files/1126685/OZW_Log-Pairing.txt)
[OZW_Log-Starting.txt](https://github.com/OpenZWave/open-zwave/files/1126684/OZW_Log-Starting.txt)

Fishwaldo, 2017-07-06:
can you replace the config/cooper/RF9505-T.xml with this version, exclude and reinclude the device and tell me if it resolves the issues. (this device doesn't have 255 groups - It appears to be a bug in the firmware)

[RF9505-T.xml.zip](https://github.com/OpenZWave/open-zwave/files/1126876/RF9505-T.xml.zip)

waxhell, 2017-07-06:
Looks like we are still looping -- Node 21 is the device in the attached log.
[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/1127022/OZW_Log.txt)

Did a quick google search and looks like other people have had some issues that necessitated xml edits on another Cooper device (but they edited this device's XML file).  The following seems pretty hackish:
https://www.reddit.com/r/homeautomation/comments/4qye9j/any_overwallswitch_zwave_switches/d577hes/

There's also some Cooper reference notes online for associations, but sounds like the reporting isn't matching correctly with what the device is putting out.

http://www.cooperindustries.com/content/dam/public/wiringdevices/products/documents/technical_specifications/advancedtechinfo_V2.pdf?_ga=2.88101656.245637702.1499326046-226758195.1499326046

Fishwaldo, 2017-07-06:
Doesn't look like you properly replaced the config File. Its still reporting Group #1:

2017-07-04 22:28:30.627 Info, Node021, Adding the controller to group 1 (Group #1) of node 21

That name comes from the config file.

waxhell, 2017-07-06:
I replaced the config file and made sure there was a change. Also made sure
that it was in the virtualenv that I was running.

In my xml file:

        
        <CommandClass id="133">
                <Associations num_groups="1">
                        <Group index="255" max_associations="1"
label="Group #255" auto="true"/>
                </Associations>
        </CommandClass>
</Product>

Fishwaldo, 2017-07-06:
That "Group #1" text can only come from the config file. Double check your locations.

waxhell, 2017-07-06:
My bad, replacing the file does work properly. -- not sure why homeassistant replicates the file in a second place. Helps when I look at the path in the logs.

My virtualenv is in my /src/homeassistant folder but it's pulling the configs from the /home/homeassistant/.homeassistant/deps folder.

Fishwaldo, 2017-07-07:
Glad it works. The cooper stuff is pretty outdated as far as Zwave is concerned. Might want to consider something more upto date in the future

closed 2017-07-07 by Fishwaldo