OpenZWave/open-zwave #1267: Door Lock user codes not all dispalying

Ittiz, 2017-07-07:
I noticed what seems to be a bug in OZW.  I have a Kwikset 910 Z-wave dead bolt.  It can have up to 30 user codes.  You can insert a user code into any position you want in the lock.  I set the first 4 user codes you be the first 4 user positions in the lock.  Then I have 2 guests who also use the lock.  I set their user code positions to be 10 and 11 because I wanted to have space to add more non guest users and not get the two confused.  Anyway OZWCP only shows the first 4 and seems to assume the other two at 10 and 11 do not exist.

techgaun, 2017-07-07:
@Ittiz Looks like your config might be setting `RefreshAllUserCodes` to false. You can see more information here: https://github.com/OpenZWave/open-zwave/wiki/Config-Options
The default set by ozw itself is false: https://github.com/OpenZWave/open-zwave/blob/ad0d31e407aca6be328c82e58aa8b4ddcded42a4/config/options.xml#L17

Ittiz, 2017-07-07:
I checked and it was set to false.  I set it to true, restarted the server, gave it time and woke up the lock for good measure.   I also hit the refresh too,  It still has the issue.

Fishwaldo, 2017-07-08:
There is also a button - "Refresh All User Codes" that should work. If it doesn't. I need Logs.

Ittiz, 2017-07-09:
Seems to have fixed itself.  Just took a couple days for some reason.  So I'm all set.  Thanks for your help.

closed 2017-07-11 by nechry