OpenZWave/open-zwave #1270: PHILIO PAN04-1B wrong states reported

michapr, 2017-07-12:
including PAN04-1B device with actual XML will happen following:

switching ON switch1 will be switched on switch 1
switching ON switch2 will be switched on switch 1 and 2 - switch 1 not physical
switching OFF switch2 will be switched off switch 1 and 2 - switch 1 not physical (was not on)
switching ON switch3 (common) will be switched ON only switch1 - but physical switch1 and switch2

I have seen the same in OZW panel, so it is not domoticz related.

Attached the log for using switch 2:
[philio_ozw.txt](https://github.com/OpenZWave/open-zwave/files/1142142/philio_ozw.txt)
log for using switch3 (common):
[philio_ozw2.txt](https://github.com/OpenZWave/open-zwave/files/1142143/philio_ozw2.txt)

Removing association to live line it will work... but then should poll the device for detecting change status...

Maybe you have an idea?

mpoi, 2017-08-21:
hi michapr,

I have more problems using philio PAN04-1 device. Your errors also exists in my case. Can you confirm right values of the previous reading instances? There are confusing values mixing here :-(

kind regards, zHeim

michapr, 2017-08-23:
Hi,
I do not have the device here more - it is already at customers place...
But the "only" problem was the described issue.

I have removed the association to group1 and enabled the polling for now.
It is not the right solution, but less confusion for the user.

Still waiting for any comment that this could be solved - the issue is only in OpenZwave present.

Fishwaldo, 2017-09-21:
Try adding (or changing) the config file for the device to:
`<CommandClass id="142" ForceInstances="true"/>`

mpoi, 2017-11-07:
Doesn't work. Still whirrling around values :-(

nibblerrick, 2018-02-06:
I'm using this switch with IOBroker/OpenZWave, too and have also the problem with the wrong states. Refreshing the node-info gets the right states.

don66, 2018-06-28:
I have problem as well with the PAN-04 and Endpoints using Home Assistant- it leads to wrong states for the switches , as well the sensors getting mixed up values.
It looks like the configuration parameter 3 Selected end points is wrongly defined in HA: 1: Relay 1 and Relay 2. 2: Relay 1. 3: Relay 2
It should be 1: Relay 1 2: Relay 2 3: Relay 1 and Relay 2
(Additionally the configuration parameters 7 to 10 is missing.)