OpenZWave/open-zwave #1271: Higher value in /config/northq/nq92021.xml needed

joreri, 2017-07-16:
Hi,

The energy meter NorthQ 92021can be configured to how many pulses per 10 kWh. The max value in the xml-file is 65535. My house meter has 10 000 pulses per 1 kWh, which give a value of 100 000 pulses.

Running Domoticz and are not able to configure my NorthQ with correct value.

https://www.domoticz.com/forum/viewtopic.php?f=6&t=18286

Is it possible to increase max?

  
  <CommandClass id="112">
    <Value type="short" index="1" genre="config" label="Pulse factor" units="pulses" min="1" max="65535"  value="1024">
      <Help>How many pulses are counted per 10 kWh.</Help>
    </Value>

Best regards,
joreri

nechry, 2017-07-17:
the parameter max value is related to the parameter size so the type (byte =1, short=2, int=4) 
the device accept only value with the right defined value size. 
You can try to change the size of parameter 1 in your local configuration file and check if reader accept any values changes.

michapr, 2017-07-17:
> My house meter has 10 000 pulses per 1 kWh, which give a value of 100 000 pulses.

You are sure? What kind of meter is this (type)?
This would mean that the meter reports 0,1Wh - so 100mWh ... never heared about such a device... ;)

joreri, 2017-07-17:
Hi and thanks for reply!
@nechry I will try change the max to 100 000 and report back here. Thanks!
@michapr Yes, at least it is label so on the meter. The type is pulse (sort of the question..) I think it is a very common meter in the nordic countries at least. Check my link above or picture: https://farm2.staticflickr.com/1555/24412057391_7060372c36_b.jpg

Thanks,
joreri

Fishwaldo, 2017-07-18:
looking at the technical manual and comparing to the config file, seems like the config file is not correct:
http://northq.com/wp-content/uploads/2016/01/Technical-Integration.pdf

Regarding this particular issue, Config Param 1 is actually a Int (though in our config file its a short). So you could change the config file to short and try setting it to 100000

@nechry maybe you want to look at the config file and see if we can fix it up?

joreri, 2017-07-18:
@Fishwaldo You mean change to Int? It is short by default.

Before change:
 <Value type="short"
After change:
<Value type="Int"

I'll try this and get back.

Thanks,
joreri

nechry, 2017-07-18:
ok after small digging it the same configuration 
https://github.com/OpenZWave/open-zwave/commit/a1b2811d13757b0736d66c2ea496c1a131d9f6fc
@joreri  please confirm it ok and we can close the issue

joreri, 2017-07-18:
Hi,
I can confirm that the new line #10 in config/northq/nq92021.xml works, eg my controller (Domoticz) is able to accept my desired value of 100000 pulses per kWh.

Thank you for rapid attention.

nechry, 2017-07-18:
perfect

closed 2017-07-18 by nechry