OpenZWave/open-zwave #1280: Zwave Fibaro FS-223 Switch 2 doesn't work reliably

minchik, 2017-07-21:
I have Aeotec ZW090 Z-Stick Gen5 and Fibaro System FGS223 Double Relay. It has two switches. The first one works great. But I have to toggle second switch twice to work properly.
If I turn on it at first time, it actually turns on the light but web interface doesn't recognize that it was turned on. If I turn on it again, web interface understands new status. Same thing for turning off.
This is not just my issue, it is actually wide spread across Home Assistant users. At first I though  that this is home-assistant bug, but then I repeated It using OpenZWave Control Panel.

I'v recorded [screen cast](https://youtu.be/FWz2PvbGtak)
My [zwave-log.txt](https://github.com/OpenZWave/open-zwave/files/1166901/zwave-log.txt)
[Home Assistant issue](https://github.com/home-assistant/home-assistant/issues/7076)

nechry, 2017-07-22:
The controller need to be with instance 1 in Lifeline group

minchik, 2017-07-22:
Ok, I'm newbie in z-wave and openzwave particularly. Do you mean this:
[image]

nechry, 2017-07-22:
Yes you must have 1.1 not 1

minchik, 2017-07-22:
But there is no such option
[image]

nechry, 2017-07-22:
Its suppose to be automatically perform during the inclusion process. It's a fresh new included device? I don't know how to force the instance 1 with your tool, In jeedom gateway its always available to choose  instance 1 or not.

minchik, 2017-07-23:
My controller with nodeId 1 is in Lifeline group of fibaro switch. I think that the issue is not in associations, as first channel of the fibaro switch works perfect. The problem is in the second channel only.

Could you read **OpenHAB Integration** part of that article https://www.nilsschneider.de/wp/2016/09/11/fibaro-double-switch-2-fgs-223-und-openhab/ I think this is my case

nechry, 2017-07-23:
I have this device and i fully tested. Without specifing the instance 1 I got same bad behavior with the device. With 1.1 (controller and instance) the 2 end points work as expected

minchik, 2017-07-23:
May be my answer is stupid 😄 , but this device doesn't have group 1.1 in associations. 

http://manuals.fibaro.com/switch-2/

nechry, 2017-07-24:
1.1  representing controller.instance, usually node 1 is the controller so 1.1 in group Lifeline

minchik, 2017-07-24:
You mean 1.1 should be there?

[image]

minchik, 2017-07-24:
By the way, all the screenshots are taken from [open-zwave-control-panel](https://github.com/OpenZWave/open-zwave-control-panel) which, I think, is developed by the owners of open z-wave project

nechry, 2017-07-24:
In jeedom it 's possible to do this. A change is required is ozcp to allow choose instance 1 of the controller.

nechry, 2017-07-24:
ensure your device configuration files are up to date in your ozcp. Exclude the node and include it again. Is supposed to be automatically get the 1.1

nechry, 2017-07-24:
see: https://github.com/OpenZWave/open-zwave/blob/master/config/fibaro/fgs223.xml#L267
ForceInstances="true" is the key

nechry, 2017-07-24:
small note: check my commit
https://github.com/OpenZWave/open-zwave/commit/d39a11ed43856437f719088eb4850d74535ee36e#diff-be80d77d3784d4c46d8a2cacd2acbe5b
your device is included in secure ? if yes me be is your issue

minchik, 2017-07-24:
No my device is not included in secure mode. So I have to include it securely?

minchik, 2017-07-24:
If that doesn't help, is it possible to install  jeedom, change this option and than use my software? Is that config parameter stored directly in relay?

nechry, 2017-07-24:
I got a issue if is included securely, so no stay in normal way.
Check you local fibaro/fgs223.xml file and ensure is same a repo file.

Yes you can install jeedom with zwave plugin do to this, your network topology is stored in your dongle and the association group and configuration in your device. If you try jeedom you will keep it... lol!

nechry, 2017-07-24:
if you wan try jeedom, start from a fresh Debian 8 install
`
wget https://raw.githubusercontent.com/jeedom/core/stable/install/install.sh
chmod +x install.sh
./install.sh
`

minchik, 2017-07-24:
Now I know what was my issue. I've added a relay to the z-wave network as described in the user manual of the Aeotec stick. I have pulled out the stick from the computer and then pressed the button on the stick. But this process is error prone. All devices should be added to z-wave network directly from ozcp.
@nechry Thank you very much!!!
 I think this issue can be closed. Will you close it or I should close it myself?

nechry, 2017-07-24:
great new, you welcome

closed 2017-07-24 by nechry